greiman/SdFat #318: No instance of overloaded function "SdFile::read"

adamalfath, 2021-08-03:
I have previous code that use SdFile class with read function on opened file. On the latest version it giving me this error `no instance of overloaded function "SdFile::read" matches the argument list -- argument types are: (const char [30], int) -- object type is: SdFileC/C++(304)`
Does it get removed? What is the equivalent method to handle that operation?

PlatformIO 5.1.1
Lib 2.0.7

greiman, 2021-08-03:
You will need to provide more info.  I ran the following:
```
#include "SdFat.h"
SdFat sd;
SdFile file;
char buf[30];
void setup() {
  Serial.begin(9600);
  sd.begin(SS);
  file.open("test.txt");
  size_t n = file.read(buf, 30);
  Serial.write(buf, n);
}
void loop() {}
```
The output is:
> Content of test.txt

I don't understand  'const char' in: 

> no instance of overloaded function "SdFile::read" matches the argument list -- argument types are: (const char [30], int) -- object type is: SdFileC/C++(304)

adamalfath, 2021-08-06:
Thanks for the response. Unfortunately (or fortunately) I can't reproduce the issue, I don't know why. But now I have other issue, when SPIFFS library is added it throwing this error `no instance of overloaded function "SdFile::open" matches the argument list -- argument types are: (const char [15], const char [2]) -- object type is: SdFileC/C++(304)`. It's similar with previous issue, I wonder if it is compatibility issue?

Here's my program:
```
bool testFunc(void) {
  SdFat sd;

  if (sd.begin(SD_SS_PIN)) {
    SdFile file;
    const char filename[15] = "test.txt";
    char text[30] = "Hello from MCU!";

    if (file.open(filename, FILE_WRITE)) {
      file.print(text);
      file.close();
    } else {
      return false;
    }

    if (file.open(filename, FILE_READ)) {
      file.read(text, 30);
      file.close();
    } else {
      return false;
    }
  } else {
    return false;
  }

  return true;
}
```

greiman, 2021-08-06:
I suspect you are using ESP32.  New versions of the ESP32 often conflict with SdFat.  

About 12 years ago Arduino did a wrapper for SdFat and defined symbols FILE_WRITE, FILE_READ like this.
```
#define FILE_READ O_READ
#define FILE_WRITE (O_READ | O_WRITE | O_CREAT | O_APPEND)
```
I decided to add these definitions to SdFat.  

Recently ESP systems have change many definitions in their version of libraries.  Most definitions are in the file FS.h which I suspect SPIFFS uses.  Here are the conflicting definitions:
```
#define FILE_READ       "r"
#define FILE_WRITE      "w"
```
This causes open to fail since FILE_READ and FILE_WRITE are now char[2], not int.
You will likely have continuing problems using SdFat on ESP systems.  Best to use SD.h on ESP32.