greiman/SdFat #319: Suggestion for overrun flag separation for adc[]

ClaudeJGreengrass, 2021-08-11:
In my ignorance I modified ExFatLogger functions logRecord(), printRecord() to integrate Adafruit's LSM6DS33 sensor to store the accelerometer and gyro data in the adc[] array.  Unfortunately this data is type float so I used this union:
union i_a_o {
  float sensorData;
  uint16_t EFLdata[2];
} in_and_out;
to get data into the the array adc[] and to read it back out.  My ignorance was that the logger code uses the top bit 0X8000 of adc[0] to indicate an overrun.

I suggestion is the add a overrunflag to handle this situation:
const size_t ADC_COUNT = 4;
struct data_t {
  uint16_t adc[ADC_COUNT];
  uint16_t overrunflag;
};

Yes it costs a couple of bytes more which are in limited supply on the Uno but to me, it is much cleaner to separate this flag from other user data.  Alternatively, just flag this as an area that the user needs to modify when modifying the code.

The limitation of data on the Uno forced me off it and onto a Mega.  Adafruit's libraries and structures are easy to use but are not particularly efficient in data usage for any particular data sensor type.

all the best and thank you for the brilliant and very useful code.

greiman, 2021-08-11:
>  Alternatively, just flag this as an area that the user needs to modify when modifying the code.

I did flag it and tried to indicate with === to delimit the section to be replaced.
```
//==============================================================================
// Replace logRecord(), printRecord(), and ExFatLogger.h for your sensors.
void logRecord(data_t* data, uint16_t overrun) {
...

//==============================================================================
```
I mainly used the high bit for overrun because if you log a single ADC channel, adding a 16-bit flag doubles the record size and the size of the binary file. This double the write cost.  I didn't use an 8-bit flag since in many cases it will be padded to 16-bit alignment.

There are now Arduino boards with 16-bit ADCs but by default they run at 10-bits unless you call `analogResolution(bits)`.

ClaudeJGreengrass, 2021-08-13:
I understand and I now see that my suggestion was not very good.  Thanks for the prompt reply.  Please close this issue.