greiman/SdFat #320: STM32_test example does not compile on latest version and STM core

toogooda, 2021-08-12:
Firstly I would like to say the capability and documentation for this library is amazing so thank you :)

The latest update of SdFat seems to have made the STM32_test.ino example out of date.

`static SPIClass mySPI2(2);`   no longer works as there is no matching constructor for  
> SPIClass::SPIClass(int)

You now need to specify the SPI pins something like this:

`static SPIClass SPI_2 (PB15, PB14, PB13);`   For SPI2

greiman, 2021-08-13:
Which STM32 board support package are you using? There are several.  

I have not decided how to support STM32 so have not modified examples.  

At this point there are almost 200 Arduino style boards so I can only provide specialized examples for a limited number of boards.

toogooda, 2021-08-15:
Thanks for your response,
The official core by ST `stm32duino/Arduino_Core_STM32` latest version 2.0.0
I am using F103RET but I am pretty sure they all use the same SPI constructor...
ST also make it easy by making the SPI/I2c/USART on the same pins across most of there range so a simple example with SPI 1 and SPI2 should work on their entire range.

greiman, 2021-08-15:
Yes, the ST package uses these constructors for all boards:
```
    SPIClass();
    SPIClass(uint8_t mosi, uint8_t miso, uint8_t sclk, uint8_t ssel = (uint8_t)NC);
```

I wrote the example when the only STM32 support was [here](https://github.com/rogerclarkmelbourne/Arduino_STM32) with this constructor:
`  SPIClass(uint32 spiPortNumber);`

I will provide an option to use either package.  

You are lucky to use the F103RET.  I use high end STM32 boards with [ChibiOS](https://www.chibios.org/dokuwiki/doku.php?id=start) but many high end boards fail with the ST Arduino package.

[Here](https://github.com/stm32duino/Arduino_Core_STM32/issues/1294) is an issue for NUCLEO-H743ZI2 I posted on Feb 8, 2021.  ST planed a fix for 2.0.0 but now plans for 2.1.0 or later.

toogooda, 2021-08-16:
Both options sounds perfect :)
Thanks again for this great library!