greiman/SdFat #323: Convert library from using SD.h to SdFat.h

hicksan, 2021-09-11:
Hi Bill, I am using your SdFat library in a project with an ESP8266 main processor and an SD card reader. Thus far all fine but I now need to parse a CSV file so would like to use the (short) CSV-Parser-for-Arduino library (github: https://github.com/michalmonday/CSV-Parser-for-Arduino). Unfortunately this is written for SD.h so I need to convert it to SdFat in order to get my sketch to compile without errors. It has only one function that uses an SD process; only a few lines in total.

Looking at CSV-Parser.cpp, line 4 :
#include <SD.h>
I have replaced this with:
#include <SdFat.h>
using namespace sdfat;
SdFat sd;

Then line 111:
File csv_file = SD.open(f_name);
if (!csv_file) {
...
becomes:
File csv_file;
if (!csv_file.open(f_name, FILE_READ)) {
...

This appears to work but I would like to check I have done it all correctly.
Thanks very much indeed.

greiman, 2021-09-11:
That should be OK.  I am not sure about the current internal version of SD.h for ESP8266, ESP32 has conflicts with SdFat.

hicksan, 2021-09-11:
Yes I know. SdFat works with ESP8266 once you delete the standard version of the library. It would be good to find a better way.
I got the above solution to work and it runs fine. Thanks very much for all your work on this superb library, especially for picking up support threads which must be a huge burden. It is very much appreciated.

marcboon, 2021-10-30:
> Yes I know. SdFat works with ESP8266 once you delete the standard version of the library. It would be good to find a better way.

A better way is to edit the library.json of the latest SdFat library, and replace 'target=*' with target=avr,esp8266' or whatever architecture you are using. They key is to explicitly name the esp8266 architecture as a target, so Arduino's library conflict resolver will choose the latest SdFat in favour of ESP8266SdFat.

Sologix, 2021-11-10:
Why doesn't SdFat have a compatible interface with SD? I'm trying to play Wav files with the Teensy audio library but it uses SD and I want the ability to use long file names. If the interface was compatible this could've been done with some typedefs, but it now looks I'll need to modify the source code.

greiman, 2021-11-10:
SD.h on Teensy is a wrapper for SdFat and supports long file names.  You will need to ask about the Teensy Audio Library in the Teensy forum.

greiman, 2021-11-10:
Actually a number of board packages use an SdFat wrapper for their SD.h.  Arduino still used a 2009 version of SdFat.  Teensy uses a more recent version of SdFat. ESP2866 uses another version of SdFat.

They all add or change something so SdFat will never "be compatible" with SD.h.

Sologix, 2021-11-12:
I got it working by upgrading my Teensyduino installation!