greiman/SdFat #327: SdFat with ESP8266

hicksan, 2021-09-23:
Hi I am using an ESP8266 within Arduino IDE 1.8.16 with SdFat.
On compiling I get loads of errors relating to SdFat. I have tried various board manager libraries for the ESP8266, from 3.0.2 back to 2.5.0 but get errors with all of them, mostly libraries clashing and the IDE not sure which to use. I read on forums to delete the Arduino version, but the directory appears to have restricted permissions and will not give me access. Any suggestions please
[error.txt](https://github.com/greiman/SdFat/files/7220750/error.txt)
?

greiman, 2021-09-24:
ESP8266 is using SdFat internally for SD.h so you can't use two versions of SdFat in your program.

marcboon, 2021-10-29:
To use the latest SdFat with ESP8266 I changed SdFat's library.json to target=esp8266 (instead of *). This way the library conflict resolver will pick the SdFat library instead of the ESP8266SdFat library. Make sure you don't have any references to SD in your code. SD will clash with SdFat. Now you can #include <SdFat.h> and use SdFs/FsFile instead of SD/File.

pankaj-gecko, 2022-04-01:
Can you please elaborate how did you made this work? I am also facing the same issue during compilation on my Wemos D1 R1.

marcboon, 2022-04-01:
> Can you please elaborate how did you made this work? I am also facing the same issue during compilation on my Wemos D1 R1.

My fix was from last year. Since then several updates to ESP8266SdFat and Greimann's SdFat have been released, so I don't know what the status is today. I haven't touched it since then. The main thing is to make sure Arduino picks only one version of SdFat. I did that by changing the target definition in the file libraries/SdFat/library.json but in the current version that file doesn't exist. If you still get conflicts between the two libraries, I suggest you temporarily remove one of them.

pankaj-gecko, 2022-04-01:
Thanks for your reply. I was trying to downgrade current latest sdfat library but id did not worked out. I tried deleting it (Deleting it's Adafruit counterpart which I had installed earlier) and even uninstalling arduino ide but no go. However, to my amazement I just went ahead and deleted the sdfat library and then recompiled it for Wemos D1 R1... And to my surprise it didn't even detected the micro sd card reader but also printed the contents inside sd card successfully. It's weird but issue resolved.Thanks for following up with me. ---- On Fri, 01 Apr 2022 13:01:19 +0530  ***@***.***  wrote ----

Can you please elaborate how did you made this work? I am also facing the same issue during compilation on my Wemos D1 R1.

My fix was from last year. Since then several updates to ESP8266SdFat and Greimann's SdFat have been released, so I don't know what the status is today. I haven't touched it since then. The main thing is to make sure Arduino picks only one version of SdFat. I did that by changing the target definition in the file libraries/SdFat/library.json but in the current version that file doesn't exist. If you still get conflicts between the two libraries, I suggest you temporarily remove one of them.