greiman/SdFat #328: RAW write to the SD-Card

karstengit, 2021-10-04:
This is not a bug - it is a feature request.

Is it possible to write direct to sectors of the SD-card using SdFat?

This could be helpful when a filesystem is not needed.
But actual there is another problem with a virus that should be solved without using an OS:
https://www.easeus.com/partition-master/sd-card-mounted-as-read-only.html

We have tried everything to get rid of the write protection, but every OS (Linux/Windows) always sees the card as write protected medium and denies to write on it.

Your formatting example is working and says it has cleared the card, but afterwards nothing has happened.
The card is still write protected and shows a (bad) file system on the card in Linux.

So it seems your format never touch the partition table that must be overwritten now.

```
 init time: 0x1e7 ms

Card type: SDXC

Manufacturer ID: 0X28
OEM ID: BE
Product:
Version: 0.2
Serial number: 0X3B0300BA
Manufacturing date: 2/2010

cardSize: 128109.77 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: true

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0X0,0X2,0X0,0XEE,0XFF,0XFF,0XFF,1,250214399
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

volumeBegin failed. Is the card formatted?
```

How can this be done with SdFat ?

The example RawWrite.ino seems to write to a file too and cannot overwrite the partition table.

karstengit, 2021-10-04:
That's the way the partition of the card is shown in Linux
```
Warning: Unable to open /dev/sdc read-write (file system read-only).
/dev/sdc has been opened read-only.
Model: Generic STORAGE DEVICE (scsi)                                      
Disk /dev/sdc: 128GB
Sector size (logical/physical): 512B/512B
Partition Table: gpt
Disk Flags: 

Number  Start   End     Size    File system  Name            Flags
 1      1049kB  17,8MB  16,8MB               android_meta
 2      17,8MB  128GB   128GB                android_expand
```

A complete clean of the SD-Card is needed.

greiman, 2021-10-04:
There are many custom formats and partition layouts used for SD cards but SdFat only supports the SD association standard format for SD cards.  SD card hardware internals and firmware algorithms for modern SD cards are optimized for this format.  

You can try their [formatter](https://www.sdcard.org/downloads/formatter/) but will probably get the same result.

The SdSpicard class can be used to for raw read/write of SD cards and the core library has support for multiple partition.  I just don't have time or interest to support all the issues that occur with apps that do raw I/O or non-standard SD formats.

Here is an example that can do raw read/write and supports multiple partitions.  

```
#include "SdFat.h"
SdSpiCard card;
FatVolume vol;
FatFile file;
uint8_t buf[512];
#define SD_CONFIG SdSpiConfig(SS,SHARED_SPI, SD_SCK_MHZ(20))
#define MBR_PART 1
#define error(msg) {Serial.println(msg);while (true);}

void setup() {
  Serial.begin(9600);
  if (!card.begin(SD_CONFIG)) error("card.begin");
  //  Read sector print signature.
  if (!card.readSector(0, buf)) error("card.readSector");
  Serial.print(buf[510], HEX);
  Serial.println(buf[511], HEX);  
  if (!vol.begin(&card, MBR_PART)) error("vol.begin");
  if (!file.open("test.txt", O_RDWR | O_CREAT)) error("file.open");
  if (file.write("test line\n") != 10) error("file.write");
  file.close();
  Serial.println("done");
}
void loop() {
}
```
The output is:
> 55AA
> done

Note that FatFile does not support Print.  The File class is a template class that is derived from Stream and FatFile.

karstengit, 2021-10-04:
Thank you for your quick response!
I am really thankful for your wonderful library that makes it possible to log data direct as files on an SD-Card.
So Thank You for your work and this library!

You are really not forced to give support for all the issues that occur with apps that do raw I/O. :-)
In this case the result was the work of an virus on an Android device.

Any tool that works on an PC or other Device with an "real OS" like Posix or Windows fails to recover the SD-Card, because the SD-Card is identifying somehow as read-only device and the kernel is blocking any write afterwards.

That is the reason i tried to recover it with an Microcontroller (STM32), that has no kernel with "unneeded routines" to check such nasty things. ;-)

In your example SdInfo.ino you read the MBR:
```
bool mbrDmp() {
  MbrSector_t mbr;
  bool valid = true;
  if (!sd.card()->readSector(0, (uint8_t*)&mbr)) {
    cout << F("\nread MBR failed.\n");
    errorPrint();
    return false;
  }
  cout << F("\nSD Partition Table\n");
  cout << F("part,boot,bgnCHS[3],type,endCHS[3],start,length\n");
```

It's hard to understand what you are doing there, but at least i only need to overwrite the first sectors with 0.
Is there a corresponding sd.card()->writeSector() and how it can be used?

That's all support i need. :-)

karstengit, 2021-10-04:
O.K. Found the documentation now:

writeSector()
bool SdSpiCard::writeSector 	( uint32_t  sector, const uint8_t *  src ) inline

Writes a 512 byte sector to an SD card.

Parameters
    [in]	sector	Logical sector to be written.
    [in]	src	Pointer to the location of the data to be written.

Returns
    true for success or false for failure. 

Is there already an buffer of 512 Byte that can be filled and used?

greiman, 2021-10-04:
To zero the first sector just add the following to the above example.
```
  memset(buf, 0, 512);
  if (!card.writeSector(0, buf)) error("card.writeSector");
```
After this the MBR sector will be entirely zero so a reformat will be needed.

greiman, 2021-10-04:
I just noticed the card is GPT.  The MBR is not used.  It is just [protective](https://en.wikipedia.org/wiki/GUID_Partition_Table#Protective_MBR_(LBA_0)).

Did you try the SdFormatter example with the 'E' option.  It uses the hardware erase option to clear every sector of flash to the default erase state.

karstengit, 2021-10-05:
Yes - all options are tested without success.
And as already told every option in Linux and Windows was tested to format and to repartition.

The interesting thing is the kernel message "[sdc] Write Protect is on" so there must be an option for it.
Maybe such a thing has been specified as replacement for the write protect switch on the SD-Card?
Nobody uses it and a hacker has find it and has fun with it.

greiman, 2021-10-05:
You may be able to save the card if simple write protection was used.

https://www.diskpart.com/articles/remove-write-protection-on-micro-sd-card-sandisk.html

You will need a utility like the AOMEI Partition Assistant.

greiman, 2021-10-06:
The card was definitely formatted as internal in an Android device.  I went back and found this earlier in your post. 

> Number  Start   End     Size    File system  Name            Flags
>  1      1049kB  17,8MB  16,8MB               android_meta
>  2      17,8MB  128GB   128GB                android_expand

Look for articles like this:

https://forum.fairphone.com/t/a-little-guide-to-using-sd-cards/37297

It should not be easy to access data on a secure card so that's what you get if you select secure/encrypted.

Look for Android articles about Internal (adopted) storage and External (portable) storage.

karstengit, 2021-10-07:
Thanks again.

It seems that as internal storage the external SD is somehow mounted into the root filesystem.
Google has blemished more and more the Linux kernel in Android and there are tons of mounts in a running system.

`Formatting the SD card as internal storage on the FP3 under android 10 is confirmed to break storage on the card and to make some things unusable. Forum topic here`

When i remember correct, the issue with the unwritable flash comes suddenly in the usage of the smartphone, without a new format or change / update of the Android. That was the reason to think for the work of an virus.

The data content is irrelevant - the aim is to get the SD-Card usable again for further use.
I don't know if the card has been become full or what the user has done in detail.

greiman, 2021-10-07:
You can see why I have avoided using any of the mysterious "Secure Digital" features of cards.  Some users have asked for this. 

I don't even support an API for the simplest write protect of micro-SD cards, it's too easy to lock them forever since there is write once protect bit.

Apple and Google want to sell products that are secure at the level of stopping the CIA. 

Using an SD card with encryption and DRM is too fragile.  I think many manufacturers like Samsung have disabled the internal/adopted feature for SD cards in the latest Android.

greiman, 2021-10-07:
Don't blame the SD-Card manufacturer - it was never intended for this use.