greiman/SdFat #340: Calling timestamp to set specific date/tims on a file on ExFat volume can corrupt the volume

KurtE, 2021-10-29:
As you know a few of us including @mjs513 are working on trying to integrate MTP and MSC support on the Teensy.  

One of the things we have run into, is that at times when we are running our MTP stuff on an exFat disk, and the host wants to send us an object, some of the paramters may in include the create and modify dates and times, which we then use to set the new file we created from the data. 

Some of the time these volumes become corrupted.  After experimenting it appears like this happens when set the dates and times at the end of process of receiving the file.  

To isolate cause, I created a quick and dirty sketch, that uses the new T4 SD wrapper, that demonstrates it.

```
/*
  SD card read/write

  This example shows how to read and write data to and from an SD card file
  The circuit:
   SD card attached to SPI bus as follows:
 ** MOSI - pin 11, pin 7 on Teensy with audio board
 ** MISO - pin 12
 ** CLK - pin 13, pin 14 on Teensy with audio board
 ** CS - pin 4, pin 10 on Teensy with audio board

  created   Nov 2010
  by David A. Mellis
  modified 9 Apr 2012
  by Tom Igoe

  This example code is in the public domain.

*/

#include <SD.h>
#include <SPI.h>

File myFile;

// change this to match your SD shield or module;
// Arduino Ethernet shield: pin 4
// Adafruit SD shields and modules: pin 10
// Sparkfun SD shield: pin 8
// Teensy audio board: pin 10
// Teensy 3.5 & 3.6 & 4.1 on-board: BUILTIN_SDCARD
// Wiz820+SD board: pin 4
// Teensy 2.0: pin 0
// Teensy++ 2.0: pin 20
const int chipSelect = BUILTIN_SDCARD;
#define BUFFER_SIZE 2048
uint8_t write_buffer[BUFFER_SIZE];

void setup()
{
  while (!Serial && millis() < 5000);
  Serial.begin(9600);

  if (CrashReport) {
    Serial.print(CrashReport);
  }
  Serial.println("\n" __FILE__ " " __DATE__ " " __TIME__);
  delay(3000);
  Serial.print("Initializing SD card...");

  if (!SD.begin(chipSelect)) {
    Serial.println("initialization failed!");
    return;
  }
  Serial.println("initialization done.");

  // open the file.
  Serial.println("Write first large file");
  myFile = SD.open("Logic+Setup+2.3.39-master.xxx", FILE_WRITE);
  uint32_t bytes_remaining = 162667688;
  // if the file opened okay, write to it:
  if (myFile) {
    DateTimeFields dtf={0, 10, 7, 0, 22, 7, 121};
    Serial.print("Writing to Logic+Setup+2.3.39-master.xxx...");
    memset(write_buffer, 1, sizeof(write_buffer));
    uint32_t write_size = BUFFER_SIZE;
    uint32_t loop_counter = 0;
    while (bytes_remaining) {
      if ((loop_counter++ & 0x3ff)== 0) Serial.print(".");
      if (bytes_remaining < write_size) write_size = bytes_remaining;
      myFile.write(write_buffer, write_size);
      bytes_remaining -= write_size;
    }
    myFile.setCreateTime(dtf);
    myFile.setModifyTime(dtf);
    myFile.close();
    Serial.println("\ndone.");
  }

  bytes_remaining = 26917168;
  // if the file opened okay, write to it:
  myFile = SD.open("sixaxispairtoolsetup-0.3.1.xxx", FILE_WRITE);
  if (myFile) {
    DateTimeFields dtf={0, 2, 5, 0, 22, 9, 121};
    Serial.print("Writing to sixaxispairtoolsetup-0.3.1.xxx...");
    memset(write_buffer, 1, sizeof(write_buffer));
    uint32_t write_size = BUFFER_SIZE;
    uint32_t loop_counter = 0;
    while (bytes_remaining) {
      if ((loop_counter++ & 0x3ff)== 0) Serial.print(".");
      if (bytes_remaining < write_size) write_size = bytes_remaining;
      myFile.write(write_buffer, write_size);
      bytes_remaining -= write_size;
    }
    myFile.setCreateTime(dtf);
    myFile.setModifyTime(dtf);
    myFile.close();
    Serial.println("\ndone.");
  }
}

void loop()
{
  // nothing happens after setup
}
```
Sorry quick and dirty, creating two large files.... As these were the sizes of the files I received with MTP... 

Without the calls:
```
    myFile.setCreateTime(dtf);
    myFile.setModifyTime(dtf);
```
I could take the disk back to windows and windows say no issues on either a Fat32 (32GB) SDor a ExFat(128 GB SD Card)

But with these lines,  I do see corruption. 
Windows gives me the error popup, and if I try to run cmd line
```
Access is denied.

The volume is in use by another process. Chkdsk
might report errors when no corruption is present.
Volume Serial Number is 688F-47C5
Windows is verifying files and folders...
Volume label is SD_EXFAT.
Corruption was found while examining files in directory \ (7).
Corruption was found while examining files in directory \ (8).
Corruption was found while examining files in directory \ (9).
Corruption was found while examining files in directory \ (10).
Corruption was found while examining files in directory \ (11).
Corruption was found while examining files in directory \ (12).
Corruption was found while examining files in directory \ (13).
Corruption was found while examining files in directory \ (14).
An error occurred while examining files and directories.

Windows has checked the file system and found problems.
Run CHKDSK with the /F (fix) option to correct these.
```

Again have not seen these issues with FAT32...

Also note:   The function in SD:
```
	virtual bool setCreateTime(const DateTimeFields &tm) {
		if (tm.year < 80 || tm.year > 207) return false;
		return sdfatfile.timestamp(T_CREATE, tm.year + 1900, tm.mon + 1,
			tm.mday, tm.hour, tm.min, tm.sec);
	}
	virtual bool setModifyTime(const DateTimeFields &tm) {
		if (tm.year < 80 || tm.year > 207) return false;
		return sdfatfile.timestamp(T_WRITE, tm.year + 1900, tm.mon + 1,
			tm.mday, tm.hour, tm.min, tm.sec);
	}
```

Now to continue trying to see where the issue might be...

greiman, 2021-10-29:
Probably the exFAT directory set checksum is not correctly updated in

`bool ExFatFile::timestamp()`

greiman, 2021-10-29:
Try modifying  ExFatFileWrite.cpp at about line 496

change this:
`  for (uint8_t is = 0;; is++) {`
to
`  for (uint8_t is = 0; is <= m_setCount ; is++) {`

I don't understand how your example even ran.  Not sure this will fix the problem but it fixed a simple test program I ran.

KurtE, 2021-10-29:
Thanks That appears to fix the problem.   At least with my test case, which I will assume it will fix

Should I close this one...

closed 2022-02-14 by KurtE