greiman/SdFat #341: How to open utf-8 files?

marcboon, 2021-10-29:
Using SdFat 2.1.0 I can open directories and list files with utf-8 file names, but I could not open such files using the utf-8 file name.

For example this code snippet:
```
#define USE_UTF8_LONG_NAMES
#include <SdFat.h>

FsFile dir = sd.open(path, O_RDONLY); // this fails if path contains non-ascii chars
  if(!dir) return false;
  FsFile file;
  char name[256];
  while(file.openNext(&dir, O_RDONLY)) {
    file.getName(name, sizeof(name)); // this will get the utf-8 file name, but I can't use it to open the file if it contains non-ascii chars
```

What am I'm doing wrong?

greiman, 2021-10-29:
You must enable utf-8 by editing SdFatConfig.h

At about line 37 of SdFatConfig.h remove the comment slashes form this:
`// #define USE_UTF8_LONG_NAMES 1`

marcboon, 2021-10-30:
> You must enable utf-8 by editing SdFatConfig.h
> 
> At about line 37 of SdFatConfig.h remove the comment slashes form this: `// #define USE_UTF8_LONG_NAMES 1`

As you can see in my code I #define USE_UTF8_LONG_NAMES 1 before  #include <SdFat.h>.
SdFatConfig.h contains the lines:
````
#ifndef USE_UTF8_LONG_NAMES
#define USE_UTF8_LONG_NAMES 0
#endif  // USE_UTF8_LONG_NAMES
````
So my code uses utf-8 long file name (USE_LONG_FILE_NAMES is defined to be 1 by default). That's why I can SEE utf-8 names in directory listings. Without defining USE_UTF8_LONG_NAMES filenames containing non-ascii characters display those characters as '?'.

My problem is however: how do I OPEN files with a name containing non-ascii characters?
Is there a maybe way to obtain a 8.3 name for those files as well, and use that to open the file? (like it was customary in Windows many years ago)?

greiman, 2021-10-30:
https://github.com/greiman/SdFat/blob/master/examples/UnicodeFilenames/UnicodeFilenames.ino

marcboon, 2021-10-30:
> https://github.com/greiman/SdFat/blob/master/examples/UnicodeFilenames/UnicodeFilenames.ino

Running this example I get:
`sd.mkdir failed
`
I'm using a Samsung EVO+ SDXC card of 128GB capacity. I can read utf-8 files names from this card, but can not open those files (other files with only ascii names are fine). I also tried a 1GB SD1 card, result is the same.

greiman, 2021-10-30:
Here is the result I get with a 128GB Samsung EVO Plus.
```
Type any character to begin
ls:
      131072 😀/
            20 россиянин
            17 très élégant
             9 狗.txt
After remove and rmdir
Done!
```
I used Arduino IDE 1.8.16.

I have no idea what your problem is.
The card was freshly formatted by the SD Association formatter.

marcboon, 2021-10-30:
Very mysterious indeed.
Could it have anything to do with the underlying SPI library? I don't expect so, since it would treat all data simply as bytes, without any knowledge of encoding, right? All utf-8 specific stuff is handled in your SdFat library, as far as I know.
I'm also using Arduino 1.8.16 (and 2.0.0-beta) with esp8266-3.0.2 (latest), but made sure that your SdFat library 2.1.0 is used, and not ESP8266SdFat.

greiman, 2021-10-30:
> Could it have anything to do with the underlying SPI library?

The ESP8266 library is junk.  Try reducing the SPI rate to 4MHz from 16MHz here:

```
// Try to select the best SD card configuration.
#if HAS_SDIO_CLASS
#define SD_CONFIG SdioConfig(FIFO_SDIO)
#elif ENABLE_DEDICATED_SPI
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SD_SCK_MHZ(16))
#else  // HAS_SDIO_CLASS
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SD_SCK_MHZ(16))
#endif  // HAS_SDIO_CLASS
```
I also use a custom wrapper that tries to speed up ESP that uses block transfers.  It works well with ESP32.

Try disabling the optimization by selecting slow byte transfers. by setting SPI_DRIVER_SELECT  to one.
```

// Driver options
/**
 * If the symbol SPI_DRIVER_SELECT is:
 *
 * 0 - An optimized custom SPI driver is used if it exists
 *     else the standard library driver is used.
 *
 * 1 - The standard library driver is always used.
 *
 * 2 - An external SPI driver of SoftSpiDriver template class is always used.
 *
 * 3 - An external SPI driver derived from SdSpiBaseClass is always used.
 */
#ifndef SPI_DRIVER_SELECT
#define SPI_DRIVER_SELECT 0
#endif  // SPI_DRIVER_SELECT
/**
 * If USE_SPI_ARRAY_TRANSFER is non-zero and the standard SPI library is
 * use, the array transfer function, transfer(buf, size), will be used.
 * This option will allocate up to a 512 byte temporary buffer for send.
 * This may be faster for some boards.  Do not use this with AVR boards.
 */
#ifndef USE_SPI_ARRAY_TRANSFER
#define USE_SPI_ARRAY_TRANSFER 0
#endif  // USE_SPI_ARRAY_TRANSFER
```

marcboon, 2021-10-30:
> The ESP8266 SPI library is junk. Try reducing the SPI rate to 4MHz from 16MHz here:

I'm already using 4MHz. In my code I call
```
sd.begin(SD_CS, SD_SCK_MHZ(4));
```

> Try disabling the optimization by selecting slow byte transfers. by setting SPI_DRIVER_SELECT to one.

Tried that, but didn't make a noticeable difference.

I did notice another oddity: the cardsize is often misread. My 128GB card is listed as 2MB, while my 1GB and 4GB cards are listed correctly as 1024MB and 3957MB. I use the same code as in your SdInfo example (after sd.begin(), of course):
```
csd_t csd;
uint32_t sizeMB = 0.000512 * sdCardCapacity(&csd);
```

greiman, 2021-10-30:
I suggest you format newer SD card with [Sd Formatter](https://www.sdcard.org/downloads/formatter/).  Modern card's buffing, flash management and other algorithms  are designed for the standard layout.

Here is what the SdInfor example gives for my Samsung Evo Plus 128GB.

I have a bug fix for Manufacturing date that will be in the next release.  The year is in binary, not BCD:

`  cout << (2000 + m_cid.mdt_year_low + 10 * m_cid.mdt_year_high) << endl;`
replaced by
`  cout << (2000 + m_cid.mdt_year_low + 16 * m_cid.mdt_year_high) << endl;`

> SdFat version: 2.1.0
> 
> Assuming the SD is the only SPI device.
> Edit DISABLE_CS_PIN to disable an SPI device.
> 
> Assuming the SD chip select pin is: 10
> Edit SD_CS_PIN to change the SD chip select pin.
> 
> type any character to start
> init time: 3 ms
> 
> Card type: SDXC
> 
> Manufacturer ID: 0X1B
> OEM ID: SM
> Product: ED4QT
> Version: 3.0
> Serial number: 0X3535D25
> Manufacturing date: 6/2018
> 
> cardSize: 128043.71 MB (MB = 1,000,000 bytes)
> flashEraseSize: 128 blocks
> eraseSingleBlock: true
> 
> OCR: 0XC0FF8000
> 
> SD Partition Table
> part,boot,bgnCHS[3],type,endCHS[3],start,length
> 1,0X0,0XA,0X9,0X2,0X7,0XFE,0XFF,0XFF,32768,250052608
> 2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 
> Scanning FAT, please wait.
> 
> Volume is exFAT
> sectorsPerCluster: 256
> clusterCount:      976640
> freeClusterCount:  976634
> fatStartSector:    49152
> dataStartSector:   65536

Here is a newer 64GB Evo Plus:

> SdFat version: 2.1.0
> 
> Assuming the SD is the only SPI device.
> Edit DISABLE_CS_PIN to disable an SPI device.
> 
> Assuming the SD chip select pin is: 10
> Edit SD_CS_PIN to change the SD chip select pin.
> 
> type any character to start
> init time: 2 ms
> 
> Card type: SDXC
> 
> Manufacturer ID: 0X1B
> OEM ID: SM
> Product: EC1S5
> Version: 3.0
> Serial number: 0X6A578D15
> Manufacturing date: 11/2020
> 
> cardSize: 64088.96 MB (MB = 1,000,000 bytes)
> flashEraseSize: 128 blocks
> eraseSingleBlock: true
> 
> OCR: 0XC0FF8000
> 
> SD Partition Table
> part,boot,bgnCHS[3],type,endCHS[3],start,length
> 1,0X0,0XA,0X9,0X2,0X7,0XFE,0XFF,0XFF,32768,125140992
> 2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 
> Scanning FAT, please wait.
> 
> Volume is exFAT
> sectorsPerCluster: 256
> clusterCount:      488704
> freeClusterCount:  488699
> fatStartSector:    49152
> dataStartSector:   65536

greiman, 2021-10-30:
If you ever let an app or other OS format a card it may be impossible to correctly format it unless you "power wash" it.

I just let the Micro Soft media maker for Windows 11 make an install disk on a 64GB SD in a usb adapter.  It made the SD appear as a 32GB FAT volume and I had to zero the first part of the card to reformat it correctly.

marcboon, 2021-10-31:
I ran your SdInfo.ino again on my 128GB card, and everything seems right. It's a brand new card, I never formatted it myself, just copied my music library onto it. Here's the output of SdInfo:
```
Card type: SDXC

Manufacturer ID: 0X1B
OEM ID: SM
Product: ED2S5
Version: 3.0
Serial number: 0X8A63B3E5
Manufacturing date: 9/2021

cardSize: 128177.93 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 
blockseraseSingleBlock: true

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0XA,0X9,0X2,0X7,0XFE,0XFF,0XFF,32768,250314752
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

Scanning FAT, please wait.

Volume is exFAT
sectorsPerCluster: 256
clusterCount: 977664
freeClusterCount: 695694
fatStartSector: 49152
dataStartSector: 65536
```
There must be a difference somewhere between my application code and SdInfo, in the section obtaining the card size. Or it might be hardware related (it's still wired up on a breadboard). I never had problem reading (big) files though. It's just the utf-8 filenames that won't open. Everything else works fine. I will investigate further. Thanks for your time!

marcboon, 2021-10-31:
I played a bit with your UnicodeFilenames example, and it only breaks when using non-ascii characters in file names. When I change the example file names to contain only ascii, those files can be opened/created, even when declared as a u8"" string. However, if the name contains one or more 8 bit characters (>0x7f), the file fails to open. 

I wonder, did you run this example on ESP8266?

marcboon, 2021-10-31:
> I did notice another oddity: the cardsize is often misread. My 128GB card is listed as 2MB, while my 1GB and 4GB cards are listed correctly as 1024MB and 3957MB. I use the same code as in your [...]

Fixed! I forgot to include: sd.card()->readCSD(&csd)

greiman, 2021-10-31:
I tested on ESP32 and it works.

I no longer test on ESP8266 since SdFat is used in SD.h on ESP8266.

There are now over 200 Arduino like boards so when a version of SdFat is used in the board support package, I stop testing on that board.  There are too many problems with multiple versions of a library for board.

I also only test with the Arduino IDE editor since other editors sometimes don't support utf-8. 

Be warned that I will soon close issues about SdFat on ESP8266 with advice to contact maintainers of the SD.h wrapper for SdFat.

marcboon, 2021-10-31:
> I tested on ESP32 and it works.

Ok. I will test it on a ESP32S2 board I have here, but my intention was to use ESP8266 (D1 mini) for this project.
 
> I no longer test on ESP8266 since SdFat is used in SD.h on ESP8266.
> Be warned that I will soon close issues about SdFat on ESP8266 with advice to contact maintainers of the SD.h wrapper for SdFat.

The problem is that ESP8266 is way behind, it still uses SdFat 2.0.2. I already contacted the maintainer about that, and he wrote that he will try to catch up. In the meantime I removed SD from my ESP8266 projects, in favour of your latest version of SdFat. I mentioned earlier in other threads that it's just a matter of changing the target in SdFat's library.json, replacing the wildcard (*) with explicit targets. Thanks a lot for the great work that you're doing!

greiman, 2021-10-31:
I found an old Adafruit HUZZAH ESP8266.  The UnicodeFilenames example runs with only one mod.

`#define SD_FAT_TYPE 3 // <<<--- was zero.`
```
Type any character to begin
ls:
      131072 😀/
            20 россиянин
            17 très élégant
             9 狗.txt
After remove and rmdir
Done!
```
I just don't have time for more of this!

marcboon, 2021-10-31:
> I found an old Adafruit HUZZAH ESP8266. The UnicodeFilenames example runs with only one mod.
> 
> `#define SD_FAT_TYPE 3 // <<<--- was zero.`

I already included that from the start, for exFAT support.

> I just don't have time for more of this!

I understand. I won't bother you again until I find the solution to my problem :)
I also installed 2.1.1 already. Thanks again!

marcboon, 2021-10-31:
I think I know what the problem is: it looks like ExFatFile::parsePathName() does not support utf-8 names.
While FatFile::parsePathName() in FatFileLFN.cpp has conditional code for USE_UTF8_LONG_NAMES, the exFatFile version does not, and it fails when chars are >0x7f
I found out that this function fails when it is called from sd.open() on exFat. At least in my application it does.

greiman, 2021-10-31:
> I think I know what the problem is: it looks like ExFatFile::parsePathName() does not support utf-8 names.

The conditional for ExFatFile::parsePathName() is hidden in this function which is called from parsePathName.

```
inline bool lfnLegalChar(uint8_t c) {
#if USE_UTF8_LONG_NAMES
  return !lfnReservedChar(c);
#else  // USE_UTF8_LONG_NAMES
  return !(lfnReservedChar(c) || c & 0X80);
#endif  // USE_UTF8_LONG_NAMES
}
```

Post the utf-8 string for the filename that fails.

marcboon, 2021-10-31:
I added some debug output to ExFatFile::parsePathName() where it fails:
```
    if (!lfnLegalChar(c)) {
      Serial.printf("parsePathName: fail on illegal char: %02x\n", c);
      DBG_FAIL_MACRO;
      goto fail;
    }
```
And then when I call sd.open("/Björk/") I get:
`parsePathName: fail on illegal char: c3`
because the ö in Björk is encoded as c3 b6 in utf-8

marcboon, 2021-10-31:
I just discovered that  USE_UTF8_LONG_NAMES is set to 0 at some point. Even though I #define it to be 1 at the start.
This is output of my code, not the change of USE_UTF8_LONG_NAMES, which I print at various points in y code, the last one in parsePathName, just before it bails out:

```
Using SdFat 2.1.1
USE_UTF8_LONG_NAMES = 1
SD Card initialized
Card size: 128 GB
Card type: SDXC
File system: exFAT

HTTP server started
USE_UTF8_LONG_NAMES = 1
GET /index.html
GET /code.js
GET /style.css
GET /sd/
GET /sd/Björk/
USE_UTF8_LONG_NAMES = 0
parsePathName: fail on illegal char: c3
Error opening /Björk/
```
I will try to find out how that happened

greiman, 2021-10-31:
That explains why the utf-8 escape 0XC3 is illegal in an ASCII string. 

One of the first things I told you was:

> You must enable utf-8 by editing SdFatConfig.h
> 
> At about line 37 of SdFatConfig.h remove the comment slashes form this:
> // #define USE_UTF8_LONG_NAMES 1

Here is the test I did on a exFAT SD
```
Type any character to begin
ls:
      131072 Björk/
            20 россиянин
            17 très élégant
             9 狗.txt
After remove and rmdir
Done!
```
You really test my patience.

marcboon, 2021-10-31:
> One of the first things I told you was:
> 
> > You must enable utf-8 by editing SdFatConfig.h
> > At about line 37 of SdFatConfig.h remove the comment slashes form this:
> > // #define USE_UTF8_LONG_NAMES 1

Yes, you told me that, and then I replied:

> As you can see in my code I #define USE_UTF8_LONG_NAMES 1 before #include <SdFat.h>.
> SdFatConfig.h contains the lines:
>
> #ifndef USE_UTF8_LONG_NAMES
> #define USE_UTF8_LONG_NAMES 0
> #endif  // [...]

So I was under the impression that it should work like this, because I could read utf-8 chars in filenames, instead of ?s 
The reason I did it like this is because I don't like to edit the library files for a specific project, while another project might need other settings. So I put all #defines in my sketch, relying on the #ifndef blocks in SdFatConfig.h. Now it turns out the #defines in my sketch don't propagate to all library code when compiled. I still don't understand why the directory listing did work with utf-8, but opening files didn't.

But I should have listened to you and do it as you intended it, because now it does work. Very sorry to have wasted your time. Have a nice evening!

closed 2021-10-31 by marcboon

greiman, 2021-10-31:
The Arduino IDE does not have a way to set global defines so some part of your code was compiled with your define but most of a library is compiled in a separate unit.

I have the conditional defines for more advanced build methods like make, cmake or platformIO.