greiman/SdFat #343: Hidden files

Miguel0101, 2021-11-05:
How to hide a file?

marcboon, 2021-11-06:
Start the filename with a dot.

Miguel0101, 2021-11-06:
> Start the filename with a dot.
But I would need to create an if to check if the file or folder starts with "."

marcboon, 2021-11-06:
There is a function FatFile::isHidden() to test if a file is hidden. It returns true if the file is a hidden file.

greiman, 2021-11-06:
There is no function to set the hidden attribute.  My to do list has writing a function like the windows [atrib](https://docs.microsoft.com/en-us/windows-server/administration/windows-commands/attrib) command.

The function would set or clear the Read-only, Archive, System, and Hidden attribute.

Probably more like the Linux `int chmod(const char *path, mode_t mode);`

mode would be the OR of :
```
const uint8_t FS_ATTRIB_READ_ONLY = 0x01;
const uint8_t FS_ATTRIB_HIDDEN    = 0x02;
const uint8_t FS_ATTRIB_SYSTEM    = 0x04;
const uint8_t FS_ATTRIB_ARCHIVE   = 0x20;

  // This call would set hidden and clear Read-only, System, and Archive.
  sd.attrib("myfile.txt", FS_ATTRIB_HIDDEN);
```
attrib(path) would return the attributes of a file.

It will first appear in SdFat-beta at some point.

Miguel0101, 2021-12-16:
Sounds good.

greiman, 2021-12-17:
I have implemented an attrib() function in[ SdFat-beta](https://github.com/greiman/SdFat-beta).

Here is how it works:

```
  /**
   * \return user settable file attributes for success else -1.
   */
  int attrib() {
    return isFileOrSubDir() ? m_attributes & FS_ATTRIB_USER_SETTABLE : -1;
  }
  /** Set file attributes
   *
   * \param[in] bits bit-wise or of selected attributes: FS_ATTRIB_READ_ONLY,
   *            FS_ATTRIB_HIDDEN, FS_ATTRIB_SYSTEM, FS_ATTRIB_ARCHIVE.
   *
   * \note attrib() will fail for set read-only if the file is open for write.
   * \return true for success or false for failure.
   */
  bool attrib(uint8_t bits);
```
You get the current fie attributes with file.attrib(), clear or set any to be changed then call file.attrib(bits) with the desired attributes.

For example to hide a file
```
 int bits = file.attrib();   // should test for -1 here.
 file.attrib(bits | FS_ATTRIB_HIDDEN);  // should test return
```
This will show up in the release version in a month or two when all the mods in the beta are tested for a while.

Miguel0101, 2021-12-17:
Very nice, I will test this feature later.

Miguel0101, 2021-12-29:
The attrib function is functional, so I will close this issue. Thanks @greiman.

closed 2021-12-29 by Miguel0101