greiman/SdFat #344: How to open specific file like /01.mp3?

podaen, 2021-11-10:
I know the files are indexed and I need to know the index to open that file. But I don't know the index of that file... How can I know?

greiman, 2021-11-10:
FAT and exFAT directories are not indexed so you need to maintain your own index.

If you want to use open by index, open the file by path then get the index for future use using dirIndex().

```
  File dir;
  File file;
  // open the directory containing the file.
  if (!dir.open(dirPath)) { // handle error}
  // Use name of file in the directory - replace O_RDWR with desired options
  if (!file.open(&dir, fileName, O_RDWR)) { // handle error}
  // use uint16_t for FAT, uint32_t for exFAT
  uint32_t index = file.dirIndex();
  file.close();
  // ...
  // open later by index
  if (!file.open(&dir, index, O_RDWR)) { // hnadle error}

```

podaen, 2021-11-10:
It seems difficult in the beginning to do. I was thinking on a using way and forgot I was the bot. It's clear to me now and I get how it works. Thanks for the help.

podaen, 2021-11-10:
I have a FAT32... So I will start with that.

[image]

greiman, 2021-11-10:
[Here](https://github.com/greiman/SdFat/blob/master/examples/examplesV1/LongFileName/LongFileName.ino) is an old example that still seems to work.

Copy [these files](https://github.com/greiman/SdFat/tree/master/examples/examplesV1/LongFileName/testFiles) into the root directory of the SD. Then run the example.

podaen, 2021-11-10:
On track, I just had to use out of an other example of you

```
#define SPI_SPEED SD_SCK_MHZ(4)

if (!sd.begin(SD_CS_PIN, SPI_SPEED)) {
```
I quid not understanding what the index means. When I add this

> Serial.print("index: ");
> Serial.print(dirIndex[n]);

to the plot I always recieve zero's. Even with other folders. If I open a file by index those numbers needs too be different I think. I don't know.

greiman, 2021-11-10:
If you put the print after this code you will always get zero since n is incremented.

```
      dirIndex[n] = file.dirIndex();

      // Print the file number and name.
      Serial.print(n++);
```

podaen, 2021-11-10:
It's not like an ID are someting, I will never know what m is coming next!  If I do openNext... Will I loop through all fills in that directory?

greiman, 2021-11-10:
openNext() starts at the current position of the directory, opens the next file, and leaves the position after the directory entry for the file it opens.

podaen, 2021-11-10:
If I understand it good. It will go to the next directory when all next files in that directory are done. How do I get all files in that directory?

greiman, 2021-11-10:
If you rewind the directory file so it is positioned at zero and call openNext until it returns false, you will have opened all files in the directory.

podaen, 2021-11-10:
Ok, I know what to do how to make my playlist. Thanks for helping me out.

closed 2021-11-10 by podaen