greiman/SdFat #345: List files an folders on SD card

podaen, 2021-11-11:
Is there any shetch I could use to list all files from folders and subfolders on one SD card?

marcboon, 2021-11-11:
Try [examples/DirectoryFunctions.ino](https://github.com/greiman/SdFat/blob/master/examples/DirectoryFunctions/DirectoryFunctions.ino)

podaen, 2021-11-11:
Feeling bad to say... What I am asking can not be done whitout leveling and I cann't find it in this example. 
It is a good example for directory tasks and is familiare whit what I am asking, but I haven't find where I am looking for.

marcboon, 2021-11-11:
> What I am asking can not be done whitout leveling

What do you mean by that?

Just as a hint: 
```
sd.ls("/", LS_R);
``` 
lists all files and directories on the card, and does that recursively, meaning it will open subdirectories until no more are found.

greiman, 2021-11-11:
> Is there any shetch I could use to list all files from folders and subfolders on one SD card?

Just use  sd.ls("/", LS_R); form the DirectoryFunctions.ino example.

If your programming skill are good enough you can write a recursive function using openNext to traverse the directory tree for the SD.
Here is the code for ls:
```
bool ExFatFile::ls(print_t* pr, uint8_t flags, uint8_t indent) {
  ExFatFile file;
  if (!isDir()) {
    DBG_FAIL_MACRO;
    goto fail;
  }
  rewind();
  while (file.openNext(this, O_RDONLY)) {
    // indent for dir level
    if (!file.isHidden() || (flags & LS_A)) {
      for (uint8_t i = 0; i < indent; i++) {
        pr->write(' ');
      }
      if (flags & LS_DATE) {
        file.printModifyDateTime(pr);
        pr->write(' ');
      }
      if (flags & LS_SIZE) {
        file.printFileSize(pr);
        pr->write(' ');
      }
      file.printName(pr);
      if (file.isDir()) {
        pr->write('/');
      }
      pr->write('\r');
      pr->write('\n');
      if ((flags & LS_R) && file.isDir()) {
        file.ls(pr, flags, indent + 2);
      }
    }
    file.close();
  }
  if (getError()) {
    DBG_FAIL_MACRO;
    goto fail;
  }
  return true;

 fail:
  return false;
}
```

podaen, 2021-11-11:
My experiences are not bad for smaller projects. You are right... I currently learning how to use classes. 

When I place that on top

```
Print Cpr;//is not allowed
uint8_t Cflags;
uint8_t Cindent;
ExFatFile Cls;
```

Is my question what do I need to fill in?

And I placed this in the setup.

`Cls(Cpr,Cflags,Cindent);`

podaen, 2021-11-11:
Ok. Diden't see that marcboon has answered. Sorry about that. I got it running now.

```
#include <SPI.h>
#include "SdFat.h"
#include "FreeStack.h"
// **************** SD CARD
const int SD_CS_PIN = SS;
#define SPI_SPEED SD_SCK_MHZ(4)
// SD_FAT_TYPE = 0 for SdFat/File as defined in SdFatConfig.h,
// 1 for FAT16/FAT32, 2 for exFAT, 3 for FAT16/FAT32 and exFAT.
#define SD_FAT_TYPE 0

#if SD_FAT_TYPE == 0
SdFat sd;
File dir;
File file;
#elif SD_FAT_TYPE == 1
SdFat32 sd;
File32 dir;
File32 file;
#elif SD_FAT_TYPE == 2
SdExFat sd;
ExFile dir;
ExFile file;
#elif SD_FAT_TYPE == 3
SdFs sd;
FsFile dir;
FsFile file;
#else  // SD_FAT_TYPE
#error invalid SD_FAT_TYPE
#endif  // SD_FAT_TYPE
void setup() {
	Serial.begin(9600);
	while (!Serial);
	Serial.println("SD Card directory sorting...");
	if (!sd.begin(SD_CS_PIN, SPI_SPEED)) {
		Serial.println("SD card error!");
		while (true);
	}
	sd.ls("/", LS_R);
}
void loop() {
  
}
```

> > What I am asking can not be done whitout leveling
> 
> What do you mean by that?

Here it is called indent i think.

marcboon, 2021-11-11:
Glad you got it running :)
You can see in Bill's source code for the ls function where the indentation happens. Two spaces are added for each level deeper, and then the same function is called recursively (it calls itself):
```
     if ((flags & LS_R) && file.isDir()) {
        file.ls(pr, flags, indent + 2);
      }
```

podaen, 2021-11-11:
Indentation! I could have know... The word indented I know.

> (it calls itself)

I hope soo. :-)

Ps: English is not my natural language.

closed 2021-11-11 by podaen