greiman/SdFat #346: Get false on openNext

podaen, 2021-11-12:
I don't know what I am doing wrong... I still getting false on open next. It's not that there is nothing in the directory. I am looking for quite a while to it and tried a few things but I am out of idees.

```
#include <SPI.h>
#include "SdFat.h"
#include "FreeStack.h"
#include <stdio.h>
// **************** SD CARD
const int SD_CS_PIN = SS;
#define SPI_SPEED SD_SCK_MHZ(4)
// SD_FAT_TYPE = 0 for SdFat/File as defined in SdFatConfig.h,
// 1 for FAT16/FAT32, 2 for exFAT, 3 for FAT16/FAT32 and exFAT.
#define SD_FAT_TYPE 1

#if SD_FAT_TYPE == 0
SdFat sd;
File dir;
File file;
#elif SD_FAT_TYPE == 1
SdFat32 sd;
File32 dir;
File32 file;
#elif SD_FAT_TYPE == 2
SdExFat sd;
ExFile dir;
ExFile file;
#elif SD_FAT_TYPE == 3
SdFs sd;
FsFile dir;
FsFile file;
#else  // SD_FAT_TYPE
#error invalid SD_FAT_TYPE
#endif  // SD_FAT_TYPE
// Store error strings in flash to save RAM.
#define error(s) sd.errorHalt(&Serial, F(s))
```

```
void setup() {
	Serial.begin(115200);
	if (!sd.begin(SD_CS_PIN, SPI_SPEED)) {
		Serial.println("SD card error!");
		sd.initErrorHalt(&Serial);
		while (true);
	}
	CreatePlaylist("/02/" , "test.csv");//("/21/" , "001.mp3")
}
```

```
bool CreatePlaylist(char* dirPath, char* fileName) {
	// Open root directory
	//if (!dir.open(dirPath)) {
	//	sd.errorHalt(&Serial, F("dir.open failed"));
	//}
	if (sd.exists(dirPath)) {
		//Serial.println(String(PathWithFilename(dirPath, fileName)) );
		if (sd.exists(CombineStrings(dirPath, fileName))) {
			if (!file.open(CombineStrings(dirPath, fileName), FILE_WRITE)) {
				error("create file in folder failed");
			}
			else
			{
				Serial.println("write song to playlist");
				SongsInFolder(dirPath);
				file.close();
			}
		}
		else {
			if (!file.open(CombineStrings(dirPath, fileName), FILE_WRITE | O_CREAT)) {
				error("create file in folder failed");
			}
			else
			{
				Serial.println("write song to new playlist");
				SongsInFolder(dirPath);
				file.close();
			}
		}
	}
	else
	{
		Serial.println("path does not exist");
	}
}
```

```
void SongsInFolder(char* dirPath) {
	char fileName[255];
	char CdirPath[255];
	int No = 1;
	//// Open root directory
	if (!dir.open(dirPath)) {
		/*sd.errorHalt(&Serial, F("dir.open failed"));*/
		error("dir.open failed");
	}
	dir.getName(CdirPath, sizeof(CdirPath));
	Serial.println(CdirPath);
	dir.rewind();
	Serial.println("Start open next");
	while (file.openNext(&dir, O_RDONLY)) {
		Serial.println("opening next");
		if (!file.isHidden()) {
			if (!file.isDir()) {
				file.getName(fileName, sizeof(fileName));
				WriteSongToCsvPlaylist(No, fileName);
				No++;
			}
		}
	}
	Serial.println("create song list done");
}
```

podaen, 2021-11-12:
Ok, solved it myself. hehe, your welcome! :-)

The cause of the issue was that I diden't close my file before I was calling an other folder. Feel free to make some changes.

greiman, 2021-11-12:
open() fails if the file is already open.  This is to catch errors since failure to close a file can cause file system corruption if the file was opened for write.

podaen, 2021-11-13:
I'm getting the point.

closed 2021-11-13 by podaen