greiman/SdFat #347: Csv Go To Line

podaen, 2021-11-13:
I having problems to go to the next line without overwriting the previous line. Can someone help me out with this?

```
void WriteSongToPlaylist(int No, char* dirPath, char* filename, char* Dbfilename) {
	char* WriteString;
	char cNoStr[16];
	if (!file.open(CombineStrings(dirPath, Dbfilename), O_WRITE)) {
		error("opening file in folder failed");
	}
	else
	{
			itoa(No, cNoStr, 10);//convert number in to string
			if (No == 0) {//First line
				WriteString = CombineStrings(cNoStr, ";");
				WriteString = CombineStrings(WriteString, filename);
				WriteString = CombineStrings(WriteString, "\r\n");
			}
			else {//all other lines
				WriteString = "\r";
				for (int i = 1; i < No; i++) {
					WriteString = CombineStringsSpecialChar(WriteString, "\r");
				}//end go to line
				WriteString = CombineStrings(WriteString, cNoStr);
				WriteString = CombineStrings(WriteString, ";");
				WriteString = CombineStrings(WriteString, filename);
			}
			file.print(F(WriteString));
		file.close();
	}
}
```

podaen, 2021-11-13:
An other question... how is this working... opening next?

If I add a file to the directory, will it automaticly added the file AFTER all files that where allready in the directory?  Or is there someting like sorting the files?

podaen, 2021-11-13:
One solution is by not closing the csv file until all lines are written.

greiman, 2021-11-14:
You should read the documentation in libraries/SdFat/doc/.  Unzip html.zip and open SdFat.html in a browser.

You will find options for opening a file, like at end.  Here is documentation for oflag in open(path, oflag)

[image]

Also look at more examples.

podaen, 2021-11-14:
intresting... Thanks.

podaen, 2021-11-17:
Can I open/creat multiple files at the same time?

greiman, 2021-11-17:
> Can I open/creat multiple files at the same time?

Yes

closed 2021-11-17 by podaen