greiman/SdFat #350: sd.stop

podaen, 2021-12-03:
I am using the SdFat32 begin before I use a library that also calls that. How can I stop it when I am done?

greiman, 2021-12-03:
You should close all open files then call sd.end(). Currently the only function of sd.end() is to call end() in the SPI library.

marcboon, 2021-12-03:
Bill Greiman schreef op 03-12-2021 13:18:

> You should close all open files then call sd.end(). Currently the only function of sd.end() is to call end() in the SPI library.

Maybe it's an idea to first close all open files in sd.end(), and then
call SPI.end()? For the lazy people, and to make it more robust?

greiman, 2021-12-03:
`Maybe it's an idea to first close all open files in sd.end(),`

I can't do that since the status of files is in the file class.

podaen, 2021-12-04:
> call sd.end()

Don't have that member on SdFat32 sd;

podaen, 2021-12-05:
Oh... I see now like this. 	SPI.end(); 

But my task task still won't start.

podaen, 2021-12-05:
Now it does by only using this without my code

```
	if (!sd.begin(SD_CS_PIN, SPI_SPEED)) {
		Serial.println("SD card error!");
		sd.initErrorHalt(&Serial);
		while (true);
	}

        SPI.end();
```

podaen, 2021-12-05:
It is something within my code why the task won't start. I closed all files and still there is a lot memorie in used.

greiman, 2021-12-05:
> It is something within my code why the task won't start. I closed all files and still there is a lot memorie in used.

Closing files does not release memory.  You need to learn the basics of how Arduino works. I don't have time to teach you how programs on Arduino work.  I am closing this.

sd.end() is in V 2.1.2:
```
#include "SdFat.h"
SdFat32 sd;

void setup() {
  Serial.begin(9600);
  if (!sd.begin(SS)) sd.initErrorHalt();
  Serial.print("SdFat32: ");
  Serial.println(SD_FAT_VERSION_STR);
  sd.end();
}
void loop() {}
```
output:
`SdFat32: 2.1.2`

closed 2021-12-05 by greiman