greiman/SdFat #351: De-initialize/Reset SD Card to save power

savejeff, 2021-12-10:
Hi, 

I'm working with an ESP32 based board that has already an inbuild SD Card Slot. 
Looking into sleep current I noticed that the SD Card takes more power after sd.end than before sd.begin.

If I completely remove the sd card code but have the SD Card plugged in I get ~1mA (might be lower, limited by decimal points of measurement)
when I init the sd card and de-init it again, i get around 13-20 mA. 
here is the interesting thing: in deep sleep, when i take out the sd card and put it in again it goes down to ~1mA. 

Thus I conclude the sd card is after sd.end still held in some initialized state. 
Is there a way besides cutting the power to the sd card with a transistor (which is not possible as the board is a given) to reset/de-init the sd card. 

I read another thread from earlier that just stopping communicating with the sd would put it into a sleep state, I'm not seeing this. I tried putting all sd spi spins in input/input_pullup/input_pulldown and it didn't really help

savejeff, 2021-12-10:
I Found a nice solution here:

https://forum.arduino.cc/t/resolved-how-can-i-reboot-an-sd-card/349700

sending 0xFF over SPI often enough will cause the SD Card to reset and send 0xFF back. 
It can be used to reset a card that is stuck or not responding after a crash but also to "unmount" the sd card.

this simple code 
```

pinMode(PIN_SPI_CS_SD, OUTPUT);
	digitalWrite(PIN_SPI_CS_SD, LOW);

	for(int i = 0; i < 10; i++)
	{
		byte v = SPI.transfer(0xFF);
		if(v == 0xFF)
			break;
	}

	digitalWrite(PIN_SPI_CS_SD, HIGH);
```

reduces the power consumption to sub 1mA after sd.end. 

Would it be possible to add this to the sd.end code or add a sd.reset() function to also use it to reset a sd card if sd.begin fails?

greiman, 2021-12-10:
Some cards remain active after chips select is high until they see a few clocks.  

Try sending a 0XFF byte something like this with SD chip select high.

```
SPI.begin();
SPI.beginTransaction(SPISettings(400000, MSBFIRST, SPI_MODE0));
SPI.transfer(0XFF);
SPI.endTransaction();
SPI.end();
```
I will experiment with the sd.end() call to see if adding a transfer of n extra byte improves sleep.

You can't really put a card back in the initial power on state.  You can only reinitialize a card.  For example once a card is in SPI mode you must cycle power to use 4-bit SDIO mode.

savejeff, 2021-12-10:
looks like i was a minutes faster with the solution. 

Would look to have a sd.reset function added. 

Might be helpful for some people when they have a system crash. 

i can test this with a couple of sd cards. i have around 7 different types of sd cards here

closed 2021-12-10 by savejeff