greiman/SdFat #352: USB MSC Driver

Miguel0101, 2021-12-22:
```
// Edit SdFatConfig.h and enable generic block devices.
// #define USE_BLOCK_DEVICE_INTERFACE 1
#include "UsbMscDriver.h"

USB usb;
BulkOnly bulk(&usb);
UsbMscDriver usbKey(&bulk);

FsVolume key;
FsFile file;

void createUSBTask(void *pv_Parameters);

void setup()
{
  Serial.begin(115200);
  while (!Serial)
    ;

  Serial.println(F("Type any character to start"));

  while (!Serial.available())
    ;

  if (!usbKey.initUSB(&usb))
  {
    Serial.println("initUSB failed");
    while (1)
      ;
  }

  else
  {
    Serial.println("initUSB ok");
    xTaskCreate(createUSBTask, "USB_TASK", 4096, NULL, configMAX_PRIORITIES, NULL);
  }

  // Must set USE_BLOCK_DEVICE_INTERFACE non-zero in SdFatConfig.h
  if (!key.begin(&usbKey))
  {
    Serial.println("key.begin failed");
    while (1)
      ;
  }

  if (!file.open("usbtest.txt", FILE_WRITE))
  {
    Serial.println("file.open failed");
    while (1)
      ;
  }

  file.println("test line");
  file.close();
  key.ls(LS_DATE | LS_SIZE);
}

void loop()
{
}

void createUSBTask(void *pv_Parameters)
{
  while (1)
  {
    usb.Task();
    vTaskDelay(5);
  }
}
```
When I try to start `key.begin`, it returns an error

greiman, 2021-12-22:
You are on you own with USB keys and FreeRTOS..

Miguel0101, 2021-12-23:
> You are on you own with USB keys and FreeRTOS..

Um, I didn't get it. I can't run `usb.Task` in an xTask?

greiman, 2021-12-23:
> Hmm, I didn't get it. I can't run usb.Task in an xTask?

I have no idea what you can do with your setup.

I will fix bugs in SdFat but can't debug your app.

closed 2021-12-23 by greiman

Miguel0101, 2021-12-23:
I just need help using an external storage device (Mass Storage).

Miguel0101, 2021-12-23:
`UsbMscDriver.h`:
```
#ifndef UsbMscDriver_h
#define UsbMscDriver_h
#include "SdFat.h"

// If the next include fails, install the USB_HOST_SHIELD library.
#include <masstorage.h>
//------------------------------------------------------------------------------
/** Print debug messages if USB_FAT_DBG_MODE is nonzero */
#define USB_FAT_DBG_MODE 1
//------------------------------------------------------------------------------
/** Maximum time to initialize the USB bus */
#define TIMEOUT_MILLIS 400000L
class UsbMscDriver : public FsBlockDeviceInterface
{
public:
  UsbMscDriver(BulkOnly *bulk) : m_lun(0), m_bulk(bulk) {}

  virtual bool initUSB(USB *usb);

  virtual bool isBusy()
  {
    return false;
  }

  virtual bool readSector(uint32_t sector, uint8_t *dst)
  {
    uint8_t rc = m_bulk->Read(m_lun, sector, 512, 1, dst);
    return 0 == rc;
  }

  virtual uint32_t sectorCount()
  {
    return m_bulk->GetCapacity(m_lun);
  }

  virtual bool syncDevice()
  {
    return true;
  }

  virtual bool writeSector(uint32_t sector, const uint8_t *src)
  {
    return 0 == m_bulk->Write(m_lun, sector, 512, 1, src);
  }

  virtual bool readSectors(uint32_t sector, uint8_t *dst, size_t ns)
  {
    return 0 == m_bulk->Read(m_lun, sector, 512, ns, dst);
  }

  virtual bool writeSectors(uint32_t sector, const uint8_t *src, size_t ns)
  {
    return 0 == m_bulk->Write(m_lun, sector, 512, ns, src);
  }

private:
  uint8_t m_lun;
  BulkOnly *m_bulk;
};

#endif // UsbMscDriver_h
```
`UsbMscDriver.cpp`:
```
#include "UsbMscDriver.h"

/** Simple USB init function.
 * \param[in] usb USB host object to be initialized.
 * \return true for success or false for failure.
 */
bool UsbMscDriver::initUSB(USB *usb)
{
  if (usb->Init() == 0)
  {
    return true;
  }

  else
  {
    return false;
  }
}
```

Miguel0101, 2021-12-23:
It initializes the module successfully, but returns an error when I try to initialize the key (`key.begin`).

greiman, 2021-12-23:
I can't help you.  I didn't write the underling USB library and some USB keys fail.  Most of my keys work.  You will need to debug the USB library yourself.

Here is a test I did on a Mega ADK:
```
Type any character to start
Host initialized, ms: 1003
USB powered, ms: 1003
USB state: 20, ms: 1009
USB state: 40, ms: 1203
USB state: 50, ms: 1253
USB state: 51, ms: 1253
USB state: 90, ms: 7188
2021-01-01 00:00         11 usbtest.txt
2021-06-10 07:58          0 RtcMin/
2021-04-23 09:43          0 FsRtc/
2021-04-25 11:20          0 RtcBeta/
2021-04-23 08:43          0 RTClib/
2021-06-13 09:49          0 RtcLite/
2021-06-13 10:39    1535594 Gregorian year.pdf
2021-06-13 10:37    1535594 cc-paper.pdf
2021-06-13 11:27          0 RtcLiteTest/
2021-06-13 04:34          0 sketch_jun12b/
2021-06-12 11:29          0 RtcMinTest/
2021-06-12 04:34          0 sketch_jun12a/
2021-06-11 05:00          0 sketch_jun11a/
2021-06-13 19:45          0 mods/

```

Miguel0101, 2021-12-24:
Can you send the code you used 🙂?

Miguel0101, 2021-12-24:
It worked!!!