greiman/SdFat #354: Using with ENABLE_ARDUINO_FEATURES=0

PetteriAimonen, 2022-01-04:
I would like to use SdFat on a platform that does not currently have Arduino core support.
The `SdFatConfig.h` has `#define ENABLE_ARDUINO_FEATURES 1`.

Would it be possible to add `#ifdef` for this like the other defines, or is there a reason to force it always on?

greiman, 2022-01-04:
It is forced on since there is not support for other platforms in the released version of SdFat.  

I have used a modified version on mbed, ChibiOS/RT and FreeRTOS but don't plan to release it anytime soon.

PetteriAimonen, 2022-01-04:
Ok, I guess I'll just use a local modified copy then also. Seems to work pretty well with small hacks. For future reference I'll include my changes below. They are quite hacky for now but compile and work:

<pre>
diff --git a/src/SdFatConfig.h b/src/SdFatConfig.h
index 141ffd9..6942738 100644
--- a/src/SdFatConfig.h
+++ b/src/SdFatConfig.h
@@ -46,11 +46,11 @@
 // if they are in a #ifndef/#endif block below.
 //------------------------------------------------------------------------------
 /** For Debug - must be one */
-#define ENABLE_ARDUINO_FEATURES 1
+#define ENABLE_ARDUINO_FEATURES 0
 /** For Debug - must be one */
-#define ENABLE_ARDUINO_SERIAL 1
+#define ENABLE_ARDUINO_SERIAL 0
 /** For Debug - must be one */
-#define ENABLE_ARDUINO_STRING 1
+#define ENABLE_ARDUINO_STRING 0
 //------------------------------------------------------------------------------
 #if ENABLE_ARDUINO_FEATURES
 #include "Arduino.h"
@@ -58,6 +58,9 @@
 // Only defined if a Particle device.
 #include "application.h"
 #endif  // PLATFORM_ID
+#else
+#define SS 0
+extern "C" unsigned long millis();
 #endif  // ENABLE_ARDUINO_FEATURES
 //------------------------------------------------------------------------------
 /**
diff --git a/src/common/PrintBasic.h b/src/common/PrintBasic.h
index f174371..bb9efa5 100644
--- a/src/common/PrintBasic.h
+++ b/src/common/PrintBasic.h
@@ -40,6 +40,7 @@ class __FlashStringHelper;
 #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
 #else  // defined(__AVR__)
 #define F(str) (str)
+struct __FlashStringHelper {};
 #endif  // defined(__AVR__)
 #endif  // F
</pre>