greiman/SdFat #355: writing on AVR arduino leonardo takes too much time

roque-canales, 2022-01-17:
Hello,
Using V2, I call this log every 100ms in my program:

  file.print(F("FLT,"));
    file.print(millis());    file.print(F(",0,"));
    file.print(galt, 2);        file.print(F(","));
    file.print(clms, 2);        file.print(F(","));
    file.print(digitalRead(CCF));        file.print(F(","));
        #ifdef TRIO
        file.print(digitalRead(CCF2));        file.print(F(","));
            file.print(digitalRead(CCF3));        file.print(F(","));
    #endif
    file.print(analogRead(TROIVTROI) * 0.0049);        file.print(F(","));
    file.print(lipov * 0.0641);         file.print(F(","));
    file.print(analogRead(GATE) * 0.084);         file.print(F(","));
    file.print(analogRead(CHKSOL) * 0.0641/*0.0075*/);         file.print(F(","));
    file.print(analogRead(CHKMOS) * 0.084/*0.0075*/);         file.print(F(","));
    file.print(URGENCE);     file.print(F(","));
    file.print(state);     file.println();
    file.sync();

If I don't put file.sync(); nothing is writted on the board......

How I can optimize writing time?

Can SDFAT write automatically buffer to sd each 256 byte ?
How to setup it?

Also I found, ENDL_CALLS_FLUSH, tryed it set to 1 or 0, and in both case, if I don't put sync function, nothing is writed on the sd.......

greiman, 2022-01-17:
Calling sync() is like closing the file in terms of I/O to the SD.  Don't call sync() unless there is a danger of a crash. sync() does not just write data to the SD, it updates the directory entry and the file allocation table.

Call close() at the end of a run.  This will properly flush buffers and update the file system structures.

> Can SDFAT write automatically buffer to sd each 256 byte ?

SdFat automatically writes data to the SD when a 512 byte sector is in the cache.

Calling sync causes about 2048 bytes of I/O each time it is called. The current data sector is read, updated and written.  The sector with the file directory entry is read, updated and written.  You are probably increasing I/O by a factor of ten by calling sync().

roque-canales, 2022-01-17:
Hello,

Thank you for this explanations.

So in our app there is risk of crash due that user can cut power at anytime.

So in this case you tell me to use sync()?

If we use close() this can corrupt sd?

Envoyé de mon iPhone

Le 17 janv. 2022 à 13:43, Bill Greiman ***@***.***> a écrit :

Calling sync() is like closing the file in terms of I/O to the SD. Don't call sync() unless there is a danger of a crash. sync() does not just write data to the SD, it updates the directory entry and the file allocation table.

Call close() at the end of a run. This will properly flush buffers and update the file system structures.

Can SDFAT write automatically buffer to sd each 256 byte ?

SdFat automatically writes data to the SD when a 512 byte sector is in the cache.

Calling sync causes about 2048 bytes of I/O each time it is called. The current data sector is read, updated and written. The sector with the file directory entry is read, updated and written. You are probably increasing I/O by a factor of ten by calling sync().

greiman, 2022-01-18:
Closing the file will not corrupt the SD, it is the way most apps work.  It's rare for an app to use sync() the way you do.

Some control power like a PC, the user presses a power button but the app controls the actual power switch.  Some pre-allocate and erase a file so at most one sector is lost.