greiman/SdFat #365: Updated SDFat

bvernham, 2022-03-02:
SDFat was updated through Arduino file manager and now I am getting an warning I never had before

Arduino\libraries\SdFat\src/SdFat.h:452:2: warning: #warning File not defined because __has_include(FS.h) [-Wcpp]
 #warning File not defined because __has_include(FS.h)

I never saw this warning before.

How do solve?

Thanks

Michael-Monaldi, 2022-03-02:
@bvernham Are you uploading sketches through Arduino IDE, Arduino 2.0, or platformio?

What board ?

I just went thru a horrible experience with this last week

bvernham, 2022-03-02:
Due

greiman, 2022-03-03:
You must be using ESP32.  ESP32 may define the File class so SdFat no longer defines File in this case.

You can replace File with FsFile.  It is best to also replace SdFat with SdFs like this:
```
  SdFs sd;
  FsFile file;
```

It may not be possible to use SdFat on ESP32 in the future.