greiman/SdFat #367: 'SysCall' has not been declared

landret, 2022-04-07:
Hi,
I am using the latest version of the Arduino IDE v1 and I recently installed and tested the library. However I met the following compile error with example QuickStart.ino for an Arduino Uno board:

C:\Users\christophe\Documents\Arduino\libraries\arduino_279335\examples\QuickStart\QuickStart.ino:` In function 'void setup()':
QuickStart:71:5: error: 'SysCall' has not been declared
     SysCall::yield();
     ^~~~~~~
C:\Users\christophe\Documents\Arduino\libraries\arduino_279335\examples\QuickStart\QuickStart.ino: In function 'void loop()':
QuickStart:109:5: error: 'SysCall' has not been declared
     SysCall::yield();
     ^~~~~~~
QuickStart:185:5: error: 'SysCall' has not been declared
     SysCall::yield();
     ^~~~~~~
'SysCall' has not been declared

This is solved by removing every occurence of SysCall:: from this example.

landret, 2022-04-08:
It looks like no. I used the Arduino library manager to update the library. The library version is now 2.1.2 but it looks like the Quickstart example was not updated. The same for the Bench example which also contains Syscall.

greiman, 2022-04-08:
I have seen glitches in the library manager in the past.  The mod to remove SysCall done on October 31, 2021 in 2.1.1 and you are the first to report a problem.  Many other examples were changed.

landret, 2022-04-09:
Indeed, there was 3 copies but I can't remember how they got here. Removing them solved the problem. Thanks and sorry about that.

closed 2022-04-12 by landret