greiman/SdFat #370: Teensy 4.1 - Vanilla SdFat or Arduino Framework ?

savejeff, 2022-04-21:
Hi!,

since yesterday im playing around with the Teensy 4.1. 
I would like to test out the SDIO functionality. I noticed that the arduino core for teensy uses the SdFat library but in version 2.0.5-beta.1 and it seems to be slightly modfied.

in generall i would prefere to use the latest sdfat library as it also reduces problems when switching between architectures. i almost only use SdFat everywhere now. 

so my question: Should i use the Vanuilla up to date Library or the one from the Arduino Core? Did they change some thing to get everything out of the hardware or is it just not keept up to date with the lastest SdFat version?

savejeff, 2022-04-21:
Thanks for the quick response. most often there is no clear answer. just pro and cons. 

With what you said ill go with the vanilla SdFat Library and I'll take the overall configuration from the teensy core.

closed 2022-04-21 by savejeff

savejeff, 2022-04-23:
just a small followup. im getting crazy fast write speeds with pre allocation and 512byte write blocks. 
with a fast Class 10 A1 Card I'm getting over 2.5Mbytes/s with only 10ms write durations at max. 
Very nice!

[image]

PaulStoffregen, 2022-05-07:
@savejeff - Teensyduino 1.57 will have SdFat updated to version 2.1.2.  It is a modified copy because we're using SdFat for both SD cards and USB drives, and also because Teensyduino has a FS abstraction layer which needs minor changes.  Beta testing for 1.57 just started, so if you want to give it a try, get it here.

https://forum.pjrc.com/threads/70196-Teensyduino-1-57-Beta-1

If you want to use unmodified SdFat, you certainly can.  Any library you install in {Documents}/Arduino/libraries will override the modified copy Teensyduino installed.  Just make sure you use "FsFile" rather than "File" so you're directly accessing SdFat's API.