greiman/SdFat #371: Files are all non writable

landret, 2022-04-23:
Hi,
I am using version 2.1.2 with an Arduino Pro Mini, and an sd card formatted fat32.

When going through a folder using openNext, I noticed that all files are marked as non-writable (isWritable() returns false for all), while they are definitelly writable.

This entails that no files are found when using openNext in write mode.

This behavior can be seen with example OpenNext. If O_RDONLY is replaced by O_WRONLY, no files are found.

greiman, 2022-04-23:
If a file is opened with O_RDONLY, isWritable() will return false.

If you use openNext with O_WRONLY and a directory is encountered, openNext will fail.  openNext with O_WRONLY is only useful in a subdirectory with no directories.

openNext was first added to the Arduino SD.h wrapper for SdFat in 2010.  I added openNext to SdFat and made it function like the Arduino wrapper version.

You can use openNext together with seekSet to navigate to writable files: 

```
  while (true) {
    uint32_t pos = dir.curPosition();
    if (!file.openNext(&dir, O_RDONLY)) {
      break;
    }
    file.printName(&Serial);
    // Check file attributes on SD.
    if (file.isDir() || file.isReadOnly()) {
      file.close();
      Serial.println(" Skip");
      continue;
    }
    dir.seekSet(pos);
    file.close();
    if (!file.openNext(&dir, O_WRONLY)) {
        Serial.println(" Open Error");
        break;
    }
    Serial.println(" Opened for write");
    file.close();
  }
  if (dir.getError()) {
    Serial.println("openNext failed");
  } else {
    Serial.println("Done!");
  }
```
Here is output for a test case with directories and a read only file.
```
Type any character to start
System Volume Information Skip
SubDir Skip
ReadOnly.txt Skip
WriteOk.txt Opened for write
Done!
```

greiman, 2022-04-23:
You might want to skip files that return true for the attributes isHidden() and isSystem.

landret, 2022-04-24:
Hi Bill, thanks for this great answer. openNext() + seekSet() is what I needed. 
It could be interesting to add this to the "OpenNext" example, to show user how to edit/remove files using openNext().

closed 2022-04-24 by landret