greiman/SdFat #372: Need for speed

ClaudeJGreengrass, 2022-04-26:
I've had good success with ExFatLogger on a Mega2560 logging 6 x floats from an Adafruit LSM6DS33 6-DOF IMU at 250Hz.  But now I've moved on to an ARM M4.  More memory, faster processor.....  and I was hoping to move the sampling rate to 400Hz or higher.  The actual board is an [Adafruit PyGamer](https://www.adafruit.com/product/4242)

The IMU can sample at 5KHz but moving the sample rate in ExFatLogger more than 250Hz gives a "Rate too fast" error

I would appreciate any suggestions to upping the data rate.  thx in advance

ClaudeJGreengrass, 2022-04-27:
I can loan you a board if that would help though it sounds as if it is not very hopeful.

What board(s) would you suggest for best throughput.  Currently I am recording 6 floats which I know is not optimal.  Future expansion plans suggest a 3x to 5x increase in data volume and I would like to be able to hit 1000hz sampling rate if  at all possible.  I know that is a big ask for a microcontroller but it is where the big boys (researcher) are starting to play.  

Of course fidelity of the sample rate needs to be maintained as small variations in frequency of the tremor (Essential Tremors) being measured is important.  Even sampling at the current 250Hz, a loss of 1ms in 4 seconds of data can badly skew the results.

uzi18, 2022-04-27:
https://www.avrfreaks.net/forum/samd21-slow-spi-slave

greiman, 2022-04-27:
The SAMD SPI driver is often slower than UNO or Mega2560.  I have given up on SAMD so I don't want a board.

To log at high rates you need something like this [Teensy program](https://github.com/greiman/SdFat/blob/master/examples/TeensySdioLogger/TeensySdioLogger.ino) that logs an ADC as text at 25,000 samples per second.

I have logged at over 1,000,000 samples per second with DMA ADC in binary on a Teensy.

ClaudeJGreengrass, 2022-04-27:
Thanks for the heads up on where to do for fast data logging.

closed 2022-04-27 by ClaudeJGreengrass