greiman/SdFat #374: Can't delete files and folder

FernandoGarcia, 2022-04-27:
Hi!

Sorry by bother you here with a question but I can't find an answer.

Could you tell me please why I can't delete the folder "veiculos" and the files inside using this code?

```cpp
#include <SdFat.h>

SdFat SD;
File file;
File root;

String folder = "veiculos";

#define LOG(...) Serial.print(__VA_ARGS__)
#define LOGLN(...) Serial.println(__VA_ARGS__)

void setup() {
  SD.begin (SPI_QUARTER_SPEED, 53);
  Serial.begin(115200);

  char currentFile[15];

  if (!root.open(folder.c_str()))
  {
    LOG(F("Can't open the folder "));
    LOGLN(folder);
  }
  else
  {
    LOG(F("The folder "));
    LOG(folder);
    LOGLN(F(" is open"));
  }

  while (file.openNext(&root, O_RDONLY))
  {
    file.getName(currentFile, 15);
    file.close();

    if (file.isOpen())
    {
      LOGLN(F("The file is open"));
    }
    else
    {
      LOGLN(F("The file is closed"));
    }

    if (file.remove(currentFile))
    {
      LOG(currentFile);
      LOGLN(F(" was deleted"));
    }
    else
    {
      LOG(currentFile);
      LOGLN(F(" wasn't deleted"));
    }
  }

  SD.chdir("/");
  
  SD.ls();

  if (SD.rmdir(folder))
  {
    LOG(F("The folder "));
    LOG(folder);
    LOGLN(F(" was deleted."));
  }
  else
  {
    LOG(F("The folder "));
    LOG(folder);
    LOGLN(F(" wasn't deleted"));
  }

}

void loop() {}
```

The output for this code is:

```
The folder veiculos is open
The file is closed
175628d2.txt wasn't deleted

The file is closed
20fb6e1b.txt wasn't deleted

The file is closed
ff28f40a.txt wasn't deleted

The file is closed
fe29f30b.txt wasn't deleted

The file is closed
fd2af20c.txt wasn't deleted

The file is closed
fc2bf10d.txt wasn't deleted

The file is closed
fb2cf00e.txt wasn't deleted

The file is closed
70dfe427.txt wasn't deleted

The file is closed
e77e23d2.txt wasn't deleted

The file is closed
7038d102.txt wasn't deleted

The file is closed
411f7527.txt wasn't deleted

imagens/
veiculos/
usuarios/
envios/
The folder veiculos wasn't deleted
```
Thanks in advance!
Best regards.

FernandoGarcia, 2022-04-27:
The files used were created in this way:

```cpp
  if(!SD.exists(folder))
  {
    if (!SD.mkdir(folder))
    {
      LOG(F("Can't create the folder "));
      LOGLN(folder);
      return;
    }
    else
    {
      LOG(F("The folder "));
      LOG(folder);
      LOGLN(F(" was created"));
    }
  }
```

```cpp
    file = SD.open(fileName, FILE_WRITE);

    if (!file)
    {
      LOG(F("Can't create "));
      LOGLN(fileName);
      return;
    }

    String data = "{\"nome\":\"";
    data += Json["adi"][i]["nome"].as<String>();
    data += "\",\"tag\":";
    data += Json["adi"][i]["tag"].as<String>();
    data += "}";

    file.println(data);

    file.close();
```

FernandoGarcia, 2022-04-27:
Looks like it doesn't works too.

I tried:

```cpp
      String filePath = folder + "/" + String(currentFile);
      LOG(F("Full path: "));
      LOGLN(filePath);

      if(file.remove(filePath.c_str()))
      {
        LOG(currentFile);
        LOGLN(F(" was deleted"));
      }
      else
      {
        LOG(currentFile);
        LOGLN(F(" wasn't deleted"));
      }
```
The output is:
```ini
The folder veiculos exists

The folder veiculos is open

Full path: veiculos/175628d2.txt

175628d2.txt wasn't deleted

Full path: veiculos/20fb6e1b.txt

20fb6e1b.txt wasn't deleted
```

greiman, 2022-04-27:
Here is proof that remove and rmdir work for "veiculos/175628d2.txt"

I created the folder and file on an SD then ran this program:

```
#include "SdFat.h"
SdFat sd;
File file;
void setup() {
  Serial.begin(9600);
  while(!Serial) {}
  Serial.println("Type any to continue");
  while (!Serial.available()) {}
  if (!sd.begin(SS)) sd.initErrorHalt(&Serial);
  Serial.println("ls before:");
  sd.ls("/", LS_R | LS_A);
  if (!sd.remove("veiculos/175628d2.txt")) sd.errorHalt("remove failed");
  if (!sd.rmdir("veiculos")) sd.errorHalt("rmdir failed");
  Serial.println("ls after:");
  sd.ls("/", LS_R | LS_A);
  Serial.println("Done");
}
void loop() {}
```
The output:
```
Type any to continue
ls before:
System Volume Information/
  IndexerVolumeGuid
  WPSettings.dat
veiculos/
  175628d2.txt
ls after:
System Volume Information/
  IndexerVolumeGuid
  WPSettings.dat
Done
```

FernandoGarcia, 2022-04-27:
`sd.remove()` worked with your code but `rmdir()` doesn't. :(

There's about 100 files in "veiculos" folder.

[image]

Thanks for your patience!

FernandoGarcia, 2022-04-27:
Can you try my code using the function `openNext()` and `getName()` with this files please?
[veiculos.zip](https://github.com/greiman/SdFat/files/8574492/veiculos.zip)

greiman, 2022-04-27:
You must delete all files from a folder before calling rmdir.

I am closing this since I have no time to debug you program.

closed 2022-04-27 by greiman

FernandoGarcia, 2022-04-27:
Thank you for try help me.

Maybe you should change this comment in your source code.

bool rmdir() ---> Recursively delete a directory and all contained files.

greiman, 2022-04-27:
The comment is for rmRfStar(), not ridir!

/** Recursively delete a directory and all contained files.
   *
   * This is like the Unix/Linux 'rm -rf *' if called with the root directory
   * hence the name.
   *
   * Warning - This will remove all contents of the directory including
   * subdirectories.  The directory will then be removed if it is not root.
   * The read-only attribute for files will be ignored.
   *
   * \note This function should not be used to delete the 8.3 version of
   * a directory that has a long name.  See remove() and rmdir().
   *
   * \return true for success or false for failure.
   */
  bool rmRfStar();

FernandoGarcia, 2022-04-27:
Ok, my fault sorry.

FernandoGarcia, 2022-04-28:
Hi!

I was trying to understand why the `remove()` function was working in other part of my program and I have noticed that I was using `SD.remove()` instead `file.remove()` as in your proof code.

I have no idea why `remove()` works with `SdFat` instance (?) but doesn't works with `File`. I think do you know.

So the final solution is:

```cpp
      String filePath = folder + "/" + String(currentFile);
      LOG(F("Full path: "));
      LOGLN(filePath);

      if(SD.remove(filePath.c_str()))
      {
        LOG(currentFile);
        LOGLN(F(" was deleted"));
      }
      else
      {
        LOG(currentFile);
        LOGLN(F(" wasn't deleted"));
      }
```

Thanks for your help.

Best regards.

FernandoGarcia, 2022-04-28:
Final code is:

```cpp
#include <SdFat.h>

SdFat SD;
File file;
File root;

String folder = "veiculos";

#define LOG(...) Serial.print(__VA_ARGS__)
#define LOGLN(...) Serial.println(__VA_ARGS__)

void setup() {
  SD.begin (SPI_QUARTER_SPEED, 53);
  Serial.begin(115200);

  char currentFile[15];

  if (!root.open(folder.c_str()))
  {
    LOG(F("Can't open the folder "));
    LOGLN(folder);
  }
  else
  {
    LOG(F("The folder "));
    LOG(folder);
    LOGLN(F(" is open"));
  }

  while (file.openNext(&root, O_RDONLY))
  {
    file.getName(currentFile, 15);
    file.close();

    String filePath = folder + "/" + String(currentFile);

    if (SD.remove(filePath.c_str()))
    {
      LOG(currentFile);
      LOGLN(F(" was deleted"));
    }
    else
    {
      LOG(currentFile);
      LOGLN(F(" wasn't deleted"));
    }
  }

  if (SD.rmdir(folder))
  {
    LOG(F("The folder "));
    LOG(folder);
    LOGLN(F(" was deleted."));
  }
  else
  {
    LOG(F("The folder "));
    LOG(folder);
    LOGLN(F(" wasn't deleted"));
  }

}

void loop() {}
```

greiman, 2022-04-29:
> I have no idea why remove() works with SdFat instance (?) but doesn't works with File. I think do you know.

If you open a file O_RDONLY file.remove() will fail.

greiman, 2022-04-29:
The file must be open for write to delete with file.remove()

Here is an example that removes all files in a directory.
The file names are 0.TXT through some max number like 123.TXT
```
/*
 * remove all files in dir.
 */
void deleteFiles(FatFile* dir) {
  char name[32];
  file_t file;

  // open and delete files
  for (uint16_t n = 0; ; n++){
     sprintf(name, "%u.TXT", n);

     // open start time
     uint32_t t0 = millis();

     // assume done if open fails
     if (!file.open(dir, name, O_WRONLY)) return;

     // open end time and remove start time
     uint32_t t1 = millis();
     if (!file.remove()) error("file.remove failed");

     // remove end time
     uint32_t t2 = millis();

     Serial.print(F("RM "));
     Serial.print(n);
     Serial.write(' ');

     // open time
     Serial.print(t1 - t0);
     Serial.write(' ');

     // remove time
     Serial.println(t2 - t1);
  }
}
```

FernandoGarcia, 2022-04-29:
Ok, thanks!