greiman/SdFat #379: Can't init SD card - Excepet when the SD is physically removed

RodEnry, 2022-05-26:
Hi, 
I'm facing an issue with an ESP32 (I tried with core arduino espressif32@3.3.2, @3.4.0 and 3.5.0) and the SPI SdFat library version 2.1.2.

I can't init the SD card properly.
The code slows down a lot when calling this function: `(!SD.begin(SD_CONFIG))`
where the SD_CONFIG is: 
`#define SPI_CLOCK SD_SCK_MHZ(25)
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)`

If I remove and re-apply the SD card runtime it starts againg working properly (basically the initialization works again).
Unfortunately the SD card is sealed inside a waterproof enclosure and I can't do this trick when needed.

Full function:
`void initializeSD() {
    SPI.begin(SD_SCK, SD_MISO, SD_MOSI, SD_CS_PIN);
    pinMode(SD_CS_PIN, OUTPUT);
    digitalWrite(SD_CS_PIN, HIGH);
    
    if (!SD.begin(SD_CONFIG)) {
        #ifdef DEBUG
        Serial.println("Can't Initialize SD CARD ");
        #endif
        SD_present = false;
    } else {
        #ifdef DEBUG
        Serial.println("SD card ready ");
        #endif
        SD_present = true; 
    }
}`

Many thanks
Enry

greiman, 2022-05-26:
Have you tried other cards?   I have seen a few cards that require a power cycle to accept a second software reset command.

RodEnry, 2022-05-26:
Hello @greiman,
thanks for the fast reply.

The SPI BUS is dedicated.

I'm actually lucky enough to have a mosfet connected to the SD card and other utilities. I can power down everything and do a power cycle as you said. It's very strange because it happens quite randomly.

greiman, 2022-05-26:
The SPI bus on the Due that had problems was "dedicated" but killed the SD upon program reload and sometimes reboot.

Put a logic analyzer on the SPI bus and see if the ESP system causes spurious signals.

SD cards can lockup if they see what looks like an SDIO init.  This can happen if CS is high and there is noise on MOSI.  power must be cycled to go to SPI mode.  

Also have you tried several SD cards.