greiman/SdFat #380: Use in-built SD slot on Teensy 4.1?

SvarunSoda, 2022-05-29:
I'm writing a simple program that writes to a SD card formatted in the "exFAT" format, on a Teensy 4.1 using the standard Arduino IDE compiler with the Teensyduino plugin.

I want to use the in-built SD card slot on the Teensy, however I'm really confused on how to properly implement it;

I initially stumbled upon a forum in which it was said to use the `BUILTIN_SDCARD` enum when calling the `begin()` function, however that is apparently only supported on the standard SD library, not on the SdFat library...

I then read that I apparently needed to use the keyword `SS` when declaring my ChipSelect pin for the `begin()` function and to use `#define ENABLE_SDIO_CLASS 1`, however that causes `begin()` to fail.

I then read that I supposedly need to delcare my SD object as a `SdFatSdio` type, however my Arduino compiler states that `SdFatSdio does not name a type`.

At this point I'm a bit clueless on what to do, and any help would be appreciated!

Thank you for reading my post.

greiman, 2022-05-29:
You can just use the standard Teensy library.  It is just a wrapper for a recent version of SdFat.

If you want to use the latest version of SdFat see examples in this library.  Most work for Teensy 4.1 with no modification. HAS_SDIO_CLASS is defined by default for Teensy 3.x and 4.x.

Try [TeensySdioLogger](https://github.com/greiman/SdFat/blob/master/examples/TeensySdioLogger/TeensySdioLogger.ino), it demonstrates high performance logging.

SdFat does not use BUILTIN_SDCARD as a fake SPI chip select to indicate the SDIO port on the Teensy board.  SdFat has many options beyond chip select.

Most examples define SD_CONFIG something like this:

```
// Try to select the best SD card configuration.
#if HAS_SDIO_CLASS
#define SD_CONFIG SdioConfig(FIFO_SDIO)
#elif  ENABLE_DEDICATED_SPI
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SPI_CLOCK)
#else  // HAS_SDIO_CLASS
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SPI_CLOCK)
#endif  // HAS_SDIO_CLASS
```
Then initialize the SD like this:
```
  // Initialize SD.
  if (!sd.begin(SD_CONFIG)) {
    sd.initErrorHalt(&Serial);
  }
```

SvarunSoda, 2022-05-29:
Thank you for your reply;

Unfortunately I cannot use the standard SD library, as I'm writing to an SD in the "exFAT" format (bigger than 32 GB), so I have to use your "exFAT" library. Even if I try to compile with both the standard SD library and yours, I get an error.

I will definitely try out your other suggestions, I will let you know how it goes!

greiman, 2022-05-30:
The Teensy 4.1 standard SD library supports exFat.  Post a question on the [Teensy Forum](https://forum.pjrc.com/) for help.