greiman/SdFat #382: Can't format with SdFormatter.ino and XSTD Adafruit SPI FLASH

simon88, 2022-06-02:
Hi,

I'm working with this module [https://learn.adafruit.com/adafruit-spi-flash-sd-card/overview](url) I've just tried to format it with the SdFormatter example but after thaht my card is down, when I try a SdInfo I got 

> 14:29:07.481 -> SdFat version: 2.1.0
14:29:07.481 -> 
14:29:07.481 -> Disabling SPI device on pin 6
14:29:07.481 -> 
14:29:07.481 -> Assuming the SD chip select pin is: 5
14:29:07.481 -> Edit SD_CS_PIN to change the SD chip select pin.
14:29:07.481 -> 
14:29:07.481 -> type any character to start
14:29:08.655 -> init time: 2 ms
14:29:08.655 -> 
14:29:08.655 -> Card type: SD2
14:29:08.655 -> 
14:29:08.655 -> Manufacturer ID: 0XB
14:29:08.655 -> OEM ID: XT
14:29:08.655 -> Product: XTSDA
14:29:08.655 -> Version: 1.1
14:29:08.655 -> Serial number: 0X1479AC02
14:29:08.655 -> Manufacturing date: 7/2020
14:29:08.655 -> 
14:29:08.655 -> cardSize: 515.90 MB (MB = 1,000,000 bytes)
14:29:08.655 -> flashEraseSize: 128 blocks
14:29:08.655 -> eraseSingleBlock: true
14:29:08.655 -> 
14:29:08.655 -> OCR: 0X80FF8000
14:29:08.655 -> 
14:29:08.655 -> SD Partition Table
14:29:08.655 -> part,boot,bgnCHS[3],type,endCHS[3],start,length
14:29:08.690 -> 1,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,4294967295,4294967295
14:29:08.690 -> 2,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,4294967295,4294967295
14:29:08.690 -> 3,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,4294967295,4294967295
14:29:08.690 -> 4,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,0XFF,4294967295,4294967295
14:29:08.690 -> 
14:29:08.690 -> MBR not valid, assuming Super Floppy format.
14:29:08.690 -> 
14:29:08.690 -> volumeBegin failed. Is the card formatted?
14:29:08.690 -> 
14:29:08.690 -> type any character to start

And when I format I got

> 14:30:20.466 -> Cards up to 2 GiB (GiB = 2^30 bytes) will be formated FAT16.
14:30:20.501 -> Cards larger than 2 GiB and up to 32 GiB will be formatted
14:30:20.501 -> FAT32. Cards larger than 32 GiB will be formatted exFAT.
14:30:20.501 -> 
14:30:20.501 -> Warning, all data on the card will be erased.
14:30:20.501 -> Enter 'Y' to continue: Y
14:30:21.413 -> 
14:30:21.413 -> Card size: 0.52 GB (GB = 1E9 bytes)
14:30:21.413 -> Card size: 0.48 GiB (GiB = 2^30 bytes)
14:30:21.413 -> Card will be formated FAT16
14:30:21.413 -> 
14:30:21.413 -> Options are:
14:30:21.413 -> E - erase the card and skip formatting.
14:30:21.413 -> F - erase and then format the card. (recommended)
14:30:21.413 -> Q - quick format the card without erase.
14:30:21.413 -> 
14:30:21.413 -> Enter option: F
14:30:24.354 -> 
14:30:24.354 -> Erasing
14:30:24.528 -> ....
14:30:25.085 -> All data set to 0xff
14:30:25.085 -> Erase done
14:30:25.085 -> Writing FAT ..................................
14:30:25.433 -> Format Done
14:30:25.433 -> Run the SdInfo example for format details.

And after this new format

> 14:31:06.698 -> Edit SD_CS_PIN to change the SD chip select pin.
14:31:06.698 -> 
14:31:06.698 -> type any character to start
14:31:09.247 -> init time: 14 ms
14:31:09.247 -> 
14:31:09.247 -> Card type: SD2
14:31:09.247 -> 
14:31:09.247 -> Manufacturer ID: 0XB
14:31:09.247 -> OEM ID: XT
14:31:09.247 -> Product: XTSDA
14:31:09.247 -> Version: 1.1
14:31:09.247 -> Serial number: 0X1479AC02
14:31:09.247 -> Manufacturing date: 7/2020
14:31:09.247 -> 
14:31:09.247 -> cardSize: 515.90 MB (MB = 1,000,000 bytes)
14:31:09.247 -> flashEraseSize: 128 blocks
14:31:09.247 -> eraseSingleBlock: true
14:31:09.247 -> 
14:31:09.247 -> OCR: 0X80FF8000
14:31:09.247 -> 
14:31:09.247 -> SD Partition Table
14:31:09.247 -> part,boot,bgnCHS[3],type,endCHS[3],start,length
14:31:09.247 -> 1,0X0,0X3,0X2D,0X0,0X6,0X9,0XF9,0XE7,233,1007383
14:31:09.247 -> 2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
14:31:09.247 -> 3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
14:31:09.247 -> 4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
14:31:09.560 -> 
14:31:09.560 -> volumeBegin failed. Is the card formatted?
14:31:09.560 -> SD errorCode: SD_CARD_ERROR_READ_TIMEOUT = 0X1D
14:31:09.560 -> SD errorData = 0XFF
14:31:09.560 -> 
14:31:09.560 -> type any character to start

Any ideas?

simon88, 2022-06-03:
@greiman Yes it' not a tru SD but it's like an SD card so I don't know if it change something or not. But I've used your library to read/wrtite everythong works perfectly, it's only the formatter has bricked my card

greiman, 2022-06-03:
I suspect you should never format the card since it is pre-formatted.  Maybe this card does not support formatting with the SD standard format.

The flash and buffer layout for cards is specified by the SD Association and  I use the SD standard layout for cards.

Contact Adafruit - I only have internal info for cards that comply with the SD Association Standard.

simon88, 2022-06-03:
Ok nice thanks for your help

simon88, 2022-06-07:
Just for you information , I've contacted the manufacturer of the device, and here is thaht answer 

> Our SD NAND default is format FAT16. Our SD NAND is the same as the micron SD card. So you can use the SD card format method, like Windows OS hard disk format. thanks!

greiman, 2022-06-07:
By default the Arduino SD.h wrapper for my 2009 version of SdFat uses a clock rate of 4 MHz for SD.begin(csPin).

In 2009 there was only the AVR 328 Arduino and I allowed 8 Mhz and 4 MHz with 4 MHz the default since the first SD modules used resistors for 5V to 3.3V level shifters and would not run at 8 MHz.

Arduino has added a more modern SD.begin(clock, csPin).

It is amazing to still see most of the 2009 SdFat library in D:\arduino\arduino-1.8.19\libraries\SD\src\utility. with the old copyright:
```
/* Arduino Sd2Card Library
   Copyright (C) 2009 by William Greiman

   This file is part of the Arduino Sd2Card Library
```