greiman/SdFat #383: Data speed transfer painfully slow in Arduino NANO 32 BLE

groblesm, 2022-06-03:
Hello,

I am testing the writting/reading speed on an Arduino NANO 32 BLE. My SD card is a SanDisk Extreme 32 GB U3 A1 micro SDHC.
The testing programme is bench.ino, from the library examples. I only change the type of file system to match the FS in the SD card, I also changed the SPI clock speed from 50 MHz to 8 MHz though no changes in data transfer are observed. The SD card is freshly formated.
I do not know whether the problem is on my side or if these results are normal. In my opinion, the latency is absurdly high since in ExFatLogger.ino which I also have tested, the recommended value for LOG_INTERVAL_USEC ranges from 250 to 4000 and I have to set it to 12000.

Any help would be appreciated.

These are the results of bench.ino for ExFAT:

`Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: -6008
Type is exFAT
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SE32G
Version: 8.0
Serial number: 0X65DBEE78
Manufacturing date: 1/2016

FILE_SIZE_MB = 5
BUF_SIZE = 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
46.00,12571,11165,11179
46.03,17548,11164,11179

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
45.83,11272,11216,11230
45.84,11271,11215,11230

Done
Type any character to start
`
And these are the results for FAT32 which are very similar:

`Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: -6064
Type is FAT32
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SE32G
Version: 8.0
Serial number: 0X65DBEE78
Manufacturing date: 1/2016

FILE_SIZE_MB = 5
BUF_SIZE = 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
46.08,22668,11161,11180
46.10,17106,11165,11179

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
45.85,11276,11217,11233
45.83,11277,11218,11234

Done
Type any character to start
`

greiman, 2022-06-03:
The NANO 32 BLE uses a mbed SPI driver and is incredibly slow.

It may be possible to get some improvement by using array transfer.  Edit SdFatConfig.h at about line 130 and set USE_SPI_ARRAY_TRANSFER to one like this:
```
/**
 * If USE_SPI_ARRAY_TRANSFER is non-zero and the standard SPI library is
 * use, the array transfer function, transfer(buf, size), will be used.
 * This option will allocate up to a 512 byte temporary buffer for send.
 * This may be faster for some boards.  Do not use this with AVR boards.
 */
#ifndef USE_SPI_ARRAY_TRANSFER
#define USE_SPI_ARRAY_TRANSFER 1  // <<---- was zero
#endif  // USE_SPI_ARRAY_TRANSFER
```

SAMD is often slower than AVR so don't expect too much.

greiman, 2022-06-03:
I managed to get a NANO 32 BLE Sense to run with array transfer but it is flaky and fails occasionally.  I am using shared SPI mode at 8 MHz.

```
Here is bench with a Samsung microSD.
Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: -5984
Type is FAT32
Card size: 32.01 GB (GB = 1E9 bytes)

Manufacturer ID: 0X1B
OEM ID: SM
Product: 00000
Revision: 1.0
Serial number: 0XE30F5501
Manufacturing date: 10/2015

FILE_SIZE_MB = 5
BUF_SIZE = 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
447.39,1208,1135,1144
447.23,1211,1134,1145

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
451.63,1170,1125,1134
452.12,1187,1125,1134
```

groblesm, 2022-06-06:
Same results as yours.
Thank you again.

```
Use a freshly formatted SD for best performance.

Type any character to start
FreeStack: -6064
Type is FAT32
Card size: 31.91 GB (GB = 1E9 bytes)

Manufacturer ID: 0X3
OEM ID: SD
Product: SE32G
Version: 8.0
Serial number: 0X65DBEE78
Manufacturing date: 1/2016

FILE_SIZE_MB = 5
BUF_SIZE = 512 bytes
Starting write test, please wait.

write speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
447.07,8567,1134,1144
447.11,1896,1134,1144

Starting read test, please wait.

read speed and latency
speed,max,min,avg
KB/Sec,usec,usec,usec
464.77,1154,1092,1100
464.77,1139,1093,1099

Done
```

groblesm, 2022-06-06:
See work around by @greiman in comment #2

closed 2022-06-06 by groblesm