greiman/SdFat #385: Cannot compile using Arduino IDE any version

BlackPaaanda, 2022-06-15:
Hi, greiman,

Thanks a lot for the library. However, when I tried to compile the code using Arduino IDE any version using the library, many error messages appeared and the compilation process halted. The major error code is attached below. Could you please help me with this?

`C:\Users\Panda\AppData\Local\Temp\cc5G3NEC.ltrans0.ltrans.o: In function `write':
c:\users\panda\documents\arduino\libraries\sdfat\common/arduinofiles.h:54: undefined reference to `FatFile::write(void const*, unsigned int)'
C:\Users\Panda\AppData\Local\Temp\cc5G3NEC.ltrans0.ltrans.o: In function `SdCardFactory::newCard(SdSpiConfig)':
C:\Users\Panda\Documents\Arduino\libraries\SdFat/SdCard/SdCard.h:58: undefined reference to `SharedSpiCard::begin(SdSpiConfig)'
C:\Users\Panda\AppData\Local\Temp\cc5G3NEC.ltrans0.ltrans.o: In function `begin':
C:\Users\Panda\Documents\Arduino\libraries\SdFat/FatLib/FatVolume.h:47: undefined reference to `FatPartition::init(SharedSpiCard*, unsigned char)'
C:\Users\Panda\AppData\Local\Temp\cc5G3NEC.ltrans0.ltrans.o: In function `chdir':
C:\Users\Panda\Documents\Arduino\libraries\SdFat/FatLib/FatVolume.h:66: undefined reference to `FatFile::close()'
C:\Users\Panda\Documents\Arduino\libraries\SdFat/FatLib/FatVolume.h:67: undefined reference to `FatFile::openRoot(FatVolume*)'
C:\Users\Panda\AppData\Local\Temp\cc5G3NEC.ltrans0.ltrans.o: In function `begin':
C:\Users\Panda\Documents\Arduino\libraries\SdFat/FatLib/FatVolume.h:54: undefined reference to `FatVolume::m_cwv'
C:\Users\Panda\Documents\Arduino\libraries\SdFat/FatLib/FatVolume.h:54: undefined reference to `FatVolume::m_cwv'
C:\Users\Panda\AppData\Local\Temp\cc5G3NEC.ltrans0.ltrans.o: In function `read':
C:\Users\Panda\Documents\Arduino\libraries\SdFat/FatLib/FatFile.h:719: undefined reference to `FatFile::read(void*, unsigned int)'
C:\Users\Panda\AppData\Local\Temp\cc5G3NEC.ltrans0.ltrans.o: In function `loop':
C:\Users\Panda\Desktop\sdfat_test/sdfat_test.ino:169: undefined reference to `FatFile::close()'
C:\Users\Panda\Desktop\sdfat_test/sdfat_test.ino:147: undefined reference to `FatFile::open(char const*, unsigned char)'
collect2.exe: error: ld returned 1 exit status
`

greiman, 2022-06-15:
Something is wrong with your installation of SdFat.   The path should have a src folder like this

C:\Users\Bill\Documents\Arduino\libraries\SdFat\src\FsLib

Your path is missing src:

C:\Users\Panda\Documents\Arduino\libraries\SdFat/FatLib

Try deleting the SdFat folder and use the IDE library manager to install SdFat.

BlackPaaanda, 2022-06-16:
Thanks a lot! I deleted the folder and re-installed the SdFat using IDE library manager, and the problem has been resolved. Thanks again for your help!

closed 2022-06-16 by BlackPaaanda