greiman/SdFat #387: SDFat is not working with Adafruit Feather M0

Dvl-s, 2022-07-06:
SDFat version=2.2.0
Board : Adafruit Feather M0

is SDFat 2.2.0 supports Adafruit Feather M0? 
      - Yes then Sdcard begin is failling.
      - No then can you suggest what to do to work SDFat with Adafruit Feather M0?

greiman, 2022-07-06:
I need more Info since it works on my [Feather M0 Adalogger](https://www.adafruit.com/product/2796).

Here is the SdInfo example:

> SdFat version: 2.2.0
> 
> Assuming the SD is the only SPI device.
> Edit DISABLE_CS_PIN to disable an SPI device.
> 
> Assuming the SD chip select pin is: 4
> Edit SD_CS_PIN to change the SD chip select pin.
> 
> type any character to start
> init time: 2 ms
> 
> Card type: SDHC
> sdSpecVer: 3.00
> HighSpeedMode: true
> 
> Manufacturer ID: 0X1B
> OEM ID: SM
> Product: 00000
> Revision: 1.0
> Serial number: 0XE30F5501
> Manufacturing date: 10/2015
> 
> cardSize: 32010.93 MB (MB = 1,000,000 bytes)
> flashEraseSize: 128 blocks
> eraseSingleBlock: true
> dataAfterErase: ones
> 
> OCR: 0XC0FF8000
> 
> SD Partition Table
> part,boot,bgnCHS[3],type,endCHS[3],start,length
> 1,0X0,0X82,0X3,0X0,0XC,0XFE,0XFF,0XFF,8192,62513152
> 2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 
> Scanning FAT, please wait.
> 
> Volume is FAT32
> sectorsPerCluster: 64
> fatStartSector:    9316
> dataStartSector:   24576
> clusterCount:      976512
> freeClusterCount:  976354

One possibility is that Adafruit still allows 24 MHz SCK.  Arduino now limits SCK to the recommended 12 MHz.  I have optimized the SD card driver and 24 MHz may be be marginal for some SAMD21 chips/boards.

The bench example runs at 24 MHz on my Feather M0 Adalogger.  Here are results at 12 and 24 MHz.  There is not much difference since the SAMD21 SERCOM SPI has huge gaps between bytes at 24 MHz, Uno is faster at 8MHz.

> bench at 12 MHz:
> 
> Use a freshly formatted SD for best performance.
> 
> Type any character to start
> FreeStack: 26260
> Type is FAT32
> Card size: 32.01 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X1B
> OEM ID: SM
> Product: 00000
> Revision: 1.0
> Serial number: 0XE30F5501
> Manufacturing date: 10/2015
> 
> FILE_SIZE_MB = 5
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 516.26,1007,984,989
> 516.21,1008,984,989
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 485.63,1058,1050,1051
> 485.63,1058,1050,1051
----------------------------------------------------------
> bench at 24 MHz.
>
> Use a freshly formatted SD for best performance.
> 
> Type any character to start
> FreeStack: 26260
> Type is FAT32
> Card size: 32.01 GB (GB = 1E9 bytes)
> 
> Manufacturer ID: 0X1B
> OEM ID: SM
> Product: 00000
> Revision: 1.0
> Serial number: 0XE30F5501
> Manufacturing date: 10/2015
> 
> FILE_SIZE_MB = 5
> BUF_SIZE = 512 bytes
> Starting write test, please wait.
> 
> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 609.46,856,832,837
> 609.38,855,832,837
> 
> Starting read test, please wait.
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 567.02,902,896,900
> 567.02,903,895,900

-----------------------------------------------

Dvl-s, 2022-07-07:
SOLVED !

There is problem in SPI clock speed, it is 24MHz by default in Feather M0.
so I Just change SD.begin(SD_CS_PIN) to SD.begin(SD_CS_PIN, SD_SCK_MHZ(12)) and code is working good

Thank to @greiman

greiman, 2022-07-07:
Adafruit should fix their SAMD driver. It has at least three bugs.  Guess I will post issues again.  

I know Limor Fried AKA Lady Ada and she is extremely capable but very busy.   She has built a business with over 100 employees and is deeply into everything.

Three bugs I know for SAMD21 SPI  are:

The max speed is checked for 24MHz even though she recently stated "SPI is a high speed, 3-wire protocol that can be clocked at up to 12MHz on the ATSAMD21".

The speed is incorrectly calculated.  It should be a speed less than or equal to the requested speed.  Instead the speed is higher unless the requested speed exactly divides 24 MHz.  so if you request 13 MHz you will get 24 MHz , the next SAMD divisor.

If you request a speed less than 95 KHz you will get some random speed up to 24MHz.

adamgarbo, 2022-07-28:
Thanks for the insights, Bill (@greiman). 

I've used the Adafruit M0 Adalogger for a number of years now and have tended to err on the side of caution by using `SD_SCK_MHZ(4)`. Interesting to know the reasons why the higher speeds don't (currently) work.

Cheers,
Adam

eugeniop, 2022-12-24:
FWIW, I had to change configuration to 12Mhz on an Adafruit Feather M0 using an external SD card board (a dual SD / e-Ink board).

jjdeprisco, 2023-02-28:
I am seeing an SDFat issue with an Emotibit sketch that appears to be related to this issue.

Here's what I am seeing on serial monitor as of today:

I2C data pin: 11
I2C clk pin: 13
hibernate pin: 6
chip sel pin: 19
Firmware version: 1.5.4
firmware_variant: charlieplex_heartbeatOnSleeveJD
vregEnablePinLogic: Active HIGH(V3+)
EmotiBit not ready. Please check if Battery and SD-Card are present on the EmotiBit.
**Setup failed: SD-Card not detected**

I am also using a Feather M0 WiFi. Can someone tell me exactly where you made the SD_SCK_MHZ change noted above? Which section in which library file? Having trouble locating it. I want to see if this solves my issue.

greiman, 2023-02-28:
Use the [Adafruit fork of SdFat](https://github.com/adafruit/SdFat) and contact Adafruit for help.

greiman, 2023-02-28:
Let me know what they say.  Adafruit software is in a poor state.  They now have about 1,600 repositories and not enough staff.
[image]

jjdeprisco, 2023-03-01:
> FWIW, I had to change configuration to 12Mhz on an Adafruit Feather M0 using an external SD card board (a dual SD / e-Ink board).

Where specifically did you make this change?

adamgarbo, 2023-03-01:
@jjdeprisco There are plenty of [examples](https://github.com/greiman/SdFat/blob/628effa1c277e6dc1fef461b4b755ad300735248/examples/ExFatLogger/ExFatLogger.ino#L68-L78) of how to configure the microSD card in Bill's library.

If you're looking for something simple, I've used the following line of code to initialize the microSD card on the Feather M0 Adalogger for a number of years:

```
if (!sd.begin(PIN_MICROSD_CS, SD_SCK_MHZ(4)))
{
    Serial.println("Warning - microSD failed to initialize.");
}
```

I've found 4 MHz to be reliable. Haven't bothered trying 12 MHz. 

Cheers,
Adam

jjdeprisco, 2023-03-01:
Thanks. I got to the bottom of things. I ended up going almost back to the very beginning, manually removing all of the libraries that were installed for my project , and then reinstalled again. It appears one can not trust the Arduino library manager to fall back on versions for a few things, so I ended up with a few multiple versions in place. Once that was resolved, I managed to get the basic sketch working.

johnmsole, 2024-05-11:
Thanks everyone. I battled with adafruit feather M0 adalogger for EXACTLY the same isue.  added SD_SCK_MHZ(12) and problem resolved.   Wahooo