greiman/SdFat #389: open fails after 4094 files

marcoberna78, 2022-07-13:
Hi all,
I'm trying to write a lot of files to an Sd Card through the Teensy 4.1 SDIO interface and after a while the open function returns an error. I tried different Sd Cards (San Disk and Samsung) and different file size (64KB, 512 KB): in all cases , starting from an empty sd card, after 4094 files written correctly, the open function started to report an error indefinitely, also after a power cycle of the Teensy. Then I removed the card from the teensy and inserted it in the PC (Win): all the 4094 files were correct. Then, using the PC, I added a file manually. At that point the teensy found correctly 4095 files and started again to working correctly for another 4095 files and so on... So the first time teensy is able to write 4094 files and then 4095 files, each time adding a dummy file using the PC.
I'm using the SD library (SD ver 2.0.0 and SdFat ver. 2.1.0). All the SD Cards are formatted with a 512KB allocation unit size. Below the code to test the strange behavior:

#include <Arduino.h>
#include <SD.h>

#define SD_BUFFER_SIZE 32*512

SdFs mysd;
FsFile myfile;
FsFile root;
static char fileName[80];
char sdBuffer[SD_BUFFER_SIZE];
bool ret;
unsigned int file_counter;

void setup()
{
  uint16_t tries=0;
  unsigned int rootFileCount=0;
  while(!Serial) continue;
  pinMode(40,OUTPUT);
  pinMode(13, OUTPUT);
  memset(sdBuffer,0,SD_BUFFER_SIZE);
  file_counter=0;
  delay(1000);
  
  while(!mysd.begin(SdioConfig(FIFO_SDIO)) && tries<10){ 
    Serial.print(tries++); Serial.print(" "); 
    delay(1000); 
  } 
  if(tries>=10) 
  {
    Serial.println("No SD storage"); 
    while(1);
  }

  if (!root.open("/")) {
    Serial.println("error opening root");
  }
  while (myfile.openNext(&root, O_RDONLY)) {
    if (!myfile.isHidden()) {
      rootFileCount++;
    }
    myfile.close();
  }
  delay(5000);
  Serial.printf("Number of files in root folder: %d\n",rootFileCount);
  Serial.println("End of Setup");
  return;
}

void loop()
{
  DateTimeFields tm;
  breakTime(rtc_get(), tm);
	sprintf(fileName, "F_%04d%02d%02d_%02d%02d%02d_%05d.bin", (short int)(tm.year)+1970, tm.mon, tm.mday, tm.hour, tm.min, tm.sec, ++file_counter); 
  Serial.printf("%s\n",fileName);
  ret=myfile.open(fileName, O_RDWR | O_CREAT);
  if(!ret){
    Serial.printf("%s - open failed\n",fileName);
  }
  else{
    int i;
    size_t bytesWritten;
    digitalWrite(13, HIGH);
    for(i=0;i<32;i++){
      bytesWritten=myfile.write(sdBuffer,SD_BUFFER_SIZE);
      if(bytesWritten!=SD_BUFFER_SIZE){
        Serial.printf("file write return %d\n",bytesWritten);
        if(bytesWritten==0){
          Serial.printf("Error code = %d\n",myfile.getWriteError());
        }
      }
    }
    myfile.flush();
    ret=myfile.close();
    digitalWrite(13, LOW);
    if(!ret){
      Serial.printf("%s - close failed\n",fileName);
    }   
  } 
 
}

greiman, 2022-07-13:
Looks like each file requires 128 byte in the directory space  so 512KB would allow 4096 files.  Probably a bug extending the root directory which starts with one cluster.

I will need to do tests since I don't see an obvious problem.  Your info will be very helpful.

greiman, 2022-07-13:
Looks like a bug that was fixed in SdFat 2.1.2 or 2.2.0.

I ran you test code with SdFat 2.20  on an 64GB card formatted with 512KB clusters and stopped the test code here:

> F_20220713_102848_08320.bin
> F_20220713_102848_08321.bin
> F_20220713_102848_08322.bin
> F_20220713_102848_08323.bin
> F_20220713_102848_08324.bin
> F_20220713_102848_08325.bin
> F_20220713_102848_08326.bin

I change this line since the year/month was wrong on my version of Teensy.

`sprintf(fileName, "F_%04d%02d%02d_%02d%02d%02d_%05d.bin", (short int)(tm.year) + 1900, tm.mon + 1, tm.mday, tm.hour, tm.min, tm.sec, ++file_counter);`