greiman/SdFat #390: Writing files larger than 4GB

nperentos, 2022-07-14:
I see that SDFat supports files > 4GB. However, I can't get this feature to work. The observed behavior is that the file size gets wrapped (e.g. if I ask for a 5GB file, I end up getting a 1GB file). Any insight would be appreciated.
Here is part of my code for setting up a file and writing to it.

// setting up a file

  uint64_t LogFileSz_MB = 5000;
  log_file_size = LogFileSz_MB * 1024 * 1024;
  FsFile dataFile;

 // Open or create file - truncate existing file.
  if (!dataFile.open(fname, O_RDWR | O_CREAT | O_TRUNC)) {
    Serial.println(F("open failed\n"));
    errBlink(REDpin);
    return;
  }
  // File must be pre-allocated to avoid huge delays searching for free clusters.
  if (!dataFile.preAllocate(log_file_size)) {
     Serial.println(F("preAllocate failed\n"));
     dataFile.close();               
     errBlink(REDpin);
     return;
  }
  
  Serial.println(F("preAllocate success\n"));
  
  if (dataFile.isContiguous()) {
    Serial.println(("dataFile is contiguous...\n"));
  }

  uint32_t first_sector, last_sector;
  if (!dataFile.contiguousRange(&first_sector, &last_sector)) {
       SERIALNAME->println(("PreallocatedFile: contiguousRange() failed"));
    // last_error = E_create_contiguous;
    //   return false;
  }
    
  // pre erase contiguous file. Is this redundant?
  uint32_t first_erase_sector = first_sector;
  const size_t erase_count = 64 * 1024; // number of sectors to erase at once
  while (first_erase_sector <= last_sector) {
    if (!sdEx.card()->erase(first_erase_sector, min(first_erase_sector + erase_count, last_sector))) {
           SERIALNAME->println(F("PreallocatedFile: erase() failed"));
      //last_error = E_erase;
      return false;
    }
    first_erase_sector += erase_count;
  }

//////////////////////////////////////////////////////
....
  static double TotalNumBytesWrittenToSD = 0;
  static double NbytesWrite;
....
// Wring to the card
    // Write data to SD card.
    if (TotalNumBytesWrittenToSD + NbytesWrite <= log_file_size) { //check to make sure there is sufficient space to write bytes
      NbytesWrite = dataFile.write((uint8_t*)bufCopy, sizeof(bufCopy));
      TotalNumBytesWrittenToSD += NbytesWrite;
      if ((unsigned) NbytesWrite != sizeof(bufCopy)) {
        SERIALNAME->println(F("not all data written to card"));
      }
    }
ite;

greiman, 2022-07-14:
Don't use floating point when you need uint64_t.  Even with 64 bit double you don't get the precision of unt64_t.

Here is an example that writes a 5000 GiB file and only uses uint64_t for sizes.
```
#include "SdFat.h"
SdFs sd;
FsFile file;
#define SD_CONFIG SdioConfig(FIFO_SDIO)
const uint64_t allocSize = 5000ULL*1024*1024;
const size_t bufSize = 32*1024;
uint8_t buf[bufSize];
void setup() {
  Serial.begin(9600);
  while (!Serial) {}
  Serial.println("\nType any character to begin");
  while (!Serial.available()) {}
  if (!sd.begin(SD_CONFIG)) {
    sd.initErrorHalt(&Serial);
  }
  
  if(!file.open("test.dat",O_RDWR | O_CREAT | O_TRUNC)) {
    sd.errorHalt("open failed");
  }
  if (!file.preAllocate(allocSize)) {
    sd.errorHalt("preAllocate failed");
  }
  uint32_t nw = 0;
  while ((file.fileSize() + bufSize) <= allocSize) {
    size_t n = file.write(buf, bufSize);
    if (n != bufSize) {
      Serial.print("\nWrite failed for n: ");
      Serial.println(n);
      break;
    }
    nw++;
    if (nw%1000 == 0) Serial.println(nw);
  }
  file.close();  
  sd.ls(LS_SIZE);
}

void loop() {
}
```

Here is how it ends:

> 158000
> 159000
> 160000
>   5242880000 test.dat

greiman, 2022-07-14:
These functions are useful for debug of uint64_t since Arduino print won't work on most boards.
```
const uint64_t allocSize = 5000ULL*1024*1024;

size_t print64(Print* pr, uint64_t n) {
  char buf[21];
  char *str = &buf[sizeof(buf) - 1];
  *str = '\0';
  do {
    uint64_t m = n;
    n /= 10;
    *--str = m - 10*n + '0';
  } while (n);
  pr->print(str);
}

size_t println64(Print* pr, uint64_t n) {
  return print64(pr, n) + pr->println();
}

void setup() {
  Serial.begin(9600);
  while (!Serial) {}
  println64(&Serial, allocSize);
}
void loop() {}
```

> 5242880000

greiman, 2022-07-15:
flush() calls sync() which does the equivalent of file.close().  So validLength and dataLength will be written to the SD.

The problem is when another system sees the file.  Android, IOS, Linux, MacOS, and Windows may truncate the file at validLength or set validLength to dataLength or somthing else.  The exFAT spec was made public after many inplementations were done  so this is the result of reverse engineering.  I even suspect Windows 10 is different than Windows 11.

I can think of two approaches.

The first would be to write a T40 program that opens the file after the crash and truncates it at fileSize().  You would lose any data after the flush() call.  I don't allow read beyond validLength so the data can't be recovered with file reads.

The second would be to erase the file after preallocate and see what happens when you open it on another system.

I think I will do an experiment on Windows and maybe Linux.  The test will be to preallocate a few MB, write some text, close the file and open it on a PC.

greiman, 2022-07-15:
Looks like Windows uses dataLength as the file-size so you can read up to dataLength.  Windows did not reset validLength when I opened the file with a binary editor.  I could look at all data up to dataLength so you might be able to recover data after validLength.

I tried the Windows scan for errors and the SD passed and validLength was not changed.

When I edited the file with a text editor it changed  both validLength and dataLength to unexpected value. When I opened the file again with a binaryviewer if dound the added line was inserted and all the zero bytes were changed to space.

So you get unexpected behavior if validLength is not equal to dataLength.

greiman, 2022-07-15:
Windows 11 does not conform to Microsoft's specification.

> 7.6.5 ValidDataLength Field
> 
> The ValidDataLength field shall describe how far into the data stream user data has been written. Implementations shall update this field as they write data further out into the data stream. On the storage media, the data between the valid data length and the data length of the data stream is undefined. Implementations shall return zeroes for read operations beyond the valid data length.
> 
> If the corresponding File directory entry describes a directory, then the only valid value for this field is equal to the value of the DataLength field. Otherwise, the range of valid values for this field shall be:
> 
>     At least 0, which means no user data has been written out to the data stream
> 
>     At most DataLength, which means user data has been written out to the entire length of the data stream

nperentos, 2022-07-15:
Thank you once more. I will implement your suggestion to re-open and truncate to file.fileSize().