greiman/SdFat #391: SD card is not initialized

Kracozebr, 2022-07-16:
Hi, I try to work with your library on Arduino Uno and Due and have the issue with initializing SD card.
What I've tried: work with example QuickStart.
And the SD  card can be seen both from Uno and Due:
```
Invalid pin number

Restarting

Enter the chip select pin number: 8

Assuming the SD is the only SPI device.
Edit DISABLE_CHIP_SELECT to disable another device.

Card successfully initialized.

Card size: 31267 MB (MB = 1,000,000 bytes)

Volume is FAT32, Cluster size (bytes): 32768

Files found (date time size name):

Success!  Type any character to restart.

```
But when I try to work with your another example AvrAdcLogger. have the issue that card is not initialized.
I slightly modified your sketch (publish only modified raws to save space):
```
#define SD_FAT_TYPE 1 

const uint8_t SD_CS_PIN = 8; 

const uint8_t PIN_LIST[] = {0, 1, 2}; 

const float SAMPLE_RATE = 500;  // Must be 0.25 or greater.

const uint32_t MAX_FILE_SIZE_MiB = 1;  // 100 MiB file.
```
So when I run this sketch on Uno have issue:
```
sd.begin failed
Check SD format.
```
I have read the documentation and the issue https://github.com/greiman/SdFat/issues/321 and format the sd card with [SD Formatter](https://www.sdcard.org/downloads/formatter/). you suggested. But issue do not disappear.

I decided to try your example SDFormatter. On Uno  I have the  output:
```
Assuming the SD is the only SPI device.
Edit DISABLE_CS_PIN to disable an SPI device.

Assuming the SD chip select pin is: 8
Edit SD_CS_PIN to change the SD chip select pin.

Type any character to start

This program can erase and/or format SD/SDHC/SDXC cards.

Erase uses the card's fast flash erase command.
Flash erase sets all data to 0X00 for most cards
and 0XFF for a few vendor's cards.

Cards up to 2 GiB (GiB = 2^30 bytes) will be formated FAT16.
Cards larger than 2 GiB and up to 32 GiB will be formatted
FAT32. Cards larger than 32 GiB will be formatted exFAT.

Warning, all data on the card will be erased.
Enter 'Y' to continue: Y

Card size: 0.00 GB (GB = 1E9 bytes)
Card size: 0.00 GiB (GiB = 2^30 bytes)
Card will be formated FAT16

Options are:
E - erase the card and skip formatting.
F - erase and then format the card. (recommended)
Q - quick format the card without erase.

Enter option: 
```
On Due have the next error:
```
error: Get sector count failed.
SD errorCode: SD_CARD_ERROR_READ_TOKEN = 0x19
SD errorData = 0x20
```
Then I tried your example SdInfo and get:
```
Assuming the SD chip select pin is: 8
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start
init time: 95 ms

Card type: SDHC
sdSpecVer: 1.01
HighSpeedMode: false

Manufacturer ID: 0X0
OEM ID: 
Product: 
Revision: 0.0
Serial number: 0XA00000
Manufacturing date: 1/2002

cardSize: 0.00 MB (MB = 1,000,000 bytes)
flashEraseSize: 128 blocks
eraseSingleBlock: false
dataAfterErase: zeros

OCR: 0XC0FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0X80,0X1,0X0,0X4,0XFE,0XFF,0XFF,0,27375616
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

volumeBegin failed. Is the card formatted?
```

The program does not see the card size. I checked 2 cards of 32 gb SDHC that are new and 100% work.
So I can not understand why it can not see the SD CARD.
And another question may be there is some way to modify AvrAdcLogger to work with arm32 (Arduino Due in my  case)?

greiman, 2022-07-16:
This is a problem with SPI/wiring

> error: Get sector count failed.
> SD errorCode: SD_CARD_ERROR_READ_TOKEN = 0x19
> SD errorData = 0x20

This card is not formatted correctly.  Looks  like the card maybe GPT. Run the [SD formatter](https://www.sdcard.org/downloads/formatter/)

> SD Partition Table
> part,boot,bgnCHS[3],type,endCHS[3],start,length
> 1,0X0,0X80,0X1,0X0,0X4,0XFE,0XFF,0XFF,0,27375616
> 2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

greiman, 2022-07-16:
AvrAdcLogger depends on the interrupt structure of the AVR ADC- thats'w why it's name starts with Avr.

Kracozebr, 2022-07-17:
> This is a problem with SPI/wiring
>
>
> This card is not formatted correctly. Looks like the card maybe GPT. Run the [SD formatter](https://www.sdcard.org/downloads/formatter/)

I ve tried another card with 2 gb and format with sd formatter you suggested. This is output of Quick Start example:
```
Enter the chip select pin number: 8

Assuming the SD is the only SPI device.
Edit DISABLE_CHIP_SELECT to disable another device.

Card successfully initialized.

Card size: 2014 MB (MB = 1,000,000 bytes)

Volume is FAT16, Cluster size (bytes): 32768

Files found (date time size name):

Success!  Type any character to restart.

```
When try  the SdInfo example:
```
Assuming the SD chip select pin is: 8
Edit SD_CS_PIN to change the SD chip select pin.

type any character to start
init time: 2 ms

Card type: SD1
sdSpecVer: 1.01
HighSpeedMode: false

Manufacturer ID: 0X0
OEM ID: 
Product: 
Revision: 0.0
Serial number: 0XC3E300E0
Manufacturing date: 1/2009

cardSize: 940.05 MB (MB = 1,000,000 bytes)
flashEraseSize: 64 blocks
eraseSingleBlock: true
dataAfterErase: ones

OCR: 0X80FF8000

SD Partition Table
part,boot,bgnCHS[3],type,endCHS[3],start,length
1,0X0,0X0,0X0,0X0,0X2,0X7,0XEF,0XC7,253,1835777
2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0

volumeBegin failed. Is the card formatted?
```
And AvrAdcLogger outputs the error:
```
sd.begin failed
Check SD format.
```

greiman, 2022-07-17:
Here are two possibilities.  

The SD card appears to be old and only seems to be a version 1.01 card.  It may not have correct support for SPI mode.  On PCs and other devices the SDIO protocol is used so on Windows, Linux, or a Mac it will work.

One of my oldest SanDisk cards has a 2006 date but is version 2.00.

Something may be marginal with with your wiring or SD shield/module.  What are you using for an SD shield/module?

The main difference between QuickStart and other examples is use of a slower SPI clock.

Try using a slower SPI rate on other examples to see if the  wiring or SD shield/module is a problem.

On the SdInfo example change SD_SCL_MHZ(16) to SD_SCL_MHZ(4) like this:
```
#if HAS_SDIO_CLASS
#define SD_CONFIG SdioConfig(FIFO_SDIO)
#elif ENABLE_DEDICATED_SPI
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, DEDICATED_SPI, SD_SCK_MHZ(4))  // <--- was 16
#else  // HAS_SDIO_CLASS
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, SHARED_SPI, SD_SCK_MHZ(4))  // <----was 16
#endif  // HAS_SDIO_CLASS
```
If SD_SCL_MHZ(4) fails try SD_SCL_MHZ(2).

greiman, 2022-07-17:
Here is an example of performance for a Due with the correct SD socket, short wires and a modern microSD.

The SD socket is [here ](https://www.ebay.com/itm/171907090277)on ebay.
 
Output from the unmodified bench example:

> write speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 4516.71,4106,110,111
> 4512.64,4176,110,111
> 
> read speed and latency
> speed,max,min,avg
> KB/Sec,usec,usec,usec
> 4528.99,113,110,111
> 4537.21,113,110,111

SdInfo for the card, a [Samsung EVO+](https://www.amazon.com/Samsung-Class-Micro-Adapter-MB-MC32DA/dp/B00WR4IJBE):

> Card type: SDHC
> sdSpecVer: 3.00
> HighSpeedMode: true
> 
> Manufacturer ID: 0X1B
> OEM ID: SM
> Product: 00000
> Revision: 1.0
> Serial number: 0XB2ED5782
> Manufacturing date: 4/2016
> 
> cardSize: 32010.93 MB (MB = 1,000,000 bytes)
> flashEraseSize: 128 blocks
> eraseSingleBlock: true
> dataAfterErase: ones
> 
> OCR: 0XC0FF8000
> 
> SD Partition Table
> part,boot,bgnCHS[3],type,endCHS[3],start,length
> 1,0X0,0X82,0X3,0X0,0XC,0XFE,0XFF,0XFF,8192,62513152
> 2,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 3,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 4,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0X0,0,0
> 
> Scanning FAT, please wait.
> 
> Volume is FAT32
> sectorsPerCluster: 64
> fatStartSector:    9316
> dataStartSector:   24576
> clusterCount:      976512
> freeClusterCount:  976323

Kracozebr, 2022-07-18:
@greiman Thanks for replying. The issue was with sd socket I changed to 4 Mhz as you suggested and it works. Also I tried 8 Mhz and the error came back, so I insert 6 Mhz and it works. There is any parity in setting frequency i.e. it should be divisible by 2 or smth else?

closed 2022-07-18 by Kracozebr

maxbrito500, 2023-03-28:
Thanks. Just to say that changing the SPI_Clock speed to 6 has worked for ESP32-WROOM.

`#define SPI_CLOCK SD_SCK_MHZ(6)`

After that was possible to format the storage card (cheap chinese microSD with 16GB)