greiman/SdFat #394: 4-bit SDIO

davepruitt, 2022-08-09:
Is there any support for 4-bit SDIO, rather than just single-bit SPI? If not, can I make this a feature request?

greiman, 2022-08-09:
There is 4-bit SDIO for Teensy 3.x and 4.x.  The NXP SDIO controller is excellent and I get read/write rates of about 22 MB/sec.  I personally developed the driver.   See [this ](https://github.com/greiman/SdFat/blob/master/examples/TeensySdioDemo/TeensySdioDemo.ino)example.

I have experimented with ESP32 and ESP32C.  The controller is not capable of extended transfers and transfers must be aligned on 4-byte boundaries.  The ESP driver often resorts to a series of 512 byte transfers in many cases and write is about 400 KB/sec.  Shared SPI on a dedicated SPI port at 40 MHz is much faster for most apps.

I think there is a port of SdFat with an SDIO driver for some STM32 boards but not for the STMicroelectronics board package.

See this package:

https://github.com/rogerclarkmelbourne/Arduino_STM32

Most microprocessor chips have poor SDIO controllers.  NXP uses technology from their Applications Processor series.

davepruitt, 2022-08-09:
I was hoping to use 4-bit SDIO on the SAMD21 or SAMD51 if possible.

greiman, 2022-08-09:
The SAMD21 has no SDIO controller.

The SAMD51 has a SDIO controller and is supported in CircuitPython 6.0 and above.  

I have not located an Arduino SAMD51 SDIO driver to test with. Any tests I have found on the web give poor results.  Here are examples:

https://www.alkgrove.com/articles/SAMD51_MicroSD_article.html
 
Also see page 38 here:

https://cdn-learn.adafruit.com/downloads/pdf/adafruit-microsd-spi-sdio.pdf

> 
> CircuitPython:
> sdcardio: For SD cards on SPI interfaces.
> sdioio: For SD cards on SDIO interfaces
> 
> SD card access with sdcardio on Grand Central:
> Reads up to 650kB/s
> Data logging at up to 700 of lines per second
> 
> SD card access with sdioio on Grand Central with SDIO breakout:
> Reads up to 1100kB/s
> Data logging at over 1000 lines per second

I can read at over 3000kB/sec on an old Due with SPI so I have no interest in writing an SDIO driver for SAMD51.