greiman/SdFat #395: visual block map

tbillion, 2022-08-09:
im looking for a way to see if a block is empty or contains data is there a function for this? 
attempting to make something like this
[image]
:
but i am  not familiar enough with the library to figure it out .

tbillion, 2022-08-09:
```
  uint8_t blk[512];
  bool blkEpty[2209] = {0};                           //each sector is 512 bytes
  uint32_t blkPerBool = volumesize / 2208;      //how many sectors are represented by a single color block.
  uint32_t bitsPerBool = blkPerBool * 512;      //how many loops it takes to verify 1 color block on lcd
  uint32_t bitCount = 0;
  uint32_t blockCount = 0;

  tft.drawString(("Sectors per Color Block: " + String(blkPerBool)), 57, 50 + tft.fontHeight() * 3, 1);
  tft.drawString(("Bytes per Color Block: " + String(bitsPerBool)), 57, 50 + tft.fontHeight() * 4, 1);

  for (uint32_t tSector = 0; tSector <= size; tSector++) {
    if ( tSector % 100 == 0 ) {
      tft.drawString(("CYCLE: " + String(tSector) + " out of: " + String(size)), 57, 50 + tft.fontHeight() * 6, 1);
    }
    if (sd.card()->readSector(tSector, blk)) {
      for (uint32_t tByte = 0; tByte <= 512; tByte++) {
        if (blk[tByte] != 255 | blk[tByte] != 0) {
          tByte = 513;
          blkEpty[blockCount] = 1;
        } else {
          blkEpty[blockCount] = 0;
        }
      }
    }
    bitCount++;
    if (bitCount >= bitsPerBool) {
      blockCount++;
      bitCount = 0;
    }
  }

```

think i figured it out.. the library reference from 2012 is wildly inaccurate. thin that was the cause of my problem.

greiman, 2022-08-10:
The current reference for SdFat classes was last updated on 4 July 2022 and is here:

Arduino/libraries/SdFat/doc/SdFat.html

Replace the content of the html folder by unzipping html.zip.
I have zipped the documentation since Doxygen changes every file each time it runs.
This makes viewing changes on GitHub difficult.

see these classes:

> DedicatedSpiCard Class Reference
> SharedSpiCard Class Reference

tbillion, 2022-08-15:
to see if a cluster or a sector on a fat table is allocated or not.. i have been all through this documentation and i cant find an answer i assume it is no, is there a way to access the fat table to see what sectors are occupied and which are free. 

what i thought would be a two day coding experiment has turned into 3 weeks .

greiman, 2022-08-15:
There is no simple way.  With exFat it is the bit map, not the FAT which is only used for chaining in non-contiguous files.  

With FAT16/FAT32 you can look at  `int8_t FatPartition::fatGet(uint32_t cluster, uint32_t* value) ` In this file:

https://github.com/greiman/SdFat/blob/master/src/FatLib/FatPartition.cpp

With exFAT look at this function `uint32_t ExFatPartition::bitmapFind(uint32_t cluster, uint32_t count)` in this file:

https://github.com/greiman/SdFat/blob/master/src/ExFatLib/ExFatPartition.cpp

greiman, 2022-08-15:
Maybe a better answer is to look at freeClusterCount().

See int32_t FatPartition::freeClusterCount() in 
https://github.com/greiman/SdFat/blob/master/src/FatLib/FatPartition.cpp
and int32_t ExFatPartition::freeClusterCount() in
https://github.com/greiman/SdFat/blob/master/src/ExFatLib/ExFatPartition.cpp

tbillion, 2022-08-15:
i already use free cluster count, is there a way to return which clusters are free (like the number of the cluster)?

greiman, 2022-08-15:
It is not practical to return a list of free clusters since a 32GB SD has about 1,000,000 clusters.  This would require a 4 MB array. exFAT volumes can have even more clusters.

You need to look at the code in freeClusterCount() and modify it for your use.

closed 2022-08-15 by tbillion