greiman/SdFat #398: Library can get stuck at this point

h4yn0nnym0u5e, 2022-09-03:
https://github.com/greiman/SdFat/blob/628effa1c277e6dc1fef461b4b755ad300735248/src/SdCard/SdioTeensy.cpp#L853

Not sure of the trigger for this, but I'm experiencing issues where this while() loop just spins forever. As similar flag waits seem to have a timeout wrapper, I suspect this should do so as well...

greiman, 2022-09-04:
>  As similar flag waits seem to have a timeout wrapper

Both read and write FIFO have the same loop.  Here is the write loop at about line 1020.

```
    while (0 == (SDHC_PRSSTAT & SDHC_PRSSTAT_BWEN)) {
    }
```

I will add a timeout to each loop.  I just copied the loops from an NXP driver.

h4yn0nnym0u5e, 2022-09-05:
Sounds good - I'll have to read up on how to layer it over Teensyduino until it gets formally adopted! RTFM...

greiman, 2022-09-05:
I am curious to know what SD you are using and what version of Teensy..  The driver has been used since Teensy 3.6, about five years ago, with those while loops.

h4yn0nnym0u5e, 2022-09-05:
SanDisk Extreme Plus 64Gb SDXC A2 with Teensyduino 1.57. 

I'm working on a variant of the SD audio playback object that executes all its reads in the foreground / sketch context, rather than inside the 2.9ms audio interrupt. This leads to a use case which has multiple open files (tested up to 16) with essentially serial access in large chunks - say 4k - 32k - buffered into PSRAM. I'm going to some lengths to ensure the chunk reads are all on 512-byte boundaries.

In a further twist, based on an idea from the forum, I have a set of stereo piano samples (should be 88 notes x 3 velocity layers, but some are MIA so it's actually only 259...), preloading 170ms of each into PSRAM, then triggering playback of the preload on a MIDI note on event, and hastily opening and seeking the relevant file, and grafting its playback seamlessly onto the end of the preloaded audio. It works pretty well if you don't thrash the keyboard too hard, but sometimes got stuck as noted; I found this forum comment, if that sheds any light: https://community.nxp.com/t5/Kinetis-Microcontrollers/K60-SDHC-hanging-problem/m-p/159111

I'm fairly sure there are still issues with my code (aren't there always!), but it'd be good to know this won't be an issue in the long run, even if I get an error return from the read so the note glitches.

h4yn0nnym0u5e, 2022-09-07:
I found my issue - I was closing a file in an ISR while other files were in the process of being read. This turns out to be a Bad Thing... 

The timeout is probably still a good idea, unless we're 100% sure the hardware can never get stuck no matter how you abuse it.

greiman, 2022-09-08:
Thanks for the info. I was really puzzled how this could happen.  I wait for clocking of data from the SD to start.  Then data should arrive even if the SD fails to put the correct data on the bus.  The error will be caught as a bad CRC.

Here is the code:
```
  static bool isBusyFifoRead() {
    return !(SDHC_PRSSTAT & SDHC_PRSSTAT_BREN);
  }

  // ...

  if (waitTimeout(isBusyFifoRead)) {  // <---  timeout for clock to start.
    return sdError(SD_CARD_ERROR_READ_FIFO);  // <-- waits for data transfer to start
  }
  for (uint32_t iw = 0 ; iw < 512/(4*FIFO_WML); iw++) {
   //  Loop should not fail since I set the controller to clock 512 bytes.
    while (0 == (SDHC_PRSSTAT & SDHC_PRSSTAT_BREN)) {  
    }
    for (uint32_t i = 0; i < FIFO_WML; i++) {
      p32[i] = SDHC_DATPORT;
    }
    p32 += FIFO_WML;
  }
```
This is the test for correct data:
`  return (m_irqstat & SDHC_IRQSTAT_TC) && !(m_irqstat & SDHC_IRQSTAT_ERROR);`

I will not put a timeout in the loop since this is the magic that allows 22 MB/sec transfers.  

It is even more important in the write loop.  The transfer of a sector in the tight loop takes about 5 μs.

This allows people to buffer data and limit write busy times to around 5 μs like this.
```
  if (!sd.isBusy() ) {
    file.write(buf, count);  //returns in about 5 μs if count <= 512.
  }
```

h4yn0nnym0u5e, 2022-09-08:
Hmmm ... 5µs on a Teensy 4.1 is about 3000 instructions, so it's hard to believe there aren't a _few_ spare cycles to allow a timeout for incompetent buffoons like me! Might be different on another Teensy, of course.

I'll have to stick a scope on and look at the timings. With FIFO_WML at 16 you'd only do a timeout check 8 times inside the outer loop, which won't slow the inner loop down at all.

Thanks for taking the time to look at this, anyway, much appreciated.

greiman, 2022-09-08:
Don't bother with a scope, I could do that.  I really have no interest in tying to catch bugs due to access to SdFat routines from interrupt code.

I am also getting more multi-thread problems, people with ESP32 try to access files from multiple threads.  There are infinite possible ways to create bugs. 

I may increase WR_WML to 128.  Then I can just copy the whole 512 bytes to the FIFO in a tight loop.  Maybe reduce the 5µs a bit more.

h4yn0nnym0u5e, 2022-09-08:
Too late, I tried it ... interesting. I get 22.4µs to read 512 bytes (reading 32 sectors in one "burst"), which corresponds pretty well with your 22MB/sec. Not sure how 5µs fits with that, it would give 100MB/sec read speed - possible for the SD card, I think, not sure about for this processor. In that time it usually spends 1.95µs waiting for the FIFO to fill, and 0.65µs reading it to memory: the first wait is a bit longer, and there's some other overhead.

The following modification appears to run with zero speed penalty on the Teensy 4.1. If it were my library I'd 100% apply this fix, at least it eliminates one way (two, including the write function) in which users' code can "just mysteriously stall for no apparent reason".

But that's your call, of course.

```
bool SdioCard::readData(uint8_t* dst) {
digitalWriteFast(25,1);
  DBG_IRQSTAT();
  uint32_t* p32 = reinterpret_cast<uint32_t*>(dst);

  if (!(SDHC_PRSSTAT & SDHC_PRSSTAT_RTA)) {
    SDHC_PROCTL &= ~SDHC_PROCTL_SABGREQ;
    noInterrupts();
    SDHC_PROCTL |= SDHC_PROCTL_CREQ;
    SDHC_PROCTL |= SDHC_PROCTL_SABGREQ;
    interrupts();
  }
  // ******** pre-loop timeout check removed, it's now inside ********
  for (uint32_t iw = 0 ; iw < 512/(4*FIFO_WML); iw++) {

  // ******** timeout added to flag check, no speed penalty on Teensy 4.1 ******** 
	if (waitTimeout(isBusyFifoRead)) {
	return sdError(SD_CARD_ERROR_READ_FIFO);
	}
  // *********************************************************************************
   for (uint32_t i = 0; i < FIFO_WML; i++) {
      p32[i] = SDHC_DATPORT;
    }
    p32 += FIFO_WML;
  }
  if (waitTimeout(isBusyTransferComplete)) {
    return sdError(SD_CARD_ERROR_READ_TIMEOUT);
  }
  m_irqstat = SDHC_IRQSTAT;
  SDHC_IRQSTAT = m_irqstat;
 return (m_irqstat & SDHC_IRQSTAT_TC) && !(m_irqstat & SDHC_IRQSTAT_ERROR);
}
```

greiman, 2022-09-09:
The 5 μs is for write. Of course read takes longer, the data is coming over the 4-bit SDIO bus at about 50MHz. 

I already decided to put the timeout in the read loop.  The RD_WML of 16  is probably not optimal for read, a smaller value is probably better.  You want the smallest RD_WML that keeps up with the SD.  That will minimize the total time to empty the FIFO. This means the timeout will limit how small RD_WML can be.

 I plan set separate RD_WML and WR_WML values.  Probably a RD_WML smaller than 16 is optimal.

I mostly worry about write since many apps logging data at high speed need minimal latency. See [this example](https://github.com/greiman/SdFat/blob/master/examples/TeensySdioLogger/TeensySdioLogger.ino#L71) at line 71 for how novice users achieve high logging rates with simple code.

With write, the 512 byte FIFO is empty so the optimal way to fill it will be a single loop with one check before the loop.

This will be replaced
```
  if (waitTimeout(isBusyFifoWrite)) {
    return sdError(SD_CARD_ERROR_WRITE_FIFO);
  }
  for (uint32_t iw = 0 ; iw < 512/(4*FIFO_WML); iw++) {
    while (0 == (SDHC_PRSSTAT & SDHC_PRSSTAT_BWEN)) {
    }
    for (uint32_t i = 0; i < FIFO_WML; i++) {
      SDHC_DATPORT = p32[i];
    }
    p32 += FIFO_WML;
  }
```
With this:
```
  if (waitTimeout(isBusyFifoWrite)) {
    return sdError(SD_CARD_ERROR_WRITE_FIFO);
  }
  for (uint32_t i = 0; i < 128; i++) {
    SDHC_DATPORT = p32[i];
  }
```

greiman, 2022-09-10:
I looked at some more sophisticated user applications and decided to leave the read loop as is.  The reason is that while in your simple application, cycles lost in the read loop seem to not matter as long as you still get 22 MB/sec reads.

SdFat is being used in complex systems with a RTOS. These systems have an SD thread running at low priority so most of the  22.4µ in the read loop are actually available for higher priority threads.  Teensy 4.x takes very few cycles to do a context switch.

Few novices use DMA with SdFat but there is a hook in SdFat for RTOS users, a call to yield in `yieldTimeout()`.  This allows users to use large read ahead or write behind queues between high priority processes and a low priority SD process.  All of the cycles that would have been used in the FIFO read/write loops are then available to high priority threads.

DMA is competitive with FIFO read/write speeds if you do 32KB reads and writes.

Your crashes were in the driver read loop but it happens in other parts of SdFat when uses have isr call or multi-thread problems.  The answer to use a hardware based debugger and fix the problem not put a patch in each area where the crash happens.

h4yn0nnym0u5e, 2022-09-10:
As I said above, it’s your call. 

Teensy users do NOT have the option of a hardware debugger, and to my mind any loop inside a library which doesn’t provide an error-trapped exit for a condition that has actually been observed by a user is a bug, pure and simple. As you’ve said yourself, doing this (my version of) properly costs no-one anything, as it doesn’t slow the code down, or increase its size - in fact my fix makes the code smaller, by moving the timed-out loop and removing the infinite one. 

I’m not requesting a thorough scan of all possible similar instances where problems _might_ occur - makes more sense to fix them if they do crop up, and it would be an unreasonable request. But I think it’s reasonable to ask for this.

closed 2022-09-10 by greiman