greiman/SdFat #400: LowLatencyLogger example fails to compile on Artemis Redboard Nano

benseleb, 2022-09-15:
Hi there,

I'm simply trying to compile the built-in SdFat example "LowLatencyLogger" on the Artemis Redboard Nano.
The example compiles fine on other boards, like the Arduino Uno. 
However, when compiling on the Sparkfun Redboard Nano, I get the following error message. `'FreeStack' was not declared in this scope` Multiple source files seem to go unused or unrecognized. I'm assuming something odd is going on under the hood -- possibly due to the Apollo3 MCU. Any advice would be appreciated!

My current versions:
- Arduino IDE: 1.8.19
- Sparkfun Apollo3 Boards: 2.2.1
- SdFat: 2.2.0
- Programmer: Ambiq Secure Bootloader

> C:\Users\Ben\Documents\Arduino\libraries\SdFat-master\examples\examplesV1\LowLatencyLogger\LowLatencyLogger.ino: In function 'void checkOverrun()':
LowLatencyLogger:150:18: error: 'FreeStack' was not declared in this scope
   Serial.println(FreeStack());
                  ^~~~~~~~~
C:\Users\Ben\Documents\Arduino\libraries\SdFat-master\examples\examplesV1\LowLatencyLogger\LowLatencyLogger.ino:150:18: note: suggested alternative: 'FreeStack_h'
   Serial.println(FreeStack());
                  ^~~~~~~~~
                  FreeStack_h
LowLatencyLogger:165:28: error: 'SdBaseFile' {aka 'class FsBaseFile'} has no member named 'firstBlock'
       Serial.print(binFile.firstBlock() + bn);
                            ^~~~~~~~~~
C:\Users\Ben\Documents\Arduino\libraries\SdFat-master\examples\examplesV1\LowLatencyLogger\LowLatencyLogger.ino: In function 'void binaryToCsv()':
LowLatencyLogger:194:18: error: 'FreeStack' was not declared in this scope
   Serial.println(FreeStack());
                  ^~~~~~~~~
C:\Users\Ben\Documents\Arduino\libraries\SdFat-master\examples\examplesV1\LowLatencyLogger\LowLatencyLogger.ino:194:18: note: suggested alternative: 'FreeStack_h'
   Serial.println(FreeStack());
                  ^~~~~~~~~
                  FreeStack_h
C:\Users\Ben\Documents\Arduino\libraries\SdFat-master\examples\examplesV1\LowLatencyLogger\LowLatencyLogger.ino: In function 'void createBinFile()':
LowLatencyLogger:259:16: error: 'SdBaseFile' {aka 'class FsBaseFile'} has no member named 'createContiguous'; did you mean 'isContiguous'?
   if (!binFile.createContiguous(TMP_FILE_NAME, 512 * FILE_BLOCK_COUNT)) {
                ^~~~~~~~~~~~~~~~
                isContiguous
C:\Users\Ben\Documents\Arduino\libraries\SdFat-master\examples\examplesV1\LowLatencyLogger\LowLatencyLogger.ino: In function 'void recordBinFile()':
LowLatencyLogger:367:39: error: 'class FsVolume' has no member named 'cacheClear'
   emptyStack[0] = (block_t*)sd.vol()->cacheClear();
                                       ^~~~~~~~~~
LowLatencyLogger:379:38: error: 'SdBaseFile' {aka 'class FsBaseFile'} has no member named 'firstBlock'
   if (!sd.card()->writeStart(binFile.firstBlock())) {
                                      ^~~~~~~~~~
LowLatencyLogger:383:18: error: 'FreeStack' was not declared in this scope
   Serial.println(FreeStack());
                  ^~~~~~~~~
C:\Users\Ben\Documents\Arduino\libraries\SdFat-master\examples\examplesV1\LowLatencyLogger\LowLatencyLogger.ino:383:18: note: suggested alternative: 'FreeStack_h'
   Serial.println(FreeStack());
                  ^~~~~~~~~
                  FreeStack_h
C:\Users\Ben\Documents\Arduino\libraries\SdFat-master\examples\examplesV1\LowLatencyLogger\LowLatencyLogger.ino: In function 'void setup()':
LowLatencyLogger:572:18: error: 'FreeStack' was not declared in this scope
   Serial.println(FreeStack());
                  ^~~~~~~~~
C:\Users\Ben\Documents\Arduino\libraries\SdFat-master\examples\examplesV1\LowLatencyLogger\LowLatencyLogger.ino:572:18: note: suggested alternative: 'FreeStack_h'
   Serial.println(FreeStack());
                  ^~~~~~~~~
                  FreeStack_h
Using library SPI at version 2.0.0 in folder: C:\Users\Ben\AppData\Local\Arduino15\packages\SparkFun\hardware\apollo3\2.2.1\libraries\SPI 
Using library SdFat-master at version 2.2.0 in folder: C:\Users\Ben\Documents\Arduino\libraries\SdFat-master 
exit status 1
'FreeStack' was not declared in this scope

greiman, 2022-09-15:
The LowLatencyLogger  example was released for SdFat V1 for FAT16/FAT32.  

By default boards like Redboard with Apollo3 support exFAT and FAT16/FAT32. Not all V1 API functions are available.

You could try to get the old example to work by editing SdFatConfig.h and restrict support to FAT16/FAT32 and delete and use of FreeStack functions.

I don't plan to update LowLatencyLogger  so this is not a good choice.  

The new example with low latency logging is [ExFatLogger](https://github.com/greiman/SdFat/blob/master/examples/ExFatLogger/ExFatLogger.ino) which supports FAT16/FAT32 and exFAT. I suggest you try it.  There will not be support for FreeStack but the example should still work since there is a test for FreeStack support and no stack use info will be printed.

Most compile errors will go away for LowLatencyLogger  if you edit SdFatConfig.h and restrict support to FAT16/FAT32 here:
```

/**
 * File types for SdFat, File, SdFile, SdBaseFile, fstream,
 * ifstream, and ofstream.
 *
 * Set SDFAT_FILE_TYPE to:
 *
 * 1 for FAT16/FAT32, 2 for exFAT, 3 for FAT16/FAT32 and exFAT.
 */
#ifndef SDFAT_FILE_TYPE
#if defined(__AVR__) && FLASHEND < 0X8000
// 32K AVR boards.
#define SDFAT_FILE_TYPE 1
#else  // defined(__AVR__) && FLASHEND < 0X8000
// All other boards.
#define SDFAT_FILE_TYPE 3   // <<<---------------------- change three to one.
#endif  // defined(__AVR__) && FLASHEND < 0X8000
#endif  // SDFAT_FILE_TYPE
```
`

benseleb, 2022-09-16:
Thanks for all the info! As you advised, I've moved on to ExFatLogger. It also fails to compile initially, with the FillStack() function being undefined.  
Compiler error message: 
```
c:/users/ben/appdata/local/arduino15/packages/sparkfun/tools/arm-none-eabi-gcc/8-2018-q4-major/bin/../lib/gcc/arm-none-eabi/8.2.1/../../../../arm-none-eabi/bin/ld.exe: sketch\ExFatLogger.ino.cpp.o: in function `setup':
ExFatLogger.ino.cpp:(.text.setup+0x2e): undefined reference to `FillStack()'
collect2.exe: error: ld returned 1 exit status
exit status 1
Error compiling for board RedBoard Artemis Nano.
```
I see that FillStack is defined in FreeStack, which you mentioned wont work. I commented out the call to FillStack, and it compiled fine. 

The default sensor test works well. However, when I try to record it stops behaving. As I can tell from the serial monitor, a binary file is successfully created and 1Gb is successfully formatted. 
```
Type any character to begin
1024 FIFO entries will be used.

type: 
b - open existing bin file
c - convert file to csv
l - list files
p - print data to Serial
r - record data
t - test without logging

ExFatLogger00.bin
preAllocated: 1024 MiB
```

The code hangs there, without moving on. Using some serial prints, I narrowed the freeze down to the binFile write here: 
```
 // Write dummy sector to start multi-block write.
  dbgAssert(sizeof(fifoBuf) >= 512);
  memset(fifoBuf, 0, sizeof(fifoBuf));
  if (binFile.write(fifoBuf, 512) != 512) {         // <------------- gets unhappy here 
    error("write first sector failed");
  }
  clearSerialInput();
  Serial.println(F("Type any character to stop"));

```
I reduced the size of the FIFO_SIZE_SECTOR for non-AVR boards from 16 to 8 KiB, and that seems to have resolved the issue! I'll try bumping it up a bit from there, but I know that value works. 
```
// FIFO SIZE - 512 byte sectors.  Modify for your board.
#ifdef __AVR_ATmega328P__
// Use 512 bytes for 328 boards.
#define FIFO_SIZE_SECTORS 1
#elif defined(__AVR__)
// Use 2 KiB for other AVR boards.
#define FIFO_SIZE_SECTORS 4
#else  // __AVR_ATmega328P__
// Use 8 KiB for non-AVR boards.
#define FIFO_SIZE_SECTORS 8     // <------ change this to 4 
#endif  // __AVR_ATmega328P__
```

Again, thanks for your help and great upkeep!

closed 2022-09-16 by benseleb