greiman/SdFat #401: Card will not sleep after initialization

hobbya, 2022-09-17:
SD card will not sleep after initialization with below configuration:
#define SD_CONFIG SdSpiConfig(CS, DEDICATED_SPI, SD_SCK_MHZ(25), &SPI)
if (!SD.begin(SD_CONFIG))

It will sleep with below configuration:
#define SD_CONFIG CS,SD_SCK_MHZ(25)
if (!SD.begin(SD_CONFIG))

Card can work properly with either configuration but with the first one it just will not sleep.

hobbya, 2022-09-17:
The first configuration always brings SCK pin to High after initialization but the second one always brings it Low.

hobbya, 2022-09-17:
The card will sleep if I change DEDICATED_SPI to SHARED_SPI
I only mount the card to a particular SPI port.
I am using STM32F411 which got up to 5 SPI ports and I have mounted 3 peripherals on 3 individual SPI ports.

greiman, 2022-09-17:
Shared SPI mode achieves high speed by keeping the SD active in an extended read/write  whenever possible.

You should be able to enable sleep by calling:   

`sd.card()->setDedicatedSpi(false);`

 before sleep and activate the faster mode with: 

`sd.card()->setDedicatedSpi(true);`

greiman, 2022-09-17:
A simpler solution may be this call before sleep:

`  sd.card()->syncDevice();`

I didn't think there was reason users would call this functions but it simply completes any read or write.  The SD should sleep after this call and you don't need to do anything to use dedicated SPI mode upon wake.

hobbya, 2022-09-18:
Thank you for your quick response and it worked like a charm:
`SD.card()->syncDevice();`

Now the card only draws 0.06mA after system shuts down under dedicated SPI mode.

closed 2022-09-18 by hobbya