greiman/SdFat #409: SD card never detected with SdFat, but detects with SD.h

Nowisthetime07, 2022-12-22:
I am using ESP8266 NodeMCU.
I want to use the SdFat.h library but the problem is it never detects my SD card when using SdFat.  
So I created a new sketch to detect the card and write a file in SD.h and it works fine, it detects the card and writes to a file.
But not detecting the sd card when trying SdFat.h library.  I've been going in circles with this for days and cannot find out why? 

I will paste code below for both sketches.  

I would appreciate if someone could help me get the card detected with SdFat.h sketch.

Here is the code for the SD.h sketch which always detects the card and writes a file.

`#include <SPI.h>
#include <SD.h>

File file;

void setup() {
  // put your setup code here, to run once:

  //Initialize serial and wait for port to open:
  Serial.begin(57600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  // prints title with ending line break
  delay (1000);
  Serial.println("Serial Connected!");

    if (!SD.begin(D8))
    //if (!SD.begin(15))
    {
      Serial.println("SD Initialization failed!");
      while(1){} //infinite loop
    }
    else
    {
      Serial.println("SD Initialization Succeeded!");
      file = SD.open("testlog3.txt", FILE_WRITE);
      file.close();
      Serial.println("SD.H File created!");
    }

}

void loop() {
  // put your main code here, to run repeatedly:
}
`

Here is the code for the SdFat.h sketch which never detects the card sd.begin(D8) always returns false

`
#include <SPI.h>
#include <SdFat.h>

SdFat sd;
File32 measurFile;

void setup() {
  // put your setup code here, to run once:

  //Initialize serial and wait for port to open:
  Serial.begin(57600);
  while (!Serial) {
    ; // wait for serial port to connect. Needed for native USB port only
  }
  
  // prints title with ending line break
  //Serial.println("Serial Connected!");
  delay (1000);
  Serial.println("\nSerial Connected!");

      if (!sd.begin(D8))
      {
        Serial.println("SD Initialization failed!");
        Serial.println("Try reducing to 30.");
        //while(1){}
      }
      else
      {
        Serial.println("SD Initialization Succeeded!");
        if(!measurFile.open("SdFat.csv", O_WRITE | O_CREAT | O_TRUNC))
        {
          Serial.println("Write failed!");
          while(1){}
        }
        else
        {
          measurFile.print("Time,Voltage,Current\n");
          measurFile.sync();
          Serial.println("SD.H File created!");
        }
      }
     
}
`

Would appreciate any help!
Thank you.

greiman, 2022-12-23:
I don't have a ESP8266 NodeMCU so I can't help much.

You might try a slower SPI rate.  SdFat requests 50Mhz and the ESP8266 system should use the highest supported speed.  Sometimes this speed is still too high.

Try sd.begin(CS_PIN, maxSck) where maxSck is 20000000UL or less.

You could try this fork:

https://github.com/earlephilhower/ESP8266SdFat

Nowisthetime07, 2022-12-23:
This worked "Try sd.begin(CS_PIN, maxSck) where maxSck is 20000000UL or less.".
I'm able to detect the card now.

Thanks so much!