greiman/SdFat #410: Question: To write a string to the SD < 512 bytes

el-samiyel, 2022-12-27:
Hi All,

I am moving away from Arduino standard SD library and am trying to develop a better understanding of how to effectively use an SD card in my project. Research shows that SD writes should be handled in blocks of 512 bytes. My issue is that most examples on the internet show `String x = "This is my test string."; myFile.Write(x); myFile.close();` or something similar.

My calculations show the string to be 24 bytes, so how is this written to the SD when it is far less than 512 bytes?

I suppose the real reason for asking is that with the standard SD lib, I was finding that after writing multiple strings to my SD, the SD would end up becoming unformatted.. This has led me to look for other libs i.e. SDFAT and let to a bunch of other research i.e. the 512 buffer.

A simplified explanation would be much appreciated.

S

greiman, 2022-12-28:
The standard SD.h library is just a wrapper for an old version of SdFat so file.write(x), file.print(n) etc are the same.

I suggest you look at various examples.  Start with [BackwardCompatibility](https://github.com/greiman/SdFat/blob/master/examples/BackwardCompatibility/BackwardCompatibility.ino).

el-samiyel, 2022-12-28:
Thanks, yes I realised that shortly after posing.

But I still don't understand the whole 512 buffer. Does the lib automatically handle the 512 bytes? As, `myFile.println("testing 1, 2, 3.");  myFile.close();` isn't 512 bytes...or am I getting my wires crossed?

Thanks

S

greiman, 2022-12-28:
SD cards are block devices like hard drives.  You don't need to worry about 512 bytes buffers and sectors on the SD.  That's the function of the filesystem.

Just open a file, read from the file or print/write to the file, and finally close the file.

el-samiyel, 2022-12-28:
Thanks, understood, this has helped me to understand the serial.print/write, but not entirely sure why sometimes my SD becomes unformatted. 

Still at a loss with regards to how to handle bytes, but perhaps I am asking the wrong question.

I am using a Teensy 4 to store a large byte array onto the SD.

I have a byte array of est 8112 bytes ` DMAMEM byte ui_img_msg_png_data[]  = {
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xF [... cut ...]
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
};`

When storing bins of est 1200, the byte array saves fine to the SD, but with the above array the bin file always ends up being 0 bytes.

So the real question is, do I have to handle the writing process of a large array to the SD in any particular way?

greiman, 2022-12-28:
Run this program.

It outputs this for a empty SD:

> Type any character to start
> fileSize: 8112
>         8112 TestFile.bin
```

#include "SdFat.h"

DMAMEM byte ui_img_msg_png_data[] = {
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
[... 638 lines cut ...]
  if (!file.open("TestFile.bin", O_RDWR | O_CREAT | O_TRUNC)) {
    Serial.println("open failed\n");
    return;
  } 
  if (!file.write(ui_img_msg_png_data, sizeof(ui_img_msg_png_data))) {
    Serial.println("write failed");
    return;
  }
  size_t n = file.fileSize();
  Serial.print("fileSize: ");
  Serial.println(n);
  file.rewind();
  for (size_t i = 0; i < n; i++) {
    if (file.read() != ui_img_msg_png_data[i]) {
      Serial.println("compare failed");
      break;
    }
  }
  file.close();
  sd.ls(LS_SIZE);
}

void loop() {
}
```

greiman, 2022-12-28:
You may have problems with DMAMEM  on Teensy 4.1.  There are quirks with the cache.

The array is not initialized correctly with DMAMEM.  I removed DMAMEM and the array and file are correct.

Go to the Teensy Fourm for help on DMAMEM.

https://forum.pjrc.com/

el-samiyel, 2022-12-28:
Thanks for the pointers. I am using DMAMEM due to the size of my code base (nearly ran out of space on the T4). So the idea was to save my images in bin format on an SD and read into a DMAMEM byte array on start-up. I have actually got this working, but the issue is that I've had to use python to generate the BIN as I haven't had success with the T4 doing this itself.

As mentioned above, I removed DMAMEM from the initializer, but I still get 'write failed'. I tried again with a smaller byte array:

`byte ui_img_msg_png_data[]   = {
    0x00,0x00,0x00,0x00,0x00, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0x00, 0x00, 0x00, 0x69, 0x52, 0x5A, 0x8A, 0x9A, 0x9A, 0x45, 0x99, 0x9A, 0xC7, 0x99, 0x9A, 0xC7, 0x99, 0x9B, 0xE7, 0x99, 0x9B, 0xE7, 0x99, 0x9B, 0xC7, 0x99, 0x9B, 0xC7, 0x99, 0x9B, 0xA6, 0x99, 0x9B, 0xA6, 0x99, 0x9A, 0xA6, 0x99, 0x9A, 0xA6, 0x99, 0x9A, 0xC7, 0x99, 0x9A, 0x49, 0xA2, 0x9F, 0xCB, 0x8A, 0x8F, 0x82, 0x08, 0x1D, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xE3, 0x28, 0x4C, 0xB2, 0xFC, 0xFF, 0x30, 0xFC, 0xFF, 0xCF, 0xFB, 0xFF, 0xEF, 0xFB, 0xFF, 0x10, 0xFC, 0xFF, 0x30, 0xFC, 0xFF, 0x30, 0xFC, 0xFF, 0xCF, 0xFB, 0xFF, 0xAE, 0xFB, 0xFF, 0x10, 0xFC, 0xFF, 0x10, 0xFC, 0xFF, 0x10, 0xFC, 0xFF, 0x10, 0xFC, 0xFF, 0x30, 0xFC, 0xFF, 0xB2, 0xFC, 0xFF, 0x96, 0xFD, 0xFF, 0x69, 0xC2, 0xDD, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xA2, 0x68, 0x9B, 0x0C, 0xFB, 0xFF, 0xD7, 0xFD, 0xFF,
    0x59, 0xFE, 0xFF, 0x18, 0xFE, 0xFF, 0x18, 0xFE, 0xFF, 0x18, 0xFE, 0xFF, 0xD7, 0xFD, 0xFF, 0x7D, 0xFF, 0xFF, 0xBE, 0xFF, 0xFF, 0x18, 0xFE, 0xFF, 0x38, 0xFE, 0xFF, 0x59, 0xFE, 0xFF, 0x59, 0xFE, 0xFF, 0x59, 0xFE, 0xFF, 0xFB, 0xFE, 0xFF, 0x34, 0xFD, 0xFF, 0xA2, 0xF8, 0xFF, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0x00, 0x60, 0x90, 0xEB, 0xFA, 0xFF, 0x75, 0xFD, 0xFF, 0x14, 0xFD, 0xFF, 0x14, 0xFD, 0xFF, 0x14, 0xFD, 0xFF, 0xD3, 0xFC, 0xFF, 0x10, 0xFC, 0xFF, 0x9E, 0xFF, 0xFF, 0xFF, 0xFF, 0xFF, 0x30, 0xFC, 0xFF, 0xD3, 0xFC, 0xFF, 0x34, 0xFD, 0xFF, 0x34, 0xFD, 0xFF, 0x34, 0xFD, 0xFF, 0xB6, 0xFD, 0xFF, 0xF3, 0xFC, 0xFF, 0x00, 0xF8, 0xF3, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0x00, 0x58, 0x8C, 0xAA, 0xFA, 0xFF, 0x14, 0xFD, 0xFF, 0xB2, 0xFC, 0xFF, 0xD3, 0xFC, 0xFF, 0x30, 0xFC, 0xFF, 0x92, 0xFC, 0xFF, 0x18, 0xFE, 0xFF, 0xBE, 0xFF, 0xFF, 0xBE, 0xFF, 0xFF, 0x9A, 0xFE, 0xFF, 0x55, 0xFD, 0xFF, 0x30, 0xFC, 0xFF, 0xD3, 0xFC, 0xFF, 0xF3, 0xFC, 0xFF, 0x55, 0xFD, 0xFF, 0xD3, 0xFC, 0xFF, 0x61, 0xF8, 0xF2, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0x00, 0x58, 0x8C, 0x49, 0xFA, 0xFF, 0xB2, 0xFC, 0xFF, 0x71, 0xFC, 0xFF, 0xCF, 0xFB, 0xFF, 0xB2, 0xFC, 0xFF, 0xDF, 0xFF, 0xFF,
    0x3C, 0xFF, 0xFF, 0x5D, 0xFF, 0xFF, 0x5D, 0xFF, 0xFF, 0xBA, 0xFE, 0xFF, 0xFF, 0xFF, 0xFF, 0xF7, 0xFD, 0xFF, 0xCF, 0xFB, 0xFF, 0x92, 0xFC, 0xFF, 0xF3, 0xFC, 0xFF, 0x71, 0xFC, 0xFF, 0x41, 0xF8, 0xF2, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0x00, 0x58, 0x8D, 0xE7, 0xF9, 0xFF, 0x71, 0xFC, 0xFF, 0xCF, 0xFB, 0xFF, 0xEF, 0xFB, 0xFF, 0xFF, 0xFF, 0xFF, 0x79, 0xFE, 0xFF, 0xEB, 0xFA, 0xFF, 0x3C, 0xFF, 0xFF, 0xDF, 0xFF, 0xFF, 0x0C, 0xFB, 0xFF, 0xD3, 0xFC, 0xFF, 0xFF, 0xFF, 0xFF, 0x96, 0xFD, 0xFF, 0x8E, 0xFB, 0xFF, 0xB2, 0xFC, 0xFF, 0x51, 0xFC, 0xFF, 0x20, 0xF8, 0xF2, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0x00, 0x58, 0x8D, 0xA6, 0xF9, 0xFF, 0x10, 0xFC, 0xFF, 0xEB, 0xFA, 0xFF, 0xBA, 0xFE, 0xFF, 0x5D, 0xFF, 0xFF, 0xAA, 0xFA, 0xFF, 0xEB, 0xFA, 0xFF, 0x1C, 0xFF, 0xFF, 0x9E, 0xFF, 0xFF, 0x6D, 0xFB, 0xFF, 0xCB, 0xFA, 0xFF, 0x55, 0xFD, 0xFF, 0xDF, 0xFF, 0xFF, 0x30, 0xFC, 0xFF, 0x92, 0xFC, 0xFF, 0x8E, 0xFB, 0xFF, 0x20, 0xF8, 0xF2, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0x00, 0x58, 0x8D, 0x65, 0xF9, 0xFF, 0x8E, 0xFB, 0xFF, 0x6D, 0xFB, 0xFF, 0x9E, 0xFF, 0xFF, 0x92, 0xFC, 0xFF, 0xCB, 0xFA, 0xFF, 0x6D, 0xFB, 0xFF, 0xEF, 0xFB, 0xFF, 0xEF, 0xFB, 0xFF, 0x6D, 0xFB, 0xFF,
    0x6D, 0xFB, 0xFF, 0x0C, 0xFB, 0xFF, 0x9E, 0xFF, 0xFF, 0x14, 0xFD, 0xFF, 0x24, 0xF9, 0xFF, 0x41, 0xF8, 0xFF, 0x00, 0xF8, 0xF2, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0x00, 0x58, 0x8D, 0x24, 0xF9, 0xFF, 0x0C, 0xFB, 0xFF, 0xEF, 0xFB, 0xFF, 0x3C, 0xFF, 0xFF, 0x4D, 0xFB, 0xFF, 0xEB, 0xFA, 0xFF, 0x4D, 0xFB, 0xFF, 0xCB, 0xFA, 0xFF, 0xCB, 0xFA, 0xFF, 0x4D, 0xFB, 0xFF, 0x4D, 0xFB, 0xFF, 0xAA, 0xFA, 0xFF, 0xDB, 0xFE, 0xFF, 0xAE, 0xFB, 0xFF, 0x00, 0xF8, 0xFF, 0x00, 0xF8, 0xFF, 0x00, 0xF8, 0xF2, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0x00, 0x58, 0x8D, 0x04, 0xF9, 0xFF, 0xEB, 0xFA, 0xFF, 0x4D, 0xFB, 0xFF, 0x5D, 0xFF, 0xFF, 0x4D, 0xFB, 0xFF, 0x69, 0xFA, 0xFF, 0xCB, 0xFA, 0xFF, 0xCB, 0xFA, 0xFF, 0xCB, 0xFA, 0xFF, 0x0C, 0xFB, 0xFF, 0x6D, 0xFB, 0xFF, 0x61, 0xF8, 0xFF, 0x75, 0xFD, 0xFF, 0xEF, 0xFB, 0xFF, 0x00, 0xF8, 0xFF, 0x00, 0xF8, 0xFF, 0x00, 0xF8, 0xF2, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0x00, 0x58, 0x8D, 0x41, 0xF8, 0xFF, 0x49, 0xFA, 0xFF, 0x08, 0xFA, 0xFF, 0x5D, 0xFF, 0xFF, 0x14, 0xFD, 0xFF, 0x04, 0xF9, 0xFF, 0xAA, 0xFA, 0xFF, 0x8A, 0xFA, 0xFF, 0xCB, 0xFA, 0xFF, 0x0C, 0xFB, 0xFF, 0xE3, 0xF8, 0xFF, 0x00, 0xF8, 0xFF, 0x1C, 0xFF, 0xFF, 0x86, 0xF9, 0xFF,
    0x00, 0xF8, 0xFF, 0x00, 0xF8, 0xFF, 0x00, 0xF8, 0xF2, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0x00, 0x58, 0x8D, 0x00, 0xF8, 0xFF, 0x82, 0xF8, 0xFF, 0x86, 0xF9, 0xFF, 0x30, 0xFC, 0xFF, 0xFF, 0xFF, 0xFF, 0x69, 0xFA, 0xFF, 0xC3, 0xF8, 0xFF, 0x08, 0xFA, 0xFF, 0x28, 0xFA, 0xFF, 0x00, 0xF8, 0xFF, 0x00, 0xF8, 0xFF, 0x59, 0xFE, 0xFF, 0xF7, 0xFD, 0 [... cut ...] 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00, 0xFF, 0xFF, 0x00,
    };`

and I get 

`initialization done.
fileSize: 1200
compare failed
        1200 TestFile.bin`

Any pointers will be much appreciated.

S

el-samiyel, 2022-12-28:
Forgot to mention, I made slight code changes. `if (!sd.begin(SD_CONFIG)) {
      Serial.println("initialization failed! ");
    } else {
      Serial.println("initialization done.");
    }`

as the SD wouldn't init before. 

Even with the confirmed `fileSize: 1200`, the SD becomes unformatted. 

Kind Regards,

S

el-samiyel, 2022-12-28:
For full clarification, I am using a Teensy 4.0 on a custom board I have developed. Its got me thinking perhaps the traces were becoming induced with noise, but all other SD calls and functions are working just fine. The SD just to seems become unformatted if trying to write large amounts of data to the SD, hence the original query around how to best write large datasets (perhaps in smaller packages 512 bytes). 

NB: I fixed my original issue around the SD becoming unformatted when writing Strings, this was due to an error in my seek function.

Many Thanks

S

el-samiyel, 2022-12-28:
> NB: I fixed my original issue around the SD becoming unformatted when writing Strings, this was due to an error in my seek function.

Edit - Just hit the same snag with strings again. Seems that If I write different strings in quick succession, the SD becomes unformatted. I am using a call back to write to the SD, I will remove this and put it into the code sequentially and see if this makes a difference Or try to slow down the clock speed.

Keeping this updated as this may help someone out in future. (assuming I find a fix)

S

greiman, 2022-12-28:
You should got to: 

https://forum.pjrc.com/

The Teensy Audio Library write many files at high speed.  They use the Teensy port of SdFat.

My [SdioDemo](https://github.com/greiman/SdFat/blob/master/examples/TeensySdioDemo/TeensySdioDemo.ino) writes 32 KB blocks with no problem.

el-samiyel, 2022-12-28:
Yes, will do as I think I have something seriously wrong here, but will take a while to get to the bottom of it. I have pushed the T4 to the max so im not too surprised.

Thanks for all the help!

el-samiyel, 2022-12-29:
Ok last update from me, not ideal but seems to be working.. In my code, I have to detect when an SD write fails, once it fails I force an SD init and then I am able to write to the card.

Its not ideal as it required additional code but keeps the project moving in the right direction.

If anyone else had similar issues and managed to solve, please do let me know.

Regards

S

greiman, 2022-12-29:
You may want to try limiting the SD clock to 25 MHz.  This is set [here](https://github.com/greiman/SdFat/blob/628effa1c277e6dc1fef461b4b755ad300735248/src/SdCard/SdioTeensy.cpp#L730) at line 730.

You may need an adjustment of the drive strength for the uSDHC controller lines.  It is [here](https://github.com/greiman/SdFat/blob/628effa1c277e6dc1fef461b4b755ad300735248/src/SdCard/SdioTeensy.cpp#L339) at line 339.

greiman, 2022-12-29:
>  I have to detect when an SD write fails, once it fails I force an SD init and then I am able to write to the card.

This will insure the card becomes totally corrupt. 

Not only will the data you are trying to write be lost but data that appeared to be written will be lost.  SD cards have huge, multiples 16KB, flash pages and this data is kept in RAM buffers until an entire page is written.  This data will be lost when you init the card.  

Flash record units can be as large as 512KB.

greiman, 2022-12-30:
I assume you are not using a callback such that two threads of execution are in Sdfat at the same time.  This would result in both threads accessing the SD and cause a failure.

el-samiyel, 2023-01-24:
Hi,

Thanks for the above - got round to tinkering with it today but still with not much luck.

I changed the frequency down to 25Mhz, but unsure what the drive strength could be set to, any suggestions?

I also ran your SDIO demo which yields the following results:

`FIFO SDIO mode.

size,write,read
bytes,KB/sec,KB/sec
512,11018.91,11716.91
1024,10819.65,11723.24
2048,11083.92,11726.17
4096,11071.23,11754.79
8192,10952.82,11765.29
16384,Error: write failed
SD_CARD_ERROR_CMD12, ErrorData: 0X0`

And

`FIFO SDIO mode.

size,write,read
bytes,KB/sec,KB/sec
512,10947.26,11711.63
1024,11067.22,11709.31
2048,11002.87,11754.32
4096,11126.15,11761.19
8192,11129.55,11753.48
16384,10788.57,11750.68
32768,10900.04,Error: read failed
SD_CARD_ERROR_CMD12, ErrorData: 0X0`

Any further pointers will be much appreciated.

S