greiman/SdFat #411: Solution for the "definition of macro 'F'  #define F(str) (str)" error in the file SysCall.h

MatchingTitan93, 2022-12-28:
I was using the Arduino Nano RP2040 Connect and trying to use the SdFat library with IrRemote.
This problem was continuously popping in my output as the #define F in the SysCall.h file conflicted with the Mbed library necessary to rub the board I'm using (I think that this will happen on all Arduino's boards if you use even the IrRemote library).
The solution i found was simply renaming the F macro with 'F_MACRO' or anything else.
So, this is what i got:

Lines 50-53 in common/SysCall.h
> ```h
> #ifndef F_MACRO
> /** Define macro for strings stored in flash. */
> #define F_MACRO(str) (str)
> #endif  // F_MACRO
> ```

and replacing all the F macro declarations in SdFat.h with F_MACRO, so for example:

Lines 126-140 in SdFat.h

> ```h
>   /** %Print error info and halt.
>    *
>    * \param[in] pr Print destination.
>    */
>   void errorHalt(print_t* pr) {
>     if (sdErrorCode()) {
>       pr->print(F_MACRO("SdError: 0X"));
>       pr->print(sdErrorCode(), HEX);
>       pr->print(F_MACRO(",0X"));
>       pr->println(sdErrorData(), HEX);
>     } else if (!Vol::fatType()) {
>       pr->println(F_MACRO("Check SD format."));
>     }
>     while (true) {}
>   }
> ```

greiman, 2022-12-29:
The F macro is defined for all standard Arduino boards so I can't change it.

MatchingTitan93, 2022-12-29:
ok thanks 👍