greiman/SdFat #414: feature request: provide a write method for arduino ```F("")``` flash strings

jerabaul29, 2023-01-10:
Arduino provides the ```F("STR")``` method for keeping the string "STR\0" in flash and saving RAM.

Unfortunately, doing:

```
sd_file.write(F("STR"));
```

gives me an error (no match found), while by contrast

```
sd_file.write("STR");
```

works fine.

My feature request: could you provide a method to work directly with ```F("")``` strings, so that we can save some RAM?

greiman, 2023-01-10:
SD.h is an old version of SdFat and it does not support write(F("STR"));

```
#include "SD.h"
File file;
void setup() {
  file.write(F("STR"));
}
void loop() {
}
```
Gives this:

> In file included from C:\Users\Bill\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/Arduino.h:232:0,
>                  from sketch\sketch_jan10a.ino.cpp:1:
> C:\Users\Bill\Documents\Arduino\sketch_jan10a\sketch_jan10a.ino: In function 'void setup()':
> C:\Users\Bill\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\cores\arduino/WString.h:38:28: warning: invalid conversion from 'const __FlashStringHelper*' to 'uint8_t {aka unsigned char}' [-fpermissive]
>  #define F(string_literal) (reinterpret_cast<const __FlashStringHelper *>(PSTR(string_literal)))
>                            ~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
> C:\Users\Bill\Documents\Arduino\sketch_jan10a\sketch_jan10a.ino:4:14: note: in expansion of macro 'F'
>    file.write(F("STR"));
>               ^
> In file included from C:\Users\Bill\Documents\Arduino\sketch_jan10a\sketch_jan10a.ino:1:0:
> C:\Users\Bill\Documents\Arduino\libraries\SD\src/SD.h:36:22: note:   initializing argument 1 of 'virtual size_t SDLib::File::write(uint8_t)'
>        virtual size_t write(uint8_t);
>                       ^~~~~
> Sketch uses 6332 bytes (19%) of program storage space. Maximum is 32256 bytes.
> Global variables use 838 bytes (40%) of dynamic memory, leaving 1210 bytes for local variables. Maximum is 2048 bytes.

Use this:
`
file.print(F("STR"));`

jerabaul29, 2023-01-10:
Thanks :) . Sorry I should have included more context, I was actually already using:

```
#include "SdFat.h"

FsFile sd_file

...
```

I just tested, replacing ```write()``` with ```print(F(""))``` works perfectly, many thanks :) .

So if I understand well then ```write``` is kind of a less good / older / deprecated ```print``` in some sense? Could / should ```write``` methods maybe be marked with a compiler deprecation warning in favor of print then? :)

jerabaul29, 2023-01-10:
(I do not think such a warning is in place, right? At least I do not see it showing up in platformio; or is there a reason / case for not deprecating ```write```? :) ).

greiman, 2023-01-10:
write() is a fine function and is the base for print.

The problem is the F(literal_string) macro.

It was defined to only be used with print(F("string")) or println(F("string")) on AVR when the only Arduino was and AVR with one KB of RAM.

Then other boards came along and the definition was on a board basis.

Now Serial.write(F("string")) gives a compile error on AVR, works correctly on Due, may compile but not work correctly on other boards.  Other classes like File behave in other ways.

So the rule is only use F("...") with print() or println().

jerabaul29, 2023-01-11:
Thanks for the explanation :) .

closed 2023-01-11 by jerabaul29