greiman/SdFat #417: Error using custom SPI pins with ESP32 SDFat library

HWR20080716, 2023-02-19:
Statement: I don't speak English well. Most of the following content is obtained from translation software.
I am using the platformio ide to write programs. The following issue is generated by using chatgpt, but only half of it is correct. I still haven't successfully initialized my SD. My requirement is to use a custom spi pin to initialize, because my development board is strange. Otherwise, the SD card cannot be initialized.

> Issue Title: Error using custom SPI pins with ESP32 SDFat library.
Problem Description: I'm using an ESP32 development board and the SDFat library to read and write data to an SD card. I want to use custom SPI pins (such as using IO5 as the SCK pin), but I'm encountering an error when attempting to use the custom pins.
Detailed Problem Description: I have successfully connected an SD card to my ESP32 development board and used the SDFat library to read and write files. Now, I want to use custom SPI pins to communicate with the SD card. I found the following statement in the SDFat library source code to define the default SPI pins:
#define SD_SCK_MHZ(SPEED) SPI.beginTransaction(SPISettings(SPEED * 1000000, MSBFIRST, SPI_MODE0)); digitalWrite(SD_SCK, LOW)
I changed it to the following:
#define SD_SCK_MHZ(SPEED) SPI.beginTransaction(SPISettings(SPEED * 1000000, MSBFIRST, SPI_MODE0)); digitalWrite(5, LOW)
This should change the SCK pin to IO5, but when I try to compile and upload my code, I get the following error:
error: 'SD_SCK' was not declared in this scope
I checked my code and found that the SD_SCK variable is being used by another library, and I changed it to the number 5. I also checked the ESP32 pin mapping table and confirmed that the number 5 is a valid SPI pin.
I don't know how to solve this problem. Do you have any suggestions?
Expected Result: I want to be able to use custom SPI pins to communicate with the SD card instead of the default pins.
Thank you.

Main:
I tried to use many methods to solve this，but they don't work.
①
Use #413 's method
`SpiPort_t spi2 = SpiPort_t(PB15, PB14, PB13);`
but i got  “没有与参数列表匹配的构造函数 "SPIClass::SPIClass" 实例”（There is no instance of constructor "SPIClass:: SPIClass" matching the parameter list.）
(I already #include <SPI.h>)
②
Use #413 's method(But I change something.)
#define SD_CONFIG SdSpiConfig(SD_CS_PIN, 0, SHARED_SPI, SpiPort_t(SDCARD_SCK_PIN, SDCARD_MISO_PIN, SDCARD_MOSI_PIN))
When I try this, I just got "no matching function for call to 'SPIClass::SPIClass(int, int, int)'"
(I already #include <SPI.h>)
③
Change config like [https://github.com/greiman/SdFat-beta/issues/40](url)

Please tell me how to fix this ploblem.
Thanks.

P.S. 

1. If I do not use these methods I will get them:
(I use sd.initErrorHalt(&Serial); to get the error code.)
SS 5
SCK 18
MISO 19
MOSI 23
begin() failed
Do not reformat the SD.
SdError: 0XC,0XFF
2. I used custom spi to run SD library like this:
  pinMode(15, OUTPUT);
  digitalWrite(15, HIGH);
  SPI.begin(14, 12, 13);
  SPI.setFrequency(1000000);
(My MISO_PIN is 12 , MOSI_PIN is 13 , SCK_PIN is 14 , SS_PIN is 15)

greiman, 2023-02-19:
Try this:
```
// latest from https://github.com/greiman/SdFat-beta
#include "SdFat.h"

SdFat sd;

#define CS_PIN 15
#define SD_CONFIG SdSpiConfig(CS_PIN, USER_SPI_BEGIN, SD_SCK_MHZ(10000000))

void setup() {
  Serial.begin(115200);
  while (!Serial) {}
  pinMode(15, OUTPUT);     // probably not needed
  digitalWrite(15, HIGH);  // probably not needed
  SPI.begin(14, 12, 13);
  SPI.setFrequency(1000000);  // probably not needed
  if (!sd.begin(SD_CONFIG)) {
    sd.initErrorHalt(&Serial);
  }
  Serial.println("Success");
}

void loop() {
}
```

HWR20080716, 2023-02-22:
Sorry,I tried this but it didn't work.
It seems that 'USER_SPI_BEGIN' was not defind.

greiman, 2023-02-22:
Did you do this - first line of above program:
`// latest from https://github.com/greiman/SdFat-beta`

USER_SPI_BEGIN is defined in [SdFa-beta here](https://github.com/greiman/SdFat-beta/blob/master/src/SpiDriver/SdSpiDriver.h#L52).

Install the latest version of SdFat-beta.

HWR20080716, 2023-02-23:
Sorry,I installed SdFat rather than SdFat-bata.
I will try to install SdFat-bata.
P.S. 
Is there a solution that can be solved with only SdFat?
Thank you.

greiman, 2023-02-23:
> Is there a solution that can be solved with only SdFat?

No but the next version of Sdfat will contain all new Sdfat-beta code.

HWR20080716, 2023-02-26:
I‘m using platformio.
So I don't know how to install SdFat-bata.
Could you please help me?
When I replace SdFat by SdFat-beta , platformio would auto install SdFat instead of using my file.

HWR20080716, 2023-02-26:
It works!!!
Thank you!!!

HWR20080716, 2023-02-26:
pinMode(15, OUTPUT);     // probably not needed
  digitalWrite(15, HIGH);  // probably not needed
  SPI.begin(14, 12, 13);
  SPI.setFrequency(1000000);  // probably not needed
                                                     ↑
yes
I delete it and the code still working.

closed 2023-02-26 by HWR20080716

zuichen123, 2023-08-20:
please spell chinese(

vednildip, 2024-10-07:
MISO_PIN is 12 , MOSI_PIN is 13 , SCK_PIN is 14 , SS_PIN is 5....... SD  
MISO_PIN is 19 , MOSI_PIN is 23 , SCK_PIN is 18 , SS_PIN is 15   ...TFt 
Below  work  in  Isolation  but  **_how  to  code  VSPI  and HSPI  user  defined  dedicated  pins_** ? as  both  VSPI  and HSPI used then  SD  fail  to begin
"
SdFat sd;
#define CS_PIN 15
#define SD_CONFIG SdSpiConfig(CS_PIN, USER_SPI_BEGIN, 1000000)
void setup() {
  Serial.begin(115200);
  while (!Serial) {}
  pinMode(15, OUTPUT);     // probably not needed
  digitalWrite(15, HIGH);  // probably not needed
  SPI.begin(14, 12, 13);
  SPI.setFrequency(1000000);  // probably not needed
  if (!sd.begin(SD_CONFIG)) {
    sd.initErrorHalt(&Serial);
  }"
  Serial.println("Success");
}"

_________________________________