greiman/SdFat #419: How to deploy SdSpiSTM32.cpp

JK-Pair, 2023-02-28:
I'm currently working on STM32 Bluepil (https://github.com/rogerclarkmelbourne/Arduino_STM32) 
I would like to decrease saving time (about 20ms, which is not enough), So I saw the DMA function in the SdSpiSTM32.cpp would help me solve that problem. By looking at `examples/UserSPIDriver/UserSPIDriver.ino`, Still have no idea about how it works. 

Could anyone give me an example of using the SdSpiSTM32.cpp?

greiman, 2023-02-28:
you didn't provide enough information about your problem so I can't help.

DMA can provide better transfer rates but does not help with some SD latency problems.

JK-Pair, 2023-02-28:
My problem is the SD latency, which took time around ~15ms to save the data into SD Card, which means the MCU will be interrupted for 15ms. I don't want that. I thought the DMA function could help me solve that problem, still have no idea. Do have any suggestions about my problem? (Can't use RTOS)

greiman, 2023-03-01:
See [this](https://github.com/greiman/SdFat/blob/master/examples/ExFatLogger/ExFatLogger.ino) example for a program that overcomes latency.

[This](https://github.com/greiman/SdFat/blob/master/examples/TeensySdioLogger/TeensySdioLogger.ino) example logs data at more than 25 ksps.

JK-Pair, 2023-03-02:
Thank you so much for your help. I still want to try the DMA method, Could you also give me an example of using it?

greiman, 2023-03-02:
You won't get any latency help from DMA.  You must look at the above examples to avoid long latency while the SD is busy. 

You must pre-allocate the file, use file.isBusy() to buffer data until the SD is not busy.

to use DMA replace [this](https://github.com/greiman/SdFat/blob/628effa1c277e6dc1fef461b4b755ad300735248/examples/UserSPIDriver/UserSPIDriver.ino#L36-L42) and [this](https://github.com/greiman/SdFat/blob/628effa1c277e6dc1fef461b4b755ad300735248/examples/UserSPIDriver/UserSPIDriver.ino#L48-L53) with calls to SPI DMA functions.