greiman/SdFat #421: USE_STIMMER problem on ARM

jgfoster, 2023-03-03:
`src/common/FmtNumber.cpp:33` defines `USE_STIMMER` if the platform is not x86_64. Unfortunately, this causes compile errors on ARM. I've solved the problem with the following:

```c++
// Use Stimmer div/mod 10 on avr
#ifdef __AVR__
#include <avr/pgmspace.h>
#ifndef __x86_64__
#ifndef __aarch64__
#define USE_STIMMER
#endif  // __aarch64__
#endif  // __x86_64__
#endif  // __AVR__
```

greiman, 2023-03-03:
How does it fail?

jgfoster, 2023-03-03:
```
/Users/jfoster/Documents/Arduino/libraries/SdFat/src/common/FmtNumber.cpp:168:5: error: invalid output constraint '=d' in asm
    divmod10_asm16(n, r, tmp8);
    ^
/Users/jfoster/Documents/Arduino/libraries/SdFat/src/common/FmtNumber.cpp:73:20: note: expanded from macro 'divmod10_asm16'
      :"+r"(in32), "=d"(mod8), "=d"(tmp8) : : "r0")
                   ^
/Users/jfoster/Documents/Arduino/libraries/SdFat/src/common/FmtNumber.cpp:193:5: error: invalid output constraint '=d' in asm
    divmod10_asm32(n, r, tmp8);
    ^
/Users/jfoster/Documents/Arduino/libraries/SdFat/src/common/FmtNumber.cpp:120:20: note: expanded from macro 'divmod10_asm32'
      :"+r"(in32), "=d"(mod8), "=d"(tmp8) : : "r0")
                   ^
```

greiman, 2023-03-03:
What platform are you compiling on?  Is it Windows Arm-based?

jgfoster, 2023-03-03:
Apple Silicon (but I expect the problem would be the same on Windows).

greiman, 2023-03-03:
Loos like a problem with how asm for AVR is handled on some ARM Windows and Apple ARM devices.

You are the first to post for my libraries  but I see problems for various cross compiles on these devices.

I may do a fix with a #warning so if it is ever fixed someone will let me know.

jgfoster, 2023-03-03:
In my case I don't think it is a cross-compiler issue since I'm trying to compile against a _mock_ of the Arduino (see https://github.com/Arduino-CI/arduino_ci) and run tests in my development environment. As long as the code is pure C/C++ it should work fine. It is when you make assumptions about the hardware architecture that we run into problems.