greiman/SdFat #422: Is SDFAT_FILE_TYPE switch broken?

Datu-Puti, 2023-03-09:
Developing for Atmel (Microchip), ATmega1284p.
My program memory usage is much larger than expected and I used "avr-objdump" to determine that the SdFat library is larger than the README.md suggests it should be when `SDFAT_FILE_TYPE == 1`.  (I summed up to about 17,000 bytes just looking at some of the larger symbols that had FAT in the name.)

I then set `SDFAT_FILE_TYPE = 3` in the SdFatConfig.h file and recompiled but it didn't have any effect on size -- the library seems to be stuck using FAT16/FAT32 and exFAT regardless of SDFAT_FILE_TYPE.  I verified that the switch was being set correctly (to value 1) where used in both files "SdFat.h" and "ios.h" by using the preprocessor directives `#if (SDFAT_FILE_TYPE == 1)` and next line `#error SDFAT_FILE_TYPE is 1` and verifying I saw the error when compiled.

I then repeated the tests with same results after downloading and replacing the SdFat library.

There is one modification I made to file "SdSpiDriver.h", changing `#include SPI.h` to `#include SPI_modified.h` because I have a very minor adjustment to the default SPI library (`SPIClass::end()` returns `initialized` to let me know when SPI has been disabled).  This should have no bearing on the SdFat library.

That's as far as I got debugging.  Is this broken for more than just me?

greiman, 2023-03-10:
Please try the following.  I tried it on a Mega 2560 since I don't have a 1284 board.

I compiled this program with the default SDFAT_FILE_TYPE  defines.
```
#include <SdFat.h>
SdFat sd;
File file;
void setup() {
  sd.begin(SS);
  file.open("SizeTest.txt", O_CREAT | O_RDWR);
  file.println("Size test.");
  file.close();
}
void loop() {}
```
The result was::

> Sketch uses 19370 bytes (7%) of program storage space. Maximum is 253952 bytes.
> Global variables use 764 bytes (9%) of dynamic memory, leaving 7428 bytes for local variables. Maximum is 8192 bytes.

I edited SdFatCongig.h and changed SDFAT_FILE_TYPE  from 3 to 1.
```
/**
 * File types for SdFat, File, SdFile, SdBaseFile, fstream,
 * ifstream, and ofstream.
 *
 * Set SDFAT_FILE_TYPE to:
 *
 * 1 for FAT16/FAT32, 2 for exFAT, 3 for FAT16/FAT32 and exFAT.
 */
#ifndef SDFAT_FILE_TYPE
#if defined(__AVR__) && FLASHEND < 0X8000
// 32K AVR boards.
#define SDFAT_FILE_TYPE 1
#else  // defined(__AVR__) && FLASHEND < 0X8000
// All other boards.
#define SDFAT_FILE_TYPE 1  //  <============== Replace 3 with 1
#endif  // defined(__AVR__) && FLASHEND < 0X8000
#endif  // SDFAT_FILE_TYPE
```
The result was:

> Sketch uses 10188 bytes (4%) of program storage space. Maximum is 253952 bytes.
> Global variables use 707 bytes (8%) of dynamic memory, leaving 7485 bytes for local variables. Maximum is 8192 bytes.
>

greiman, 2023-03-10:
I tried an old 1284 board support package and found a curious problem.  I only compiled since I don't have a working 1284.

On Arduino 2.01 The size for SDFAT_FILE_TYPE 1 is:

> Sketch uses 11882 bytes (9%) of program storage space. Maximum is 130048 bytes.
> Global variables use 712 bytes (4%) of dynamic memory, leaving 15672 bytes for local variables. Maximum is 16384 bytes.

On Arduino 2.01 The size for SDFAT_FILE_TYPE 3 is:

> Sketch uses 23196 bytes (17%) of program storage space. Maximum is 130048 bytes.
> Global variables use 787 bytes (4%) of dynamic memory, leaving 15597 bytes for local variables. Maximum is 16384 bytes.

If I use Arduino 1.8.19 I get this independent of SDFAT_FILE_TYPE.

> Sketch uses 23320 bytes (17%) of program storage space. Maximum is 130048 bytes.
> Global variables use 785 bytes (4%) of dynamic memory, leaving 15599 bytes for local variables. Maximum is 16384 bytes.

Some how the Arduino 1.x does not get the compiler/linker flags to remove unused code with the 1284 package.

greiman, 2023-03-10:
Looks like the 1284 package I used is fairly new.  

https://github.com/MCUdude/MightyCore

Here is the URL for the board manager:

https://mcudude.github.io/MightyCore/package_MCUdude_MightyCore_index.json

Datu-Puti, 2023-03-10:
Sorry, I should have thought to test with the minimum viable "sketch".

I have reproduced the same as you have demonstrated -- the switch is working properly even when using my custom board and compiler switches.

So that leaves something incompatible in my code somehow (which is using 127044 (97%) of 130048 bytes unfortunately; it sure would be nice to have ~8K back.)

I'll see if I can track it down by cutting pieces off until it starts working.  If I find anything helpful I'll update this thread for others that might see a similar issue.

Thanks for your continued support of this fine library!

Datu-Puti, 2023-03-10:
Okay, it was fairly simple to determine the cause.

Instead of cutting code out of my large project I decided to add suspect pieces into the minimum viable example code.  I was using a helper unit to manage the VDD pin on the SD card (shutting power off after session was over to get a lower sleep current).

In that helper I had declared with global scope in the cpp file:
```
SdFs sd;
FsFile file;
```
...which if I replace with...
```
SdFat32 sd;
FatFile file;
```
...I now get the expected size result.

I realize these are not equivalent as I immediately lose the `FsFile::println()` method.  

Now to look into getting functionality back.

greiman, 2023-03-10:
Try using the combination

```
SdFat32 sd;
File32 file;
```

File32 is derived from the Arduino classes Stream and Print.

greiman, 2023-03-10:
If you only need Print and not the added functions of Stream you can define a FAT16/FAT32 file with print like this:

```
class PrFile : public PrintFile<FatFile> {};
SdFat32 sd;
PrFile file;
```
This may save another 700 bytes.  You can use any name for PrFile.

Here is my test code:
```
#include <SdFat.h>
SdFat32 sd;
class PrFile : public PrintFile<FatFile> {};
PrFile file;
// File32 file;
void setup() {
  sd.begin(SS);
  file.open("SizeTest.txt", O_CREAT | O_RDWR);
  file.println("Size test.");
  file.close();
}
void loop() {}
```

Datu-Puti, 2023-03-11:
Excellent, I think I'm very close to getting this implemented with the smaller code.

Embarrassingly, it took me quite a while to figure out why the minimal code example wasn't actually writing anything -- my SS pin is not default for the SD card; when using the correct pin it works (even though my helper unit was still using `FsFile`, I still noticed a huge reduction in code size).

I don't really need the print functions; I was using `write()` anyway in my actual code.  

I will need to adjust my approach a bit because I was used to the 64-bit addressability of `FsFile` (which I would never have been able to fully fill with log data but now may need to be careful) and the methods are named a bit differently (was using seek, position, etc.) but thanks to your incredibly helpful local HTML reference I think I'll be able to adjust quite quickly.  I was also in the habit of opening files using the form `sd.open()` which isn't supported by `SdFat32` but I think it makes more sense anyway via `file.open()`.

Again, many thanks for the library and the great support!

_By the way, this is my first post on GitHub and I'm not sure if I'm supposed to mark a solution or close the thread or if that's handled by the repository owner.  Let me know..._

greiman, 2023-03-11:
The extra methods for FsFile and File32 are strictly for Arduino.  When I wrote SdFat in 2009 Arduino wrote a wrapper that added sd.open() and several other functions.  Arduino never updates the version of SdFat in SD.h.

file.open() is a bit smaller than sd.open() since it is a wrapper for file.open().

The core functions like FatFile and FatVolume are independent of Arduino and I use the core on other RTOS systems.

You don't need to close issues.  Often people read issues and the open issues appear first in github.

Datu-Puti, 2023-03-11:
Just a final update:

Using the aforementioned tips I was able to free up ~11% of my codespace (it went from 97% usage to 86% -- I wasn't paying attention to exact sizes but that translates to somewhere around 14,300 bytes)!

That's a lot more than I was expecting.  Feeling rich again! 💰 🤑

closed 2023-09-05 by Datu-Puti