greiman/SdFat #423: ExFatLogger Example with ESP32

iharshadec, 2023-03-10:
This is my configuration
ESP32 dev Kit
64GB SD Card

I am trying to run ExFatLogger Example with Only 1 Analog channel read and for that I have changed the logic in **logRecord** function.
Also **ADC_COUNT** is set to **1**
```
void logRecord(data_t* data, uint16_t overrun) {
  if (overrun) {
    // Add one since this record has no adc data. Could add overrun field.
    overrun++;
    data->adc[0] = 0X8000 | overrun;
  }
  else {
    for (size_t i = 0; i < ADC_COUNT; i++) {
      data->adc[i] = analogRead(A0);
    }
  }
}
```
I have also formatted my SD card with SDFormatter example to ExFat. When I run this example code I am getting core paniced error.

> type: 
> b - open existing bin file
> c - convert file to csv
> l - list files
> p - print data to Serial
> r - record data
> t - test without logging
> 
> ExFatLogger00.bin
> preAllocated: 128 MiB
> Guru Meditation Error: Core  1 panic'ed (Unhandled debug exception)
> Debug exception reason: Stack canary watchpoint triggered (loopTask) 
> Core 1 register dump:
> PC      : 0x4000c46c  PS      : 0x00060136  A0      : 0x800d149a  A1      : 0x3ffaff30  
> A2      : 0x3ffaff70  A3      : 0x00000000  A4      : 0x00002000  A5      : 0x3ffb0060  
> A6      : 0x00000400  A7      : 0x00000200  A8      : 0x800d4bd8  A9      : 0x3ffb1ef0  
> A10     : 0x00000006  A11     : 0x00000004  A12     : 0x0000000a  A13     : 0x00000001  
> A14     : 0x00000000  A15     : 0x00000000  SAR     : 0x0000001b  EXCCAUSE: 0x00000001  
> EXCVADDR: 0x00000000  LBEG    : 0x4000c46c  LEND    : 0x4000c477  LCOUNT  : 0x000001f0  
> 
> Backtrace: 0x4000c46c:0x3ffaff30 0x400d1497:0x3ffaff40 0x400d19ad:0x3ffb1f90 0x400d4b31:0x3ffb1fb0 0x400885bd:0x3ffb1fd0
> 
> Rebooting...
> Type any character to begin

greiman, 2023-03-10:
The problem is stack overflow.  

>  Stack canary watchpoint triggered (loopTask)

On most boards large arrays can be on the stack.  ESP32 uses FreeRTOS and the user stack is fairly small.

The arrays causing problems are [here](https://github.com/greiman/SdFat/blob/628effa1c277e6dc1fef461b4b755ad300735248/examples/ExFatLogger/ExFatLogger.ino#L191) and [here](https://github.com/greiman/SdFat/blob/628effa1c277e6dc1fef461b4b755ad300735248/examples/ExFatLogger/ExFatLogger.ino#L305).

The ESP32 has lots of RAM so you can fix this by making the arrays static like this:

`  static uint32_t fifoBuf[128 * FIFO_SIZE_SECTORS];`

`  static data_t binData[FIFO_DIM];`

There are lots of other ways to fix the problem.

Also you may need to decrease the SPI clock from 50MHz.  ESP32 SPI should choose the max speed it can support less than or equal to 50 MHz but this seems to fail on some versions of the ESP32 board support package.

iharshadec, 2023-03-10:
SPI Clock is already set to 25. Let me try to fix the ram thing and see if this is fixing it

closed 2023-03-10 by iharshadec