greiman/SdFat #424: ESP32 issues

ITstreet1, 2023-03-10:
I have a hard time making this lib work with the ESP32 dev kit. 

First and above all. 
My wiring is OK as the SD lib works great. This lib is causing me a headache.

I am using the SoftwareSpi example and these pins:
const uint8_t SD_CS_PIN = 22;
const uint8_t SOFT_MISO_PIN = 13;
const uint8_t SOFT_MOSI_PIN = 12;
const uint8_t SOFT_SCK_PIN  = 14;

At first boot, I had the "wiring error".
First I had to check in the config file and select the SPI DRIVER SELECT to 2. After this, the ESP32 recognized the card, but now it says to check the SD format. Inside the sketch, there is an SD FILE TYPE set to 0. Tried 1, but nothing. Still, it says to check the SD format, which is FAT32. Any suggestions?

Question nr 2:
I have an older sketch that worked but with a constructor SdFatSoftSpi, which now is SoftSpiDriver. What should I do to make an easy transition? Just to rename it?

greiman, 2023-03-10:
I can't help with software SPI on ESP32.  Perhaps someone has a working ESP32 software SPI package and you can use it as [UserSPI](https://github.com/greiman/SdFat/blob/master/examples/UserSPIDriver/UserSPIDriver.ino).

There are too many boards for me to support with Software SPI.

greiman, 2023-03-10:
Strange, I tried the Software SPI example on an ESP32 Dev Module and it worked.

I used a 32GB FAT32 SD.

The output was:

> This line was printed using software SPI.
> Done.

greiman, 2023-03-10:
I tried these pins on another ESP32 module and they work.

```
// Chip select may be constant or RAM variable.
const uint8_t SD_CS_PIN = 22;
//
// Pin numbers in templates must be constants.
const uint8_t SOFT_MISO_PIN = 13;
const uint8_t SOFT_MOSI_PIN = 12;
const uint8_t SOFT_SCK_PIN  = 14;
```

greiman, 2023-03-10:
All I did was set SPI_DRIVER_SELECT to 2 and defined the four pins to use. 

I didn't change SD_FAT_TYPE from zero since the default is to support FAT16/FAT32 and exFAT.