greiman/SdFat #425: Some oddness after a seek()

zyeborm, 2023-03-16:
I'm writing a little SD data logger and as part of the operation I was hoping to occasionally sync then go back and read the last line just to make sure the data is actually being written successfully.

I'm preallocating the file then writing 512 byte blocks to the card every second.
Typically this is happening in <1 milliseconds 
Everything seems great.

However if just once I do a 
file.seek(file.curPosition());
All writes from then on take 2-4ms which make me suspect there's now more going on under the hood which may not be good for the life of the SD card. (read/modify/write etc)

Initially I was seeking to curPosition - 512, doing a read of 512 which should leave me at the same spot as I left off I presume. But I have narrowed it down to just that seek.

Is this something that's part of SdFat or perhaps the card itself seeing the read and not buffering the writes or some such from then on?

So, not really sure if it's an "issue" as such, but it might be?

Because I know it's your favourite platform I'm using an ESP32 and I'm using FsFile.
Thanks so much for your library, it's really very very useful.

greiman, 2023-03-16:
The flash pages on an SD are huge and the buffering and access is designed to be a sequential stream if you seek to a previous sector, I must cancel the current multi-byte write, start a read for the sector, then cancel the read and start a write.

You won't damage the card but it will take a long time.

zyeborm, 2023-03-17:
If its left writing in that state will it restart the multi-byte write when it crosses some boundary or is that only possible at file creation until it is disturbed? 
Also is there some way I can shoot you through a few dollars or something? I really appreciate the existence and quality of this library and I'd like to help a little if I can.

greiman, 2023-03-17:
The problem is internal to the SD.  When you end a multi-byte write, the SD will program a flash page which contains many, 32 or more emulated 512 byte sectors.  On high end SD cards flash pages can be much larger.  

When a new write is started, the SD must copy data form the previous flash page to a new erased page and continue the write.  You may be increasing the data rates in the SD by a factor of 50 or more.

Fortunately SD cards have excellent wear leveling so every flash page of an SD will be equally used and it is almost impossible to wear an SD out with a MPU using SPI.  You never achieve the 100 MB/sec a modern SD is capable of.