greiman/SdFat #427: Question: Calling only sync() and not close()

devbhullar, 2023-03-30:
I had an issue where opening a file was taking upwards of 2 seconds. I only observed this when the SD card already had about 6GB of data in it with > 4000 total files. With a fresh SD card, the same call would be a few ms or less.

I'm not sure what I can do to reduce this time to open the file. So, I've tried only calling `open()` once in the beginning of the program, and then just call `sync()` instead of `close()` after writing to the file with `println()`. This reduces the total time for each operation from 2s to 4ms (except for the very first time where I still need to call `open()`).

Is there another way I can reduce the time it takes to open the file? Or is it okay to just call `sync()` without ever calling `close()`? I'm doing continuous logging so there really isn't a way to gracefully call `close()` before the SD card is physically pulled out.

For reference, I'm using v1.0.16 with Particle Electron.

greiman, 2023-03-31:
FAT filesystems do not index directories so a linear search is required to find files in open.  To create a file, the entire directory must be read to insure the name is unique.

One way to decrease open time is to use a directory tree to reduce the number of files in each directory.

close() is just a call to sync()  plus the directory entry is marked closed.   

```
bool FatFile::close() {
  bool rtn = sync();
  m_attributes = FILE_ATTR_CLOSED;
  m_flags = 0;
  return rtn;
}
```

devbhullar, 2023-04-03:
Got it, so I can continue to use sync() without close(). 
I tried changing the directory structure so that the file I write to is in a separate folder and the other 4000+ files are in another folder. That didn't reduce the time to open the file. I'm assuming this is a peculiarity of the FAT filesystem but could be that I'm using v1.0.16.

greiman, 2023-04-03:
You can use sync without closing the file.

The time to open a file depends on which folders must be searched.  If you use a path, all segments of the path must be found. 

If you use `sd.chdir(subdirPath)` you can avoid the overhead of searching sub-directories.  See [this](https://github.com/greiman/SdFat/blob/master/examples/DirectoryFunctions/DirectoryFunctions.ino) example.

devbhullar, 2023-04-03:
Excellent! Using `chdir` works as expected. Time to open file is about 4ms.

closed 2023-04-03 by devbhullar